corundum/corundum #110: Basic verilog question: Why is x_reg and x_next used instead of just using x_reg?

zaidtahirbutt, 2022-07-21:
I just have a basic question. In the code forexample a signal is x_reg, then x_next is updated first and its value is transferred to x_reg on the next clk cycle. Why is on clk cycle wasted when x_reg can directly be assigned the value?

Something like this below happens which require 1 extra clk cycle. Why not update temp_tdata_reg  directly in the FSM?

```
always @(posedge clk) begin
temp_tdata_reg <= temp_tdata_next;
end
```

alexforencich, 2022-07-21:
The convention is `_next` is updated combinatorially, and `_reg` is updated sequentially from `_next`.  So `_next` will be set in a combinatorial always block in the same clock cycle, and then `_next` will be transferred into `_reg` in a separate sequential always block.

This convention is used to avoid "mixing" assignment types in the same always block (`=` vs `<=`).  There are quite a few times where results of one operation have to be used as inputs to another operation or conditional in the same clock cycle, so using `=` is required.  It would technically be possible to use all clocked always blocks and no `_next`, but then assignment types would have to be mixed, which is problematic for a number of reasons.

zaidtahirbutt, 2022-07-21:
@alexforencich got it, thanks a lot for the detailed explanation!

closed 2022-07-21 by zaidtahirbutt