corundum/corundum #114: Support of PCIe “Non-Transparent Bridge" (NTB) function

ohault, 2022-08-16:
Support PCIe “[Non-Transparent Bridge](https://events.static.linuxfound.org/sites/events/files/slides/Linux%20NTB_0.pdf)" (NTB) function could be a great interest for Corundun as a DPU enabler?

One key use case would be to support [RDMA over NTB](https://www.snia.org/sites/default/files/SDC15_presentations/hw/RolandDreier_PCI_Express_Non-Transparent.pdf ) to complement [RDMA development roadmap](https://github.com/corundum/corundum/wiki/RDMA-Development-Roadmap) 

Ref : https://github.com/jonmason/ntb

alexforencich, 2023-08-28:
Not sure exactly what this would look like.  Seems like NTB would require multiple PCIe endpoints, connected to different host CPUs?  Is there a particular multi-headed card that would even make sense for something like this?  Like maybe the 250-SoC with additional PCIe lanes connected to the expansion connectors?  Or would the data be transferred via the network end in some way - Ethernet, Aurora, etc.?

ohault, 2023-08-29:
PCIe interconnect with multiple PCIe Endpoints and hosts (FPGA, DPU, CPU, GPU, ...) can be organized in several topologies. The goal of this ticket would be to implement the support of NTB for FPGA, to provide a versatile and flexible way to support cross-domain communication at PCIe level without relying on ETH PHY.