corundum/corundum #117: support separate Tx/RX queue per CPU core？

zhouandy1985, 2022-09-22:
hi , we are now evaluating corundum。we consider multiple queue and multiple core scenario. we bind 1 tx/rx queue pairs to 1 cpu core,  4 pairs total, and found that corundum can't works well in multiple queue and multiple core, even we add lock to ensure atomic PCIE MMIO operations, only 1 queue pair works well, others seems hang there after several packets transferring.  Something I can't figure out is that corundum works well in multiple tx/rx queue bond to 1 cpu core case.  
Any suggestions?  Considering multiple queue and multiple core is key to performance. It is a must for corundum.
 Thanks

alexforencich, 2022-09-22:
This explicitly should work.  If it isn't, then there must be some sort of a bug.  Can you provide more details about your setup - what FPGA, what board, what host system, how are you generating the traffic, etc.

zhouandy1985, 2022-11-11:
fixed

closed 2022-11-11 by zhouandy1985