corundum/corundum #118: AU50 25G test bench - The AXI Stream isn't being converted to 128 bits from 64 bits in the async FIFO

zaidtahirbutt, 2022-09-30:
I ran the test bench of AU50 25G example. I saw this written in the Corundum research paper:

> For 25G Ethernet, the MAC interface is 64 bits at 390.625 MHz, necessitating a conversion to 128 bits to provide sufficient bandwidth at 250 MHz.

Please refer to the image attached below.

So I checked the simulation waveform, I thought the output rx data (wire axis_rx_async_fifo_tdata) from the async fifo (in mqnic_port_rx.v) would be 128 bits wide but it is 64 bits wide.

I checked the rx_clk is around 390 MHz and clk is 250 MHz.

Is the conversion to 128 bits happening somewhere else?

[image]

alexforencich, 2022-09-30:
Conversion takes place here: https://github.com/corundum/corundum/blob/master/fpga/common/rtl/mqnic_port_rx.v#L233

Corundum uses MACs without built-in FIFOs so that additional logic can be inserted between the MAC and FIFO, including in the application section.

zaidtahirbutt, 2022-09-30:
Yes this https://github.com/corundum/corundum/blob/master/fpga/common/rtl/mqnic_port_rx.v#L233 is the async fifo I am referring to that you have mentioned.

The input AXI-stream **_axis_rx_in_tdata_** is 64 bit wide in this [line ](https://github.com/corundum/corundum/blob/56fe10f27d9b42f1ff9abe4d735b113008e4be9d/fpga/common/rtl/mqnic_port_rx.v#L255 )and the output AXI-stream is also 64 bit wide in this [line ](https://github.com/corundum/corundum/blob/56fe10f27d9b42f1ff9abe4d735b113008e4be9d/fpga/common/rtl/mqnic_port_rx.v#L267) which is also visible in the simulation snapshot I posted in the question. I thought that the output axi stream wire **_axis_rx_async_fifo_tdata_** would be 128 bits wide as per the text in the research paper to match the through put of 25G.

I went deeper into the async fifo adapter and the input output buses of the fifo are also 64 bits wide as in the screenshot below. I still can't find where the conversion of 64 bit bus to 128 bits is happening. Can you point that to me? Thanks for the help.

[image]

alexforencich, 2022-10-01:
I just tested the 25G U50 testbench, and axis_rx_async_fifo_tdata is 128 bits.  Are you using the latest version from git, without any modifications?

zaidtahirbutt, 2022-10-03:
Thanks for testing it out!! I was running a few months old version of Corundum which was showing a 64 bit rx async fifo output.
I just tested the latest Corundum 25G U50 testbench (I had to update the cocotbext-pcie for it to run), and axis_rx_async_fifo_tdata is 128 bits as seen in the screenshot below. Cheers. 

[image]

closed 2022-10-03 by zaidtahirbutt