corundum/corundum #121: IF=4 fails to synthesize on AU50 due to address overlap issue

vaniaprkl, 2022-10-17:
Hi Alex,

I was trying to see if I can have a 4 interface design instantiated using Corundum on the Alveo U50. I am new to your FPGA hierarchy and just started looking at this code last week. A direct build with IF_COUNT=1 works and I can receive packets. Changing IF_COUNT=4 in config.tcl in the fpga_10g directory causes an address overlap. 

```
INFO: [Synth 8-251] 3 (0): c00000 / 22 -- c00000-ffffff [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:147]
INFO: [Synth 8-155] case statement is not full and has no default [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:242]
        Parameter DATA_WIDTH bound to: 32 - type: integer
        Parameter ADDR_WIDTH bound to: 24 - type: integer
        Parameter STRB_WIDTH bound to: 4 - type: integer
        Parameter AR_REG_TYPE bound to: 2'b00
        Parameter R_REG_TYPE bound to: 2'b10
        Parameter DATA_WIDTH bound to: 32 - type: integer
        Parameter ADDR_WIDTH bound to: 24 - type: integer
        Parameter STRB_WIDTH bound to: 4 - type: integer
        Parameter AR_REG_TYPE bound to: 2'b01
        Parameter R_REG_TYPE bound to: 2'b00
        Parameter S_COUNT bound to: 4 - type: integer
        Parameter M_COUNT bound to: 3 - type: integer
        Parameter DATA_WIDTH bound to: 32 - type: integer
        Parameter ADDR_WIDTH bound to: 17 - type: integer
        Parameter M_BASE_ADDR bound to: 0 - type: integer
        Parameter M_ADDR_WIDTH bound to: 96'b000000000000000000000000000100000000000000000000000000000001000000000000000000000000000000010000
        Parameter M_CONNECT_READ bound to: 12'b111111111111
        Parameter M_CONNECT_WRITE bound to: 12'b111111111111
        Parameter S_COUNT bound to: 4 - type: integer
        Parameter M_COUNT bound to: 3 - type: integer
        Parameter DATA_WIDTH bound to: 32 - type: integer
        Parameter ADDR_WIDTH bound to: 17 - type: integer
        Parameter STRB_WIDTH bound to: 4 - type: integer
        Parameter S_ACCEPT bound to: 128'b00000000000000000000000000010000000000000000000000000000000100000000000000000000000000000001000000000000000000000000000000010000
        Parameter M_REGIONS bound to: 1 - type: integer
        Parameter M_BASE_ADDR bound to: 0 - type: integer
        Parameter M_ADDR_WIDTH bound to: 96'b000000000000000000000000000100000000000000000000000000000001000000000000000000000000000000010000
        Parameter M_CONNECT bound to: 12'b111111111111
        Parameter M_ISSUE bound to: 96'b000000000000000000000000000100000000000000000000000000000001000000000000000000000000000000010000
        Parameter M_SECURE bound to: 3'b000
        Parameter S_AW_REG_TYPE bound to: 8'b00000000
        Parameter S_W_REG_TYPE bound to: 8'b00000000
        Parameter S_B_REG_TYPE bound to: 8'b01010101
        Parameter S bound to: 0 - type: integer
        Parameter S_COUNT bound to: 4 - type: integer
        Parameter M_COUNT bound to: 3 - type: integer
        Parameter ADDR_WIDTH bound to: 17 - type: integer
        Parameter M_REGIONS bound to: 1 - type: integer
        Parameter M_BASE_ADDR bound to: 0 - type: integer
        Parameter M_ADDR_WIDTH bound to: 96'b000000000000000000000000000100000000000000000000000000000001000000000000000000000000000000010000
        Parameter M_CONNECT bound to: 12'b111111111111
        Parameter M_SECURE bound to: 3'b000
        Parameter WC_OUTPUT bound to: 1 - type: integer
INFO: [Synth 8-251] Addressing configuration for axil_crossbar_addr instance  [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:144]
INFO: [Synth 8-251] 0 (0): 0 / 16 -- 0-ffff [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:147]
INFO: [Synth 8-251] 1 (0): 10000 / 16 -- 10000-1ffff [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:147]
INFO: [Synth 8-251] 2 (0): 0 / 16 -- 0-ffff [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:147]
INFO: [Synth 8-251] Overlapping regions: [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:177]
INFO: [Synth 8-251] 0 (0): 0 / 16 -- 0-ffff [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:178]
INFO: [Synth 8-251] 2 (0): 0 / 16 -- 0-ffff [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:185]
ERROR: [Synth 8-6058] Synth Error: Error: address ranges overlap (instance )  [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:192]
WARNING: [Synth 8-693] zero replication count - replication ignored [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:52]
WARNING: [Synth 8-693] zero replication count - replication ignored [/disk1/corundum-oct14th/corundum/fpga/mqnic/AU50/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:52]

```

nightseas, 2022-10-23:
The memory size mapped to PCIe BAR0 isn't large enough to assign the register addresses needed. You need to change the BAR size from 16M to a larger value, e.g. 64M.

vaniaprkl, 2022-10-24:
Thanks for your response nightseas. I'm still  a bit confused. 

IF_COUNT=4 used to work with the PCIE parameters already there. I am unsure what broke in the current drop. 
I see this in config.tcl 

```
# Control BAR (BAR 0)
configure_bar $pcie 0 0 [dict get $params AXIL_CTRL_ADDR_WIDTH] 
```
And AXIL_CTRL_ADDR_WIDTH has not changed from the working drop to the not working drop.

Any help in understanding is much appreciated.

alexforencich, 2022-10-24:
Each interface requires a certain amount of address space in the BAR.  If you change the interface count, you may need to increase the size of the BAR to fit everything.  Simple as that.  

What I should probably do is look in to reworking the TCL scripts to compute the BAR size automatically.

vaniaprkl, 2023-05-24:
Increasing BAR size helped. Thanks