corundum/corundum #125: TCAM support?

dtaht, 2022-11-16:
The libreqos.io project is using XDP + EBPF + cake to schedule packets for 10s of thousands of customers. There appears to be a performance wall we are going to hit as we try to crack 40Gbit. In looking over this project, a key feature we need is the ability to do a longest prefix match to processor to interrupt to further process the packets. Is this actually hard to do in a modern FPGA?

https://github.com/LibreQoE/LibreQoS/issues/156

I've of course always wanted to see modern network queuing algorithms like fq_codel (RFC8290) and cake implemented directly in a FPGA also, and in this case, as a transparent bridge, but I don't know what would be involved, what board to start fiddling with (even at 10Gbit), and it does go back to not needing RSS steering at all, but that pesky LPM....

alexforencich, 2022-11-16:
Sounds like you're basically asking about some sort of configurable flow steering?  It's not something that's currently supported, but in principle it should be doable.  

I have definitely seen some several TCAM implementations for FPGAs; I know Vivado has some sort of TCAM generator (https://www.xilinx.com/products/intellectual-property/ef-di-cam.html).  There is always going to be some sort of trade-off in terms of clock speed vs. lookup rate vs. CAM/TCAM size vs. CAM/TCAM features.  But, it looks like the STCAM tops out at 16K entries, which is short of your 64K requirement.  And it's closed, proprietary, probably patented IP, so making internal modifications won't be possible.  There are probably other solutions available, but this is not something that I have looked in to recently.

And I suppose there is also the higher-level question of, does it make sense to implement just flow steering, or perhaps some sort of more general match/action capability?  Or maybe both - the design is highly configurable, maybe start with flow steering and then additionally provide the option for something more full-featured.  Possibly this would be a good idea anyway as I'm sure flow steering only would be more efficient from a resource consumption perspective vs. something more comprehensive.  And since it is an FPGA, no need to worry too much about flexibility as you can just swap out the whole design if necessary.

In terms of a hardware implementation of something like fq_codel, it's worth taking a closer look.  I will note that in many cases, networking-related algorithms do not lend themselves to simple hardware implementation because they tend to be designed by programmers that either know nothing about hardware, or assume that nobody would ever try to offload it into hardware (for example, someone was asking about parsing some of the newer variable-length QUIC and HTTP fields recently....doing this with a general-purpose CPU is not too bad, but doing this in hardware is a total nightmare) so it's possible some modifications/simplifications would need to be made so that it's actually implementable.

dtaht, 2022-11-16:
Speaking as a designer of fq_codel and cake, I can agree with you that most layer 3 folk have historically had a bad interface with layer 2 folk. I've often longed for joint meetings between the IETF and IEEE where it was mandated that the engineers from both groups room, and drink together!

That said, I've looked over the sizzling potentials in your design so far (I was also a fan of SENIC), and can add some layer three insights. 1) strict round robin is not as good as the fast/slow queue distinction fq_{codel,pie,cake} make  for real network traffic, always having near zero latency for traffic that is "sparse". Relevant paper here: https://ieeexplore.ieee.org/document/8469111 - this is the key enhancement to conventional DRR that pushes the implementations of things like google swift, to the best long tail latency we know of. 2) Having zillions of queues without some form of AQM to drop packets intelligently adds latency. I would so love to get away from the essentially default fq_codel + BQL abstraction in linux today and have the hardware do all the work.  

Arguably codel's drop on dequeue behavior is hard to do in a FPGA as it is not fixed time, but with sufficient lookahead into multiple queues and shortcircuiting it, I believe it is doable to do in hardware.

Codel itself has now been put into hardware multiple times: https://queue.acm.org/detail.cfm?id=2209336

In designing cake, we leveraged some interesting hardware concepts. The 8 way hash to flow map, for example, struck us as an off-the shelf IP block for cache, the three hashes we use to create the per host/per flow fq distinction, also offloadable, and there have been multiple hardware flow dissectors also designed in the past. 

We originally designed cake for home gateways but it is now proving useful as a middlebox, wifi tower and vpn concentrator, also. 

old paper here: https://arxiv.org/abs/1804.07617

Anyway, this is actually a distraction from the relative simplicity of the original bug. A simple *large* TCAM-like structure that mapped ipv4/x and ipv6/x addresses to "the right cpu to interrupt" would be a win from where we stand today.

dtaht, 2022-11-16:
cool, thx. I really do wish more folk cared about bufferbloat, fair queuing, and AQM, at this layer, in general.

alexforencich, 2022-11-16:
TBH, it's quite hard to beat a used Alveo U200.  Has PCIe gen 3 x16, two QSFP28 ports for 100G (and 10G/25G/40G/50G), four channels of DDR4 2400, quite a large Virtex US+ FPGA (VU9P with different JTAG ID), and no Vivado license required.  For most other boards that support 10G+, you usually need to have a license for the tools, which can complicate things.  Also, if you get an Alveo board, make sure you get one with a fan if you're going to install it in a desktop machine.   The cards come as either "active" or "passive", with the passive ones relying on chassis fans to provide sufficient airflow.

dtaht, 2022-11-16:
Thank you for the steer. I had worked on the onenetswitch30 project a while back (zynq - still have a couple), and had assembled a lot of the pieces like a netfpga derived DRR module (can't find it now). Catching up to the research, like, https://arxiv.org/pdf/2101.12691.pdf this morning has been fun. I still find it difficult to care about match pipelines, I just want a TCAM... :/

dtaht, 2022-11-16:
Also the invsqrt codel depends on only requires 3k gates, and was only for some crazy reason, available as a p4 extension in the tofino. Weird that basic feature was so missing elsewhere...