corundum/corundum #126: when using pytest has error

liucunshu, 2022-11-17:
try getting started step by step. But it also have error such as:
E       TypeError: expected str, bytes or os.PathLike object, not NoneType
/usr/lib/python3.8/os.py:806: TypeError
There are 101 identical errors. When I test in another host, it correct. But it failed in this host

alexforencich, 2022-11-17:
If it's only an issue on that one machine, then it's probably an issue specific to that machine, such as something wrong with the python environment.  IIRC someone else reported running in to similar errors, and the cause was found to be a botched python installation on the machine in question.

liucunshu, 2022-11-17:
Thank you. I will follow this idea to find out how to solve it.

canchen0720, 2023-10-11:
hello, liu
When I ran tox, I have met the same problem. How did you solve it？It seems to be something wrong with the environment, but I have no idea how to solve it.

libero-x, 2024-08-29:
> hello, liu When I ran tox, I have met the same problem. How did you solve it？It seems to be something wrong with the environment, but I have no idea how to solve it.

hello chen,When I ran tox and pytest, I have met the same problem. How did you solve it？

libero-x, 2024-08-30:
When I changed the second line envlist =py3 to envlist =py3 of tox.ini，It runs success.

libero-x, 2024-08-30:
When I changed the second line envlist =py3 to envlist =py38 of tox.ini，It runs success.

libero-x, 2024-08-30:
When I changed the second line envlist =py3 to envlist =py38 of tox.ini，It runs success. 
My ubuntu is 20.04 LTS，and have installed py3.10