corundum/corundum #128: Enable dma_bench app module

multi-stream, 2022-11-23:
How do I enable app module (dma_bench specifically) and integrate into the mqnic design?

alexforencich, 2022-11-23:
Make a copy of the makefile and config.tcl for one of the design variants for your board, then make the appropriate changes.

Here is an example: https://github.com/corundum/corundum/tree/master/fpga/mqnic/fb2CG/fpga_100g/fpga_app_dma_bench

Basically, all you need to do is set the application block configuration in config.tcl and add all of the required files to the makefile.  So, for example, dma_bench requires config.tcl to be set like so:

    # Application block configuration
    dict set params APP_ID "32'h12348001"
    dict set params APP_ENABLE "1"
    dict set params APP_CTRL_ENABLE "1"
    dict set params APP_DMA_ENABLE "1"
    dict set params APP_AXIS_DIRECT_ENABLE "1"
    dict set params APP_AXIS_SYNC_ENABLE "1"
    dict set params APP_AXIS_IF_ENABLE "1"
    dict set params APP_STAT_ENABLE "1"

The application module is responsible for validating the settings (i.e. it should check APP_ID and the various _ENABLE parameters, and fail the build if the configuration is not acceptable).  For example, the DMA benchmark application really only requires APP_DMA_ENABLE, all of the others can be turned off as all of those interfaces are either tied off or looped through as appropriate.

And then the makefile needs to have

    SYN_FILES += app/dma_bench/rtl/mqnic_app_block_dma_bench.v

This one is simple and doesn't require any additional modules beyond the main application module, but in principle other application modules might require more files.

multi-stream, 2022-11-23:
Thanks for help Alex! This builds the overall design correctly for AU50. However, the mqnic_app_dma_bench.ko is crashing the kernel. The NIC function also stopped working. Is this expected? Thanks!

[28427.900941] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: mqnic_app_dma_bench_probe() called
[28427.900966] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: Allocated DMA region virt 00000000348324b4, phys 00000000652cff22
[28427.900972] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: Statistics counters
[28427.900977] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_mem_rd: 0
[28427.900982] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_mem_wr: 0
[28427.900986] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_io: 0
[28427.900990] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_cfg: 0
[28427.900995] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_msg: 0
[28427.900998] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_cpl: 0
[28427.901002] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_cpl_ur: 0
[28427.901006] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_cpl_ca: 0
[28427.901010] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_atomic: 0
[28427.901014] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_ep: 0
[28427.901018] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_hdr_dw: 0
[28427.901021] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_req_dw: 0
[28427.901025] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_payload_dw: 0
[28427.901029] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_rx_tlp_cpl_dw: 0
[28427.901033] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_mem_rd: 0
[28427.901037] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_mem_wr: 0
[28427.901041] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_io: 0
[28427.901044] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_cfg: 0
[28427.901048] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_msg: 0
[28427.901052] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_cpl: 0
[28427.901056] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_cpl_ur: 0
[28427.901059] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_cpl_ca: 0
[28427.901063] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_atomic: 0
[28427.901067] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_ep: 0
[28427.901071] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_hdr_dw: 0
[28427.901074] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_req_dw: 0
[28427.901078] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_payload_dw: 0
[28427.901082] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: pcie_tx_tlp_cpl_dw: 0
[28427.901086] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_op_count: 0
[28427.901089] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_op_bytes: 0
[28427.901093] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_op_latency: 0
[28427.901097] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_op_error: 0
[28427.901101] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_req_count: 0
[28427.901104] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_req_latency: 0
[28427.901108] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_req_timeout: 0
[28427.901112] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_op_table_full: 0
[28427.901116] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_no_tags: 0
[28427.901120] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_tx_no_credit: 0
[28427.901124] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_tx_limit: 0
[28427.901128] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_rd_tx_stall: 0
[28427.901132] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_op_count: 0
[28427.901136] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_op_bytes: 0
[28427.901140] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_op_latency: 0
[28427.901144] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_op_error: 0
[28427.901147] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_req_count: 0
[28427.901151] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_req_latency: 0
[28427.901155] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_op_table_full: 0
[28427.901159] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_tx_no_credit: 0
[28427.901163] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_tx_limit: 0
[28427.901167] mqnic_app_dma_bench.mqnic_app_dma_bench mqnic.app_12348001.0: dma_wr_tx_stall: 0
[... cut ...]
[28427.956322] R13: 0000000000000001 R14: 0000000000000001 R15: 0000000000000000
[28427.956593] FS:  00007f6c08151000(0000) GS:ffff8ed2c0f80000(0000) knlGS:0000000000000000
[28427.956868] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[28427.957143] CR2: 0000564e70115c50 CR3: 00000001143a6001 CR4: 00000000007706e0
[28427.957419] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[28427.957695] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[28427.957982] PKRU: 55555554

alexforencich, 2022-11-23:
Ah I see, it crashed with a divide by zero error.  Looks like it's reading the statistics counters as all zeros, so there is probably an issue with how that module is reading the counters.  I'll take a look and see if I can figure out what's going on.

alexforencich, 2022-11-24:
Try with the code from this repo: https://github.com/ucsdsysnet/corundum .  This should be fixed to read the counters correctly.

multi-stream, 2022-11-28:
Thanks Alex! This works well now. And just one more thing: The dma_bench app breaks the nic function. I assume dma_bench is not intended for nic function to work at the same time, right? Thanks!

alexforencich, 2022-11-28:
Define "breaks the NIC function".  Does it not work at all?  Does it stop working after loading the extension module?  Is something else going on?  The application logic should be completely independent of the core datapath, so aside from sharing the BW everything should work normally.  

I did notice that when I loaded the extension module during my testing last week, the write DMA BW was not what it should be (~75 Gbps, instead of ~100 Gbps), so perhaps there is a bug somewhere that I need to take a closer look at.

multi-stream, 2022-11-29:
So ping to peer server simply stops working. Looking closer within the driver on the tx path, it looks like the packet is created but tx dma is not proceeding when the tx_queue head pointer is updated.

alexforencich, 2022-11-29:
Okay, does this always happen with this design, or for example only after loading the extension kernel module?

multi-stream, 2022-11-29:
This always happens with or without the extension kernel module.

alexforencich, 2022-11-29:
Strange.  I will take a closer look, that shouldn't be the case.

alexforencich, 2022-12-03:
I made some headway on this; I was able to replicate the issue in HW, and it seems that for some reason the card is not issuing any interrupts.  The event queues have data when they should be empty, and /proc/interrupts shows no activity.  Are you seeing the same on your end?  I have not yet been able to replicate the issue in simulation, however.

alexforencich, 2022-12-03:
Ah.  Looks like something gets screwed up in the PCIe core config for MSI-X when the application section is enabled.

Working config, as reported by lspci:

	Capabilities: [60] MSI-X: Enable- Count=64 Masked-
		Vector table: BAR=0 offset=00010000
		PBA: BAR=0 offset=00018000

Broken config, as reported by lspci:

	Capabilities: [60] MSI-X: Enable- Count=64 Masked-
		Vector table: BAR=0 offset=00000040
		PBA: BAR=0 offset=00000050

So, the OS is looking for the MSI-X table in the wrong spot and things don't get configured correctly.  

Apparently when the application BAR is enabled, the MSI-X settings get changed and need to be re-applied.  Frapping vivado!  This should just be a change to some TCL scripts, I will let you know when I get it sorted out.  

Also, I refactored the DMA bench application and peeled the actual DMA benchmark portion into a separate verilog module.  It's looking like the DMA bench application could be a good vehicle for testing various aspects of the design, particularly the memory interfaces, so I went ahead and added targets for all of the Alveo boards, and will eventually add it for all of the boards.

alexforencich, 2022-12-04:
Try the latest code at https://github.com/ucsdsysnet/corundum .  This at least fixes the MSI-X problem so the main NIC functionality will work.  However, the DMA write test is still reporting ~76 Gbps.  Interestingly, the NIC RX still hits 100G line rate, so it looks like I still have some debugging to do.

alexforencich, 2022-12-04:
Fixed the DMA write BW limit, but I'm still not sure exactly what was going on.  Basically, PCIe flow control used to be handled more directly by the DMA engines, and now it's handled in the shims.  Initially, the number of operations in the TX pipeline was limited so that backpressure from the PCIe IP core would work correctly.  But then I rewrote the shims to take advantage of the segmented interface, and in doing so I also had to move the flow control into the shim since the shim now contained some largeish FIFOs.  When I did that, I didn't change the number of in-flight writes, so it was left at an unreasonably low number (3).  Changing that to a significantly larger number has improved the write DMA throughput, but what I don't understand at this point is why the NIC was able to issue DMA writes faster than the DMA benchmarking code, since the limit applied to the DMA engine itself and both components use the exact same DMA engine.

alexforencich, 2022-12-05:
Should be fully fixed now at https://github.com/ucsdsysnet/corundum.  I will be getting that merged back into this repo relatively soon.

alexforencich, 2022-12-07:
And the main repo has been updated.  Please let me know if you're running in to any other issues with the DMA benchmark design.

multi-stream, 2022-12-09:
Hi Alex, Thanks for taking the time investigate the issue. 
It's been busy at the office. I'll try building it from the main repo tomorrow morning and report back here as soon as I can.
Thanks!

cmegias, 2022-12-09:
I have tested it on Nexus K3P-S card and the NIC function remains working. And the output shown when running dmesg after loading the extension module seems to be fine.

multi-stream, 2023-01-03:
HI Alex! This works with the AU50 on a Intel Xeon machine. However, when I moved the card to an ARM based platform the tx dma is not working. I can see the ping from peer and ARP reply packet is created, but the tx dma seems to be stuck and no follow up tx dma completion interrupt. Could you shed some light where we should look? 
BTW, this goes beyond my original question. I can open a new issue if that works better. Thanks!

lizajoseph, 2025-11-07:
> Make a copy of the makefile and config.tcl for one of the design variants for your board, then make the appropriate changes.
>
> Here is an example: [...]
# Application block configuration
> dict set params APP_ID "32'h12348001"
> dict set params APP_ENABLE "1"
> dict set params APP_CTRL_ENABLE "1"
> dict set params APP_DMA_ENABLE "1"
> dict set params APP_AXIS_DIRECT_ENABLE "1"
> dict set params [...]

> Thanks for help Alex! This builds the overall design correctly for AU50. However, the mqnic_app_dma_bench.ko is crashing the kernel. The NIC function also stopped working. Is this expected? Thanks! [...]

Hi Alex,
I have been trying to enable the mqnic_app_dma_bench application for AU50, I had configured the config.tcl file as follows and similarly in the driver code. I has used this repo which you have mentioned in this issue #128
https://github.com/ucsdsysnet/corundum
> # Application block configuration
> dict set params APP_ID "32'h12348101"
> dict set params APP_ENABLE "1"
> dict set params APP_CTRL_ENABLE "1"
> dict set params APP_DMA_ENABLE "1"
> dict set params APP_AXIS_DIRECT_ENABLE "1"
> dict set params APP_AXIS_SYNC_ENABLE "1"
> dict set params APP_AXIS_IF_ENABLE "1"
> dict set params APP_STAT_ENABLE "1"
However I get below error , DMA bench register block not found. Could you let me know what am I missing here. The simulation testcase in this path ucsdsysnet/corundum/fpga/app/dma_bench/tb/mqnic_core_pcie_us works fine with test PASS 
for this application.

[image]

alexforencich, 2025-11-07:
Looks like all of the statistics counters read as zero, and that resulted in a divide-by-zero triggering a panic.  Not sure why they would all read as zero though.  The NIC functionality should still work, although that panic could have caused some problems in the kernel.

lizajoseph, 2025-11-07:
> Looks like all of the statistics counters read as zero, and that resulted in a divide-by-zero triggering a panic. Not sure why they would all read as zero though. The NIC functionality should still work, although that panic could have caused some problems in the kernel.

Hi Alex,
What are the statistics counters that you are referring to here ?