corundum/corundum #131: How to convert one U250 QSFP28 port to four 10g port?

xz394839, 2022-12-19:
Hi,

I am trying to run Corundum using Alveo U250.  
I generate the bit file by running the makefile in the folder ''corundum-master/fpga/mqnic/AU250/fpga_25g/fpga_10g''.
It seems that I can see two interfaces in my host after installing mqnic kernel module. 
Then I do a simple test to check the HW works well:
1.connect these two port together via a QSFP28 cable
2.tcpreplay some packets at one port
3.tcpdump at the other port
4.it works well. 

my question is, how to convert one U250 QSFP28 port to four 10g port?  I want to connect one QSFP28 port to four 10g port via a  QSFP28-4x10g SFP+ DAC cable, as shown in the picture below.
[image]

Thank you for your great work. It's a amazing.

alexforencich, 2022-12-19:
Not sure exactly what you're asking.  You should be able to simply plug in a QSFP-to-4xSFP cable like you have drawn and it should work.  Note that in the default configuration of the U250 10G/25G designs, only the first channel on each QSFP is actually connected to the datapath, you'll have to change the number of interfaces and/or the number of ports per interface so that all of the ports are usable.  Or if you only need two ports, but want to place both ports on one QSFP, you can set the PORT_MASK parameter in config.tcl to override the default behavior (first it tries to use lane 0 of each QSFP for each port, if that fails then it tries to put one interface on each QSFP, if that fails then it packs everything sequentially with no empty lanes).  Setting PORT_MASK to 255 would result in the ports being completely packed with no empty lanes in between.

xz394839, 2022-12-19:
Yes,  I once simply plugged one QSFP-to-4xSFP cable and test the connectivity but failed. I will change the number of ports per interface to 4 and try again.
Thank you for your prompt reply.

xz394839, 2022-12-20:
> Not sure exactly what you're asking. You should be able to simply plug in a QSFP-to-4xSFP cable like you have drawn and it should work. Note that in the default configuration of the U250 10G/25G [...]

Hello Alexforencich, I need two or four ports and want to place all ports on one QSFP. 
So I change the config.tcl file(IF_COUNT=1,PORTS_PER_IF=4,PORT_MASK=255) according to your suggestion and regenerate the bitfile. but I can see only one interface in my host( I retest its connectivity and of course, it works, thanks for your help).

Should there be four interface in my host? Do I miss any details? What should I do to get four interface?
Thank you in advance.

alexforencich, 2022-12-20:
In your case, I recommend setting IF_COUNT = 2 or 4 and PORTS_PER_IF = 1, along with PORT_MASK = 255.  Setting PORTS_PER_IF to something other than 1 is not currently very well supported (you have to tinker around with the schedulers or do something more custom to get TX traffic to use all of the ports, and it all appears as one interface to the OS...which, while useful for certain applications, is likely not what you want).

xz394839, 2022-12-20:
I encounter this error in synthesis stage if I set IF_COUNT = 4, PORTS_PER_IF = 1, and PORT_MASK = 255.
It seems like that the configuration causes the address overlap
Could you please help me check what the problem is? 
thank you.
------------------------------------------------------------------------------------------------------------------------

ERROR: [Synth 8-6058] Synth Error: Error: address ranges overlap (instance )  [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:192]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:52]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:58]
ERROR: [Synth 8-6156] failed synthesizing module 'axil_crossbar_addr__parameterized3' [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_addr.v:34]
ERROR: [Synth 8-196] conditional expression could not be resolved to a constant [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:165]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:48]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:57]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:60]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:63]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:66]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:69]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:72]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:75]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:78]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:81]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:84]
ERROR: [Synth 8-6156] failed synthesizing module 'axil_crossbar_wr__parameterized1' [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar_wr.v:34]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:48]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:57]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:60]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:63]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:66]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:69]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:72]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:75]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:78]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:81]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:84]
WARNING: [Synth 8-693] zero replication count - replication ignored [/data/home/xianwei/bak/corundum/corundum/fpga/mqnic/AU250/fpga_25g/lib/axi/rtl/axil_crossbar.v:87]
[... cut ...]
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/data/tools/Xilinx/Vivado/2022.1/data/ip'.
open_project: Time (s): cpu = 00:00:08 ; elapsed = 00:00:09 . Memory (MB): peak = 3253.621 ; gain = 2.020 ; free physical = 577575 ; free virtual = 729720
# reset_run impl_1
# launch_runs -jobs 4 impl_1
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Tue Dec 20 10:22:20 2022...

alexforencich, 2022-12-20:
Right, in this case you'll either need to reduce TX_QUEUE_INDEX_WIDTH (to reduce the number of transmit queues, and hence the amount of address space taken up by each interface) or increase AXIL_CTRL_ADDR_WIDTH (to increase the size of the BAR so all of the interfaces will fit).

closed 2023-02-03 by xz394839