corundum/corundum #132: Support for Alveo U55C

JuneJXS, 2022-12-21:
Hi,

Currently I have a U55C board on hand, and wanna test it out. I know U55C is similar to U50 but with one additional QSFP and HBM bank (plus more resources), so should I use the U50 program as a starting point? And what parts of configurations should I pay attention to for tuning for U55C?

Thanks!

alexforencich, 2022-12-21:
U55C = U280 with double the HBM2 and no DDR4; U55N/C1100 = U50 with double the HBM2, an additional QSFP28, and USB JTAG.  The parts even share JTAG IDs.

So use the U280 design as a starting point, and strip out all of the DDR4 stuff (use the U50 design as a reference for how the ports are tied off in the mqnic_core_pcie instance).  Not sure what else needs to be done aside from maybe some clocking changes.  Naturally IO and placement constraints will have to be updated.  I think it should be a straightforward port, though.  I will definitely add a port for the U55C upstream, once I get my hands on a board so I can verify that everything works going forward.

JuneJXS, 2022-12-21:
> 

Thanks for the quick response. Let me look into it.

vaniaprkl, 2023-07-27:
Hi, I'm also interested in getting corundum running on U55C. 
Is there a fork for it or expected release soon? Asking before I attempt this  port :)

Thanks in advance.

alexforencich, 2023-07-27:
Unfortunately, I don't have a U55C myself, so there won't be upstream support for the U55C until I get my hands on one as this is necessary for long-term maintenance

myk12, 2024-06-26:
> > 
> 
> Thanks for the quick response. Let me look into it.

Hi, have you successfully done it?