corundum/corundum #135: Adjust PHC time with offset register does not work

nightseas, 2023-02-05:
Hi,

I was trying the linuxptp `phc_ctl -- adj` function and found the adjustment of PHC with less than or equal to 1 second didn't make any change to PHC at all.

After debugging with simulation, it shows that $clog2(2**32) will return 0, which leads to OFFSET_NS_WIDTH being ignored and then an incorrect INC_NS_WIDTH.
https://github.com/corundum/corundum/blob/c708bc45cd0d5b44274d238dd9fadd16335f337d/fpga/lib/eth/rtl/ptp_clock.v#L97

Changing OFFSET_NS_WIDTH to 31 here works.
https://github.com/corundum/corundum/blob/c708bc45cd0d5b44274d238dd9fadd16335f337d/fpga/common/rtl/mqnic_ptp_clock.v#L100

BR,
Xiaohai Li

alexforencich, 2023-02-05:
Hmm, good catch.  I guess that code path hasn't really been tested as ptp4l can just slew out relatively small offsets, but naturally that doesn't work if you're trying to do something like white rabbit where you can't slew the clock.  I think using `{X{1'b1}}` instead of `{2**X}` might work correctly, though.

alexforencich, 2023-02-05:
And actually I should add a check for that, none of these can be larger than 30 bits, otherwise the ns overflow will not work correctly.  So I guess that means the `PTP_OFFSET_NS_WIDTH` parameter in `mqnic_ptp_clock` also needs to be fixed.