corundum/corundum #136: Internal error: Cannot run timing on a non-timing device

magiczixiao, 2023-02-20:
We are running `make` under `..../corundum/fpga/mqnic/AU50/fpga_100g`, the board package for au50 is installed and our vivado version is 2022.2, whith SW Build 3671981 on Fri Oct 14 04:59:54 MDT 2022, IP Build 3669848 on Fri Oct 14 08:30:02 MDT 2022;
Then we have the following error when building:
```
ERROR: [Common 17-577] Internal error: Cannot run timing on a non-timing device [/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/lib/axi/syn/vivado/axil_cdc.tcl:23]
Finished Sourcing Tcl File [/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/lib/axi/syn/vivado/axil_cdc.tcl]
Sourcing Tcl File [/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/lib/axis/syn/vivado/axis_async_fifo.tcl]
Inserting timing constraints for axis_async_fifo instance core_inst/core_inst/core_pcie_inst/core_inst/iface[0].interface_inst/port[0].port_inst/port_rx_inst/rx_async_fifo_inst/fifo_inst
ERROR: [Common 17-577] Internal error: Cannot run timing on a non-timing device [/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/lib/axis/syn/vivado/axis_async_fifo.tcl:23]
Finished Sourcing Tcl File [/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/lib/axis/syn/vivado/axis_async_fifo.tcl]
Sourcing Tcl File [/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/lib/axis/syn/vivado/sync_reset.tcl]
....
ERROR: [Common 17-577] Internal error: Cannot run timing on a non-timing device
INFO: [Common 17-206] Exiting Vivado at Mon Feb 20 11:40:54 2023...
[Mon Feb 20 11:40:55 2023] impl_1 finished
WARNING: [Vivado 12-13638] Failed runs(s) : 'impl_1'
wait_on_runs: Time (s): cpu = 00:00:02 ; elapsed = 00:01:50 . Memory (MB): peak = 1752.785 ; gain = 0.000 ; free physical = 463173 ; free virtual = 511542
# open_run impl_1
ERROR: [Common 17-69] Command failed: Run 'impl_1' failed. Unable to open
INFO: [Common 17-206] Exiting Vivado at Mon Feb 20 11:40:55 2023...
../common/vivado.mk:114: recipe for target 'fpga.runs/impl_1/fpga_routed.dcp' failed
make[1]: *** [fpga.runs/impl_1/fpga_routed.dcp] Error 1
make[1]: Leaving directory '/workspace/zxwang/corundum/fpga/mqnic/AU50/fpga_100g/fpga'
Makefile:14: recipe for target 'fpga' failed
make: *** [fpga] Error 2
```
Any solutions? Thanks!

alexforencich, 2023-02-20:
Never seen that one before.  "Internal error" makes me think that this is some sort of a Vivado problem.  Not sure if it's some sort of installation or license related problem, or if it's a deeper problem with the 2022.2 release of Vivado.  My suggestion is to try reinstalling Vivado completely, and make sure you install device support for Virtex UltraScale+ (unless there is a separate option for Alveo; I don't remember exactly what all of the options are during installation).  If that doesn't work, then try an older version of Vivado like 2022.1 or 2021.2.

magiczixiao, 2023-02-24:
Thank you for your prompt reply. We have done some tests:
This problem will reappear on ubuntu18.04+Vivado2022.2 standard version, and cannot be solved by reinstalling Vivado.
We have installed Vivado2022.2 standard version on another Centos server, and can successfully generate fpga. bit.
We can also compile successfully using ubuntu18.04+vivado2022.2 Enterprise Edition.
Maybe, as you said, this is a bug of Vivado itself.