corundum/corundum #140: IF_COUNT in NETfpga SUME

bizhaan47, 2023-03-08:
in NETfpga SUME, IF_COUNT is set to 1
also PORT_COUNT is set to 1
so, this board works with 10g ethernet
but this board has 4 sfp+ ports and must be 40g
my board has 4 sfp+ ports; can i change IF_COUNT to 4? after that, my NIC card will be a 40g ethernet card?

alexforencich, 2023-03-08:
Virtex 7 is rather long in the tooth from a timing perspective, so that's currently set to 1 interface and 1 port per interface in the interests of timing closure.  Sure, you can crank it up to 4 if you want, but you may run in to problems with timing closure.

However, 4 ports of 10G does not give you a 40G link, if you need a proper 40G link then you would need a 40G MAC and you would configure it to operate with 1 interface.  Unfortunately, I'm not aware of any open-source 40G MAC modules, so you might have to get a license for the Xilinx 40G MAC if you need to go that route.  But, if you only need 40 Gbps of aggregate bandwidth across four 10G links, then yes, that should work, assuming the design closes timing.

bizhaan47, 2023-03-09:
Thanks.
What are problems with timing closure? can you make examples?
Also, why 4 aggregated 10g links are not a 40g mac?
Is it needed to change 10g MAC module to 40g Mac?

alexforencich, 2023-03-09:
The components inside the FPGA (LUTs, routing wires, etc.) all have some associated propagation delay.  When a digital logic design is implemented on a chip (FPGA or otherwise), you have to make sure that it will actually run correctly at the required clock frequency - in other words, all possible paths have a propagation delay less than the clock period.  The clock period is specified via timing constraints, and these constraints are checked via static timing analysis.  The term "timing closure" relates to all of these constraints being met - achieving "timing closure" means that all of the constraints are met and the design will operate correctly.  If there are timing failures, then the design will not operate correctly (and in the case of a PCIe device like Corundum, this could result in crashing the host system, etc.).

More logic in a design means that there is going to be more potential for suboptimal placement, routing congestion, etc. which make it much harder for the tools to meet the timing constraints.  Manually adding placement constraints, adding/adjusting pipelining settings and such can help, but Virtex 7 is quite old so it's probably a better idea just to get a newer FPGA with faster logic so you don't waste a lot of time just trying to get Vivado to give you a design that has a possibility of working.

TL;DR: if you turn on all four ports, it might be hard to get Vivado to build a design that actually works in the hardware, at least not without making significant modifications.

Four 10GBASE-R links are not equivalent to the four 10 Gbps lanes that make up one 40GBASE-R link.  Whether you need a single 40G link (and hence need a 40G MAC) or four 10G links (ahd hence need a 10G MAC) is going to be dependent on your application.

bizhaan47, 2023-03-09:
Thanks.
But i have a question:
While i enable 4 ports, these ports are parallel and work together in parallel. so why need timing closure modification? critical path don't changed and possibly timing don't changed.

alexforencich, 2023-03-09:
Each port has logic, each interface has a LOT of associated logic.  And running at 250 MHz on Virtex 7 is not so easy, the tools have a hard enough time getting timing closure with 1 interface enabled.  Sure, if you ignore placement and routing congestion, the critical path does not change when enabling more interfaces.  But things like BRAMs are only located in specific places and as a result the tools can be forced to place things in sub-optimal ways.  Sure, you could probably make it work if you spent a few hours playing around with manual placement constraints, isolating each interface with its own pblock or whatever.  Or you could get a more modern part with a faster fabric that requires less tool hand-holding.  

Anyway, I'm just trying to provide a fair warning based on my experience of working with Corundum: if you turn on all four interfaces on a Virtex 7, you'll likely need to spend a lot of time baby-sitting the tools.  Doable?  Sure.  Worth the time?  Up to you to decide; personally I would get something more modern.

bizhaan47, 2023-03-09:
I got it.
Thanks.

daviderror, 2025-06-09:
@bizhaan47 
hello! have you solved the problem with the 4 ports on the NetFPGA board?