corundum/corundum #143: `pcie_user_reset` BUFG replication issues with VU9P (VCU1525)

KireinaHoro, 2023-04-04:
I'm working on integrating a rather big [in-network-compute cluster](https://github.com/spcl/pspin) as an application block into Corundum.  I am using Vivado 2020.2.  So far I've been having issues with congestion on SLR1, where the PCIe block is.  Timing results are quite sporadic and Vivado struggles to meet timing.

One case of timing failures I've recently seen is that the `pcie_user_reset` signal:

https://github.com/corundum/corundum/blob/c708bc45cd0d5b44274d238dd9fadd16335f337d/fpga/mqnic/VCU1525/fpga_100g/rtl/fpga.v#L804-L808

gets replicated into SLR0 and 2, but one of the replicas misses the BUFG for some reason.  The post-implementation schematics confirm this:

[image]

Note how `pcie_user_reset_reg_2_reg/Q` drives other logic without the BUFG.  This net has high fan-out and some endpoints fail timing by a small margin (TNS=-0.150).

The question is that if we need some better constraints on the pipelining registers for the reset, so that the replication happens correctly.  I would be happy to provide more details.  Thanks in advance!

alexforencich, 2023-04-04:
Vivado definitely has issues with some of its heuristics regarding replication.  For instance, it seems like it only looks at the number of loads and not where those pins have actually been placed, resulting in large timing failures when one flip flop is connected to two elements that ended up in opposite corners of the device.  Replicating the flip flop would trivially solve the problem, but that doesn't happen.  The problem is particularly bad on multi-SLR devices.  Adding that explicit BUFG seemed to help quite a bit, but it's definitely not perfect.  

In this case, I'm wondering if Vivado doesn't have the timing margin considering the bufg delay - it might be able to drive an FF directly, but maybe it can't use the BUFG when driving a LUT.  I have also seen cases where it re-pipelines stuff and ends up connecting a bunch of stuff one FF before the bufg.  I have thought about possibly adding a flip flop after the bufg, but haven't tried it yet.  It's also possible that there are certain places where the reset line should just be run through a flip-flop locally.  For instance, maybe an extra FF on the reset line to the MMCM would help.  I think I have also seen some messages about timing constraints associated with the MIGs that could be causing issues with the reset lines, possibly adding flip flops on the MIG reset inputs would help with that so those constraints don't affect the upstream bufg.

KireinaHoro, 2023-04-05:
Thanks a lot for the input!  I tried adding DONT_TOUCH on the last FF before BUFG yesterday and timing closure only became worse.  I'm reverting that and enabling `Performance_ExplorePostRoutePhysOpt` to try again.

In the meantime, I also have a few ideas of experimenting to proceed; can you comment on these as well?

- Since `pcie_user_reset` is supposed to be synchronous after the pipeline (`_int` out of the PCIe IP block is already synchronous, not sure why we needed the pipeline) and also semi static, I can put in a `set_multicycle_path` or `set_false_path` to ease the timing constraint
- Manually replicate the last FF and BUFG into all SLRs, and place them manually with constraints.  I have to figure out where the reset consumers are and fix them to the respective SLR pblocks (e.g. the MIGs).

Thanks again for the fast reply!

alexforencich, 2023-04-05:
False path and multicycle constraints effectively make the signal asynchronous, so you would then have to resynchronize it at the destination.  Multicycle constraints should never be used for resets; they're intended to be used in conjunction with some explicit mechanism to control when the signal is sampled at the destination to ensure that the multicycle constraint is valid, usually in the form of some sort of enable signal.  Similarly, false paths need to be used in conjunction with synchronizers.