corundum/corundum #144: Automatic server restart issue when programming fpga.bit

magiczixiao, 2023-04-13:
When we use the JTAG programmer to download the fpga.bit file to the Alveo u50, the server will automatically restart. After restarting, the lspci will not be able to locate the board, and the blue light on the board will turn off. After a cold restart, it returns to normal.

Does automatic restart when downloading bit files trigger any checking mechanism on the server? Is there any way to prevent him from automatically restarting?

Here are the outputs of `lspci:`
```
root@ubuntu:~# lspci -vd 10ee:
44:00.0 Processing accelerators: Xilinx Corporation Device 5020
        Subsystem: Xilinx Corporation Device 000e
        Physical Slot: 2
        Flags: bus master, fast devsel, latency 0, NUMA node 2
        Memory at c2000000 (64-bit, prefetchable) [size=32M]
        Memory at c4040000 (64-bit, prefetchable) [size=256K]
        Capabilities: [40] Power Management version 3
        Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [1c0] #19
        Capabilities: [e00] Access Control Services
        Capabilities: [e10] #15
        Capabilities: [e80] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
        Kernel driver in use: xclmgmt
        Kernel modules: xclmgmt

44:00.1 Processing accelerators: Xilinx Corporation Device 5021
        Subsystem: Xilinx Corporation Device 000e
        Physical Slot: 2
        Flags: bus master, fast devsel, latency 0, IRQ 37, NUMA node 2
        Memory at c0000000 (64-bit, prefetchable) [size=32M]
        Memory at c4000000 (64-bit, prefetchable) [size=256K]
        Memory at b0000000 (64-bit, prefetchable) [size=256M]
        Capabilities: [40] Power Management version 3
        Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [e00] Access Control Services
        Capabilities: [e10] #15
        Capabilities: [e80] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
        Kernel driver in use: xocl
        Kernel modules: xocl
```

We have tried 

```
root@ubuntu:/workspace/zxwang/corundum# sh ./fpga/lib/pcie/scripts/pcie_disable_fatal_err.sh 44:00.1
Disabling fatal error reporting on port 0000:40:03.0...
Command: 0547
Command: 0447
Device control: 012f
Device control: 012b
root@ubuntu:/workspace/zxwang/corundum# sh ./fpga/lib/pcie/scripts/pcie_disable_fatal_err.sh 44:00.0
Disabling fatal error reporting on port 0000:40:03.0...
Command: 0447
Command: 0447
Device control: 012b
Device control: 012b
```
but it cannot solve this problem.

Thanks!

alexforencich, 2023-04-13:
This is going to be entirely dependent on the server and exactly how the server's internal monitoring subsystem is set up.  Most likely something is detecting the PCIe link going down while the FPGA is being reset during bitstream loading and triggering a server reboot.  From what I have seen, this usually seems to be done via PCIe error reporting, and at some point the error gets reported to the management subsystem somehow.  If fatal error reporting is turned off on the PCIe root port, then the management subsystem won't detect the error.  But it's possible that on your server there is something different about the configuration so simply flipping those bits isn't sufficient.

What server specifically are you using, and what does the iDRAC/iLO/IPMI/etc. management interface report after the reboot?

RoyTien, 2023-04-14:
Thank you for your answer, it is  very helpful. We installed `ipmitool` in our server printed out **log 561** by running `ipmitool sel list`.
[image]

For now, we are working on turning off PCIe error reporting as you pointed out. However, we are not sure about the server **log 561** IS the PCIe error report you mentioned before. Is there any method to confirm that **log 561** is caused by FPGA programming？

Thank you so much.

RoyTien, 2023-04-14:
Thank for you reply！

`dmidecode -t1` returns:
Getting SMBIOS data from sysfs.
SMBIOS 3.0 present.
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: ZTE
	Product Name: R8500G3
	Version: 1.0.0
	Wake-up Type: Power Switch
	SKU Number: N/A
	Family: Server
`cat /proc/version` returns:
Linux version 4.15.0-112-generic (buildd@lcy01-amd64-027) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04))

Thank you again.

alexforencich, 2023-04-14:
ZTE R8500G3?  Unfortunately, that's not one that I am familiar with.  Well, it's possible you'll need to talk to the server manufacturer to ask about how to disable the automatic reboot, since that's going to be mediated by the system's firmware and management subsystems.  I'm not sure exactly what the firmware is looking at to detect that there is some issue associated with the PCIe bus.  On all of the systems that I have personally worked with, either the firmware does not reboot the server when a PCIe device falls off the bus, or flipping the couple of bits in the root port to disable error reporting is sufficient to prevent the reboot.  It seems like the firmware on that machine must be doing something else - possibly those error reporting bits are getting flipped back by something, or perhaps there is some other error reporting mechanism that the firmware is looking at.