corundum/corundum #150: After insmod mqnic.ko    execute lsmod or cat /proc/modules oops

zqchinaone, 2023-05-30:
linux 4.19  
 corundum ver
[image]

[  188.303188] Unable to handle kernel paging request at virtual address 00400bf39587062b
[  188.311099] Mem abort info:
[  188.313891]   ESR = 0x96000004
[  188.316938]   Exception class = DABT (current EL), IL = 32 bits
[  188.322836]   SET = 0, FnV = 0
[  188.325884]   EA = 0, S1PTW = 0
[  188.329016] Data abort info:
[  188.331886]   ISV = 0, ISS = 0x00000004
[  188.335707]   CM = 0, WnR = 0
[  188.338658] [00400bf39587062b] address between user and kernel address ranges
[  188.345769] Internal error: Oops: 96000004 [#1] SMP
[  188.350623] Modules linked in: at24 i2c_mux_pca954x i2c_mux mqnic(OE) scsi_transport_iscsi rfkill sunrpc ngbe(OE) uio_pdrv_genirq uio vfat fat sch_fq_codel ip_tables rtc_ds1307 ahci libahci libata i2c_designware_platform i2c_designware_core ast dm_mirror dm_region_hash dm_log
[  188.374932] Process lsmod (pid: 2081, stack limit = 0x0000000080c2e68e)
[  188.381515] CPU: 45 PID: 2081 Comm: lsmod Tainted: G           OE     4.19.90-2106.3.0.0095.kb13.dmi.mem.ky3_4.aarch64 #1
[  188.392416] Hardware name: FT2000plus Generic Board & Memsize 64G (DT)
[  188.398913] pstate: a0000005 (NzCv daif -PAN -UAO)
[  188.403700] pc : string+0x2c/0xa8
[  188.406998] lr : vsnprintf+0x5f0/0x748
[  188.410727] sp : ffff000014dcfa40
[  188.414024] x29: ffff000014dcfa40 x28: ffff000008ee821a
[  188.419309] x27: ffff000008ee821a x26: ffff8480cae200a4
[  188.424593] x25: 00000000ffffffd8 x24: 0000000000000020
[  188.429878] x23: 000000000000ff5c x22: ffff000009533000
[  188.435162] x21: ffff000008d40ce0 x20: ffff8480cae30000
[  188.440446] x19: ffff8480cae200a4 x18: ffffffffffffffff
[  188.445729] x17: 0000000000000000 x16: 0000000000000000
[  188.451012] x15: ffff000009533708 x14: ffff8480cae30000
[  188.456297] x13: ffff8480cae200a3 x12: 0000000000000008
[  188.461581] x11: 0000000005f5e0ff x10: 7f7f7f7f7f7f7f7f
[  188.466865] x9 : fefefefefefeff09 x8 : fffffffffffffffe
[  188.472149] x7 : ffff8480cae200a4 x6 : f9400bf39587062b
[  188.477433] x5 : 000000000000ffff x4 : ffffffffffffffff
[  188.482717] x3 : ffff0a00ffffff04 x2 : ffff8480cae30000
[  188.488001] x1 : ffff8480cae30000 x0 : ffff8480cae200a4
[  188.493285] Call trace:
[  188.495718]  string+0x2c/0xa8
[  188.498669]  vsnprintf+0x5f0/0x748
[  188.502056]  seq_vprintf+0x70/0x98
[  188.505438]  seq_printf+0x7c/0xa0
[  188.508737]  m_show+0xb8/0x238
[  188.511775]  seq_read+0x2cc/0x4c8
[  188.515075]  proc_reg_read+0x80/0xd8
[  188.518632]  __vfs_read+0x60/0x1a0
[  188.522014]  vfs_read+0x94/0x150
[  188.525224]  ksys_read+0x74/0xf0
[  188.528434]  __arm64_sys_read+0x24/0x30
[  188.532251]  el0_svc_common+0x78/0x178
[  188.535981]  el0_svc_handler+0x38/0x78
[  188.539710]  el0_svc+0x8/0x1f8
[  188.542749] Code: d1000488 eb0500df 540002e9 b4000344 (394000c5)
[  188.548846] ---[ end trace 210dfb35f54d5770 ]---
[  188.553440] Kernel panic - not syncing: Fatal exception
[  188.558665] SMP: stopping secondary CPUs
[  188.562579] Kernel Offset: disabled
[  188.566050] CPU features: 0x10,a0800008
[  188.569864] Memory Limit: none
[  188.572930] ---[ end Kernel panic - not syncing: Fatal exception ]--

alexforencich, 2023-05-30:
Interesting, I'll take a look at that.  Can you post the output of `uname -a`, please?  Kernel version alone might not be sufficient.  Also it looks like you might be on arm64 - I don't have any ARM systems to test on, so hopefully whatever is going on is not specific to ARM.

zqchinaone, 2023-05-30:
Linux localhost.localdomain 4.19.90-2106.3.0.0095.kb13.dmi.mem.ky3_4.aarch64 #1 SMP Tue Sep 13 11:21:02 CST 2022 aarch64 aarch64 aarch64 GNU/Linux

alexforencich, 2023-05-30:
Looks like you're on commit 03a49d7bc69c7427de118c127d66215c22753bf4?  That's from July 2022, almost a year old at this point.  Have you tried the latest version to see if the issue is present there?

I tested on my local machine, both a recent commit and 03a49d7bc69c7427de118c127d66215c22753bf4, but I have not been able to reproduce the issue.  But I am running a much more recent kernel version (5.15.53-1-lts).  What linux distribution are you using?  If this is just a kernel module issue that doesn't involve the hardware, maybe it would be possible to replicate in a virtual machine.

zqchinaone, 2023-05-30:
think you ！Wait until I have a chance to try it。
[image]

I have modified  arch/arm64/include/asm/stat.h 
Because 
./arch/arm64/include/asm/stat.h:30:8: err：‘struct stat64’ Redefine
        ^~~~~~
In file included from ./arch/arm64/include/uapi/asm/stat.h:17:0,
                 from ./arch/arm64/include/asm/stat.h:19,
                 from ./include/linux/stat.h:6,
                 from ./include/linux/sysfs.h:22,
                 from ./include/linux/kobject.h:21,
                 from ./include/linux/pci.h:30,
                 from /root/zq/mqnic/mqnic.h:41,
                 from /root/zq/mqnic/mqnic_main.c:36:
I don't know if there is an impact。