corundum/corundum #152: No interfaces on Debian

vaniaprkl, 2023-06-16:
I am seeing a very strange issue running on Debian 
> root@trinity:/disk2/jun16th/corundum/utils# uname -r
6.1.0-9-amd64

Without the driver loaded there is return on running util commands for example
> root@trinity:/disk2/jun16th/corundum/utils# lspci -kb -s 04:00.0
04:00.0 Ethernet controller: Device 1234:1001
	Subsystem: Xilinx Corporation Device 9032
root@trinity:/disk2/jun16th/corundum/utils# ./mqnic-fw -d 04:00.0
Error: Event queue block not found
Failed to create interface 0, skipping
Error: Event queue block not found
Failed to create interface 1, skipping
Error: Event queue block not found
Failed to create interface 2, skipping
Error: Event queue block not found
Failed to create interface 3, skipping
PCIe ID (device): 0000:04:00.0
PCIe ID (upstream port): 0000:00:03.0
FPGA ID: 0x04b77093
FPGA part: XCU50
FW ID: 0x00000000
FW version: 0.0.1.0
Board ID: 0x10ee9032
Board version: 1.0.0.0
Build date: 2023-02-24 18:00:35 UTC (raw 0x63f8fb43)
Git hash: c708bc45
Release info: 00000000
Application ID: 0x12340001
Flash type: SPI
Flash format: 0x01002012
Data width: 4
Manufacturer ID: 0x20
Memory type: 0xbb
Memory capacity: 0x21
Flash size: 128 MB
Write buffer size: 256 B
Erase block size: 4096 B
Flash segment 0: start 0x00000000 length 0x01002000
Flash segment 1: start 0x01002000 length 0x06ffe000
Default segment: 1
Fallback segment: 0
Selected: segment 1 start 0x01002000 length 0x06ffe000

Loading the driver seem to break util run fully
>root@trinity:/disk2/jun16th/corundum/utils# insmod ../modules/mqnic/mqnic.ko
root@trinity:/disk2/jun16th/corundum/utils# lspci -kb -s 04:00.0
04:00.0 Ethernet controller: Device 1234:1001
	Subsystem: Xilinx Corporation Device 9032
	Kernel driver in use: **mqnic**
root@trinity:/disk2/jun16th/corundum/utils# ./mqnic-fw -d 04:00.0
mmap regs failed: Invalid argument
Failed to open device
root@trinity:/disk2/jun16th/corundum/utils#

I just cant get the interfaces up and would like to get it working on this environment. What am I doing wrong?

alexforencich, 2023-06-16:
Only one thing can talk directly to the device at any given time, so once you load the driver you then have to point the userspace tools at `/dev/mqnicN` or specify the ethernet interface, instead of the PCIe ID.  Although, hmm, maybe I can adjust the userspace tools to transparently figure that one out.  

But, it looks like your main problem is a version conflict between the bitstream on the FPGA and the driver/userspace tools.  Both should be built from the same git revision.

alexforencich, 2023-06-16:
Userspace tools are now updated in my development repo to try the miscdev first when specifying a PCIe BDF, this way it should work even when the driver is loaded: https://github.com/ucsdsysnet/corundum/commit/fde1f5d341c30d307d9e28003b1c69098a5ecfca

And in your case, it looks like the FPGA config was built a while ago (`Build date: 2023-02-24 18:00:35 UTC (raw 0x63f8fb43)`), so if you have a fresh copy of the repo on your debian box, that would probably explain why it isn't working.  Simple solution is to roll back that clone to the same commit.

vaniaprkl, 2023-06-20:
New driver built from the development repo worked against fw built from the same repo. Thanks.

closed 2023-06-20 by vaniaprkl