corundum/corundum #156: `ethtool enp94s0` doesn't show the information about speed and so on.

zhangchenqi123, 2023-07-11:
I have programed the device, insmod the kernel. but when I check the Corundum NIC with ethtool, it doesn't show many information like the video in youtube.
I thought it should be:
[image]

but on my server, it is:
[image]

---
Here's the dmesg information:

[image]
[image]
[image]
[image]

What could be the reason and how can I fix it ?

alexforencich, 2023-07-11:
It's not yet implemented.  Eventually, there will be a management soft core, which the driver will query to get this sort of information in a target-independent manner.  Until then, only link up/down status is reported.

zhangchenqi123, 2023-07-11:
Should I make some modifications to the source code？

alexforencich, 2023-07-11:
Modify the source code to do what?  Implement the management soft core?

zhangchenqi123, 2023-07-11:
I saw your video in the youtube made a lot of modifications. 
[The document ](https://www.youtube.com/watch?v=MMGCctTQD7k&list=PLV-OBCE-6KmB0mQvjWgagQGXsuvyLBxxd&index=2)
Anyway, what should I do next to implement?

alexforencich, 2023-07-12:
The question is, what do you want to accomplish?  Just hack some stuff together so ethtool reports some link information?  Or would you be interested in helping to implement all of the pieces necessary to do this properly?  (Management soft core, mailbox for driver communication, probably some additional control registers, etc.)