corundum/corundum #157: How to enable flow control?

vaniaprkl, 2023-07-17:
Hi Alex,

I have been going through the source code trying to enable Flow control. I am starting to understand the concept of PCIe "credits" that both the card and host track. This credit tracking is already set in your top level by 
```
pcie_cq_np_req(1'b1),
```

Upon further deep diving into your code/document, noticed these that seems quiet relevant:
```
dma_if_pcie_us.v:    parameter READ_TX_FC_ENABLE = 0,
dma_if_pcie_us.v:    parameter WRITE_TX_FC_ENABLE = 0
pcie_us_axi_dma.v:    parameter READ_TX_FC_ENABLE = 0,
pcie_us_axi_dma.v:    parameter WRITE_TX_FC_ENABLE = 0
```
These modules don't seem to be instantiated in any upper level code. What am I missing?

Thanks in advance!! Really loving the pipeline you have created here.

alexforencich, 2023-07-18:
Which board are you using?  And can you be more specific about what it is you're trying to do?

vaniaprkl, 2023-07-18:
Hi Alex, I am currently targeting the Alveo U50 board. At the moment, I am just counting the number of packets going through to test the resilience of the pipeline. I am noticing my counters to be predictable and exactly equal to the number of pkts that were sent, even at 10Gbps line rate, which is excellent. But the host side interface counters, got through running ifconfig, does not reflect the hardware counters even at 1Gbps. Trying to figure out where the pkts are ending up and if it is due to flow control not being enabled. 

Thanks for the prompt response.

alexforencich, 2023-07-18:
Interesting.  I would need to hear more about your specific setup to try to debug what might be going on.  But, PCIe is a lossless protocol, flow control is part of the spec and cannot be disabled.  This would be handled by the PCIe hard IP core on the U50.  The read/write FC stuff in the DMA engine itself is to prevent head-of-line blocking between read and write requests, but I think this specific functionality has been moved into pcie_us_if, which then applies back-pressure to the DMA engine.  If you're seeing dropped packets on the RX path, then these would be dropped inside the FPGA before they are transferred via PCIe, it's not possible for packets to be dropped on the PCIe side.  However, performance issues on the PCIe and software end can cause things to back up in the datapath, and packets are dropped either in one of the RX FIFOs (when the FIFO is full) or by the RX engine during the receive process (in the case of an issue reading a receive descriptor for the packet in question).

vaniaprkl, 2023-07-18:
Hi Alex, in an attempt to corner down where the packet drops were actually happening, I added debug logic: 
- to count ```tlast ``` at the output of the MAC 
- hardware counters in ```mqnic_app```

The above counts **always matched** the number of pkts that were sent. Host side pkts were counted from running ```ifconfig``` on the interface but showed fewer pkts at datarates nearing 1Gbps. 

Next I checked to see if the signal ```in_tlp_ready``` from module ```pcie_tlp_fifo_raw.v```, propagates correctly to module  ```mqnic_app_block.v``` where I count the packets. The signal ```s_axis_write_data_tready``` that ```mqnic_app_block.v``` gets from module ```dma_client_axis_sink``` never seems to be pulsed low even when drops are seemingly happening. 

Just to make sure ```s_axis_write_data_tready``` is a functional back pressure I added debug logic to force the signal ```ram_wr_cmd_ready``` in module ```dma_client_axis_sink``` low and did notice ```s_axis_write_data_tready``` pulse low as expected. So back pressure works from the PCIe as far as I can tell. 

I did not count pkts out of the RX FIFOs since they seem like TLP pkts, which I am still becoming familiar with.
Is there a easy way to count the number of pkts leaving the RX FIFOs?

alexforencich, 2023-07-19:
Are you looking at just `tlast`, or `tvalid && tready && tlast`?  The `tlast` signal is not necessarily going to be high for only one cycle.  

On the PCIe TLP side, you're not going to be able to directly compare TLP counts to Ethernet packet counts for several reasons - there is descriptor and completion traffic in addition to the actual packet data, and packet data can be split across several TLPs.  And if packets are dropped due to disabled queues or insufficient descriptors, it will still be transferred into the internal scratchpad RAM as the descriptor read takes place after the packet has been fully stored into the scratchpad RAM, so the tready line should not get deasserted in this case.

alexforencich, 2023-07-19:
The scratchpad RAM is used to store the packet data temporarily while the DMA engine transfers it to/from host memory.  There are also separate scratchpad RAMs for descriptors and completions.  

The counters currently reported by the operating system should be coming from the driver itself, whenever the driver handles a packet.  So if a packet is dropped in the NIC, then it won't be reflected in the counters.

Eventually, there will be a bunch of statistics counters related to the descriptor handling and transmit/receive engines, but there is a planned rewrite of the descriptor handling logic so those will be implemented with the rewrite. For now, what you can do is look at the descriptor request status signals here: https://github.com/corundum/corundum/blob/master/fpga/common/rtl/rx_engine.v#L167-L173 .  `empty` and `error` both indicate a failed descriptor read that will result in a dropped packet.  

It doesn't really matter if backpressure is exerted or not.  If software can't keep up with the incoming packets, either you've got to explicitly drop them when the descriptor reads fail, or everything will back up and packets will get dropped in the RX FIFO.  Eventually I suppose this could be made into a config option once PFC/pause support is implemented, as this would prevent the RX FIFOs from overflowing.  Although this may also require changes to the RX engine so that packets can pass each other, but even that won't be a perfect solution as once the scratchpad RAM fills up with packets waiting for descriptors, everything will get blocked.

FC signals: https://github.com/corundum/corundum/blob/master/fpga/lib/pcie/rtl/pcie_us_if_rq.v#L115-L121

vaniaprkl, 2023-07-31:
Hi Alex, 

Thanks for that excellent pointer to ```empty``` and ```error``` signals. I redid my build counting on them and the lost packets were exactly equal to the number of times ```empty``` had pulsed. So this way, I could confirm my pkts were dropped right on the card side and that ambiguity was laid to rest. 

Going by your suggestion, on a PFC/Pause support circuit being helpful, I was going through the code and noticed a TODO here (line 751: rx_engine.v):

```
// descriptor fetch
    // wait for queue query response
    if (s_axis_desc_req_status_valid) begin

        // update entry in descriptor table
        desc_table_dequeue_ptr = s_axis_desc_req_status_tag & DESC_PTR_MASK;
        desc_table_dequeue_queue_ptr = s_axis_desc_req_status_ptr;
        desc_table_dequeue_cpl_queue = s_axis_desc_req_status_cpl;
        desc_table_dequeue_invalid = 1'b0;
        desc_table_dequeue_en = 1'b1;

        if (s_axis_desc_req_status_error || s_axis_desc_req_status_empty) begin
            // queue empty or not active
            // TODO retry if empty?

            // invalidate entry
            desc_table_dequeue_invalid = 1'b1;

            dec_active_desc_req_1 = 1'b1;
        end else begin
            // descriptor available to dequeue

            // wait for descriptor
        end
    end
```

I was able to get ```tready```  to pulse low successfully by making descriptor fetch to 'do its thing' only for descriptor, "valid:HIGH and empty:LOW"
 ``` if (s_axis_desc_req_status_valid && !s_axis_desc_req_status_empty) begin ```  in line 
[ ```https://github.com/corundum/corundum/blob/master/fpga/common/rtl/rx_engine.v#L740C5```](https://github.com/corundum/corundum/blob/master/fpga/common/rtl/rx_engine.v#L738C5-L762C8)
 
I have also now implemented a buffer in the ```mqnic_app``` with sights on the ```tready``` to buffer the data if your RX_FIFO runs out of room. BUT ... (there is always a BUT :) ) 
```tready``` never seems to go HIGH after the first time it does go LOW. 

I am fearing a stall condition that the pipeline encounters after seeing the first ```s_axis_desc_req_status_empty``` . 
Where should I be focussing on? 

Thanks in advance!!!

vaniaprkl, 2023-08-02:
Hi Alex,

So an update on my attempt to implement the PFC/Pause support circuit that was suggested.  

Since the change to look at both ```s_axis_desc_req_status_valid``` and ```s_axis_desc_req_status_empty``` in the ```rx_engine.v``` was clearly getting the pipeline stalled after seeing the first ```tready``` go low, I decided to move further back up the hierarchy and chase down the source driving this ```s_axis_desc_req_status_valid```. 

The source of both these signals were traced back to ```queue_manager.v```. Upon further study, I think the reason for the lockup is some pointer/address moving ahead even when I am telling the ```queue_manager.v```  to do nothing when seeing ```queue_empty``` go ***HIGH***. What I changed was comment out the assertion of ```m_axis_dequeue_resp_empty_next = 1'b1``` and ```m_axis_dequeue_resp_valid_next = 1'b1```

```
...
        if (!queue_ram_read_data_enable) begin
            // queue inactive
            m_axis_dequeue_resp_error_next = 1'b1;
            m_axis_dequeue_resp_valid_next = 1'b1;
        end else if (queue_empty) begin
            // queue empty
          // ***m_axis_dequeue_resp_empty_next = 1'b1; ***  <== this is what I commented out
          // ***m_axis_dequeue_resp_valid_next = 1'b1; ***   <== this is what I commented out
        end else begin
            // start dequeue
            m_axis_dequeue_resp_valid_next = 1'b1;
...
```

The change above did the same as before and locked up after tready went ***LOW*** and failed to go ***HIGH*** again.

So having moved up to the issuer of the valid and empty that generates the descriptor request, I was hoping to not discard packets when ```queue_empty``` is ***HIGH***. Having built it on a whim I soon realized this possibly  is not even the descriptor  queue I should be dealing with looking at '''queue_empty'''. Am I right on that realization?

Currently trying to figure out how to get ```tready``` to return ***HIGH*** and also not throw away the packet when there is no descriptor.

alexforencich, 2023-08-02:
So, sounds like what you want to do is not ever report queue empty, instead block everything until a descriptor is actually available?  Not sure what the best approach would be to do something like that.  Reissuing requests at various places in the datapath would be one option - reissue the descriptor fetch request, reissue the dequeue request, etc.  Another option might be to adjust the queue management logic to recycle dequeue requests internally if the queue is empty.  The hard part, I think, is going to be maintaining proper ordering when reissuing operations as you have to be able to keep the pipeline full.

vaniaprkl, 2023-08-02:
Hi Alex,

What you stated here " _instead block everything until a descriptor is actually available?_" is exactly what we are attempting to do. I agree that reissuing requests seem a bit complicated especially because of all the maintenance of proper ordering as you pointed out. Your second suggestion of adjusting the queue management logic to recycle dequeue requests internally if the queue is empty seems like what we are attempting to do here. 

In order to do this can we set a flag to stall the pipeline within the ```queue_manager.v``` ?
What I am currently looking at ``` op_req_pipe_reg <= op_req_pipe_next``` in ```queue_manager.v``` file and changing it to be ``` op_req_pipe_reg <= op_req_pipe_reg``` until the queue is not empty. Thinking of doing this hold with the use of flags as of now.

Any recommendations or am I out of my mind (o.O) trivializing what needs to be done?

alexforencich, 2023-08-02:
I guess there are a couple of potential paths forward.  One of them is to ignore traffic classes on RX and simply block everything if a single RX queue runs out of descriptors.  This would probably be the simplest thing to do, although still not very trivial.  I think the proper place to handle this would be in the descriptor management or queue management logic.  The pipeline can't be stalled for a number of reasons (for example, the register writes to enqueue descriptors go through the same pipeline, so stalling the pipeline would result in a deadlock), instead there needs to be some sort of method to recycle failed requests back through the queue manager pipeline.  There will likely need to be some provisions to handle ordering, in the case of multiple operations against the same queue.

I should mention that the descriptor fetch and queue management logic are going to be completely rewritten at some point in the near future to support variable-length descriptors, batch descriptor reads, caching, and pre-fetching, and this will also consolidate the descriptor fetch and queue management logic.  The pipeline will also get a significant overhaul to eliminate pipeline hazards to improve throughput.  I have been working on the completion side so I have some idea of what things will look like, but the descriptor fetch side is going to be a bit more complex.  Although, the pipeline will be purely for accessing the queue RAM and not performing individual enqueue/dequeue operations, so that could make waiting for new descriptors to be enqueued much simpler.

The other method is a more general setup, where different traffic classes are effectively isolated and some TCs can drop packets while others cannot.  This would require quite a few modifications, perhaps something like:

- Extract traffic class information from inbound RX traffic (VLAN PCP, IP DSCP, etc.)
- Per-TC queues on RX path
- Updated memory management in RX engine to support multiple TCs
- Per-TC operation tracking in RX engine
- RX engine support for per-TC retry-on-empty

vaniaprkl, 2023-08-07:
Hi Alex,

Thanks for all the feedback and the details. It's definitely not trivial and I'm going to just to wait for the next drop since you have an overhaul of changes coming. 

But I was curious on something, while I was going to the code earlier, I noticed the head and tail pointers of the queue are 16 bits wide but the number of queues allocated seem to be in increments of 1024(and its not a parameter). Is this some kind of restriction? My head is circling around increasing the number of allocated queues to hopefully reduce the number of packet drops. Would increasing the Queue manager and RX Engine parameters in the config.tcl help? Is there a max I have to stay under?

```
# Queue manager configuration
dict set params EVENT_QUEUE_OP_TABLE_SIZE "32"
dict set params TX_QUEUE_OP_TABLE_SIZE "32"
dict set params RX_QUEUE_OP_TABLE_SIZE "32"
dict set params CQ_OP_TABLE_SIZE "32"
dict set params EQN_WIDTH "6"
dict set params TX_QUEUE_INDEX_WIDTH "13"
dict set params RX_QUEUE_INDEX_WIDTH "8"
dict set params CQN_WIDTH [expr max([dict get $params TX_QUEUE_INDEX_WIDTH], [dict get $params RX_QUEUE_INDEX_WIDTH]) + 1]
dict set params EQ_PIPELINE "3"
dict set params TX_QUEUE_PIPELINE [expr 3 + max([dict get $params TX_QUEUE_INDEX_WIDTH] - 12, 0)]
dict set params RX_QUEUE_PIPELINE [expr 3 + max([dict get $params RX_QUEUE_INDEX_WIDTH] - 12, 0)]
dict set params CQ_PIPELINE [expr 3 + max([dict get $params CQN_WIDTH] - 12, 0)]

# TX and RX engine configuration
dict set params TX_DESC_TABLE_SIZE "32"
dict set params RX_DESC_TABLE_SIZE "32"
dict set params RX_INDIR_TBL_ADDR_WIDTH [expr min([dict get $params RX_QUEUE_INDEX_WIDTH], 8)]
```

alexforencich, 2023-08-07:
The producer and consumer pointers relate to the number of items in the queue, not the number of queues.  The size of the queue is determined by the driver, it can pick any size it wants from a handful of entries up to 32768 entries.  With the current version of Corundum, you should be able to change this at run time via ethtool.  Also, the number of TX and RX queues that are enabled by the driver can be controlled dynamically by ethtool, and by default are based on the number of CPU cores in the host system.

I'm actually not sure if changing the size of the queue will have much effect on performance.  It's one of those things that needs to be big enough to get good performance, but beyond some point will just end up wasting RAM.  What point that is is likely going to be dependent on a number of different factors.  But it's easy to tune via ethtool, so feel free to experiment.

vaniaprkl, 2023-08-14:
Hi Alex, Thanks for all the pointers.  
I'll experiment on sizing using ethtool and looking forward to your next drop.

closed 2023-08-14 by vaniaprkl