corundum/corundum #158: Will Xilinx SN1022(Alveo U45N) be supported?

PriceHuang, 2023-07-25:
Hi,
Currently I got a Alveo U45N on hand, and want to test Corundum on it. Althought the board name U45N is similar to other Alveo's, Alveo U45N just Alveo SN1022 changed its name. Which board should use be as start point?

alexforencich, 2023-07-25:
I actually just recently received an SN1022 myself, so once I can figure out how to get JTAG access I'll look in to adding support upstream.  Really not sure what the obsession is with oddball JTAG connectors.....

It looks like the U45N and SN1022 are identical - OpenNIC recently renamed "sn1022" to "au45n".  So, seems like the specific board doesn't matter, but might as well name it "au45n" and note it down in the docs as "Alveo U45N/SN1022" to make it clear that they're interchangeable.

Now, as for porting, it looks like the board has GTY transceivers wired to one QSFP and GTM transceivers wired to the other QSFP.  I have CMAC + GTY wrappers, but nothing for GTM.  So I think we can roll with a CMAC core with integrated GTM for now, and then perhaps revisit that later on if more control over the GTM channels is required.  

Also, do you have an XDC for the board with the full pin listing?  The only docs that I have been able to find so far are what are in the OpenNIC repo, which have the PCIe and GTY/GTM pins, but not the DRAM pins.  And I'm not sure if there might be other important pins to watch out for, such as interfaces for a PLL chip, which seems to be relatively common on the newer Alveo boards.

alexforencich, 2023-07-27:
Also, do you have the JTAG adapter board for the U45N?  Can you possibly post a good quality picture of the front and back of it?

shroud404, 2023-07-27:
I have it. And I also successfully ported the earlier version of Corundum without DRAM on SN1022. I put the JTAG pictures on Zulip.

PriceHuang, 2023-08-02:
> I actually just recently received an SN1022 myself, so once I can figure out how to get JTAG access I'll look in to adding support upstream. Really not sure what the obsession is with oddball JTAG [...]

The XDC file can be download from here https://www.[xilinx.com/content/dam/xilinx/support/download/alveo/u45n_xdc.xdc](https://www.xilinx.com/content/dam/xilinx/support/download/alveo/u45n_xdc.xdc)

PriceHuang, 2023-08-02:
> Also, do you have the JTAG adapter board for the U45N? Can you possibly post a good quality picture of the front and back of it?

AU45N/SN1022 using the ADK2(Alveo Debug Kit) connector as programmer.
[image]
[image]

PriceHuang, 2023-08-02:
> I have it. And I also successfully ported the earlier version of Corundum without DRAM on SN1022. I put the JTAG pictures on Zulip.

Oh that's great! Could you please share the "Zulip" link? Thank you.

alexforencich, 2023-08-02:
Zulip channel: https://corundum.zulipchat.com/#narrow/stream/341410-fpga

alexforencich, 2023-11-13:
I just took a crack at getting my SN1022 working, at least on QSF1 that uses GTYs.  I cobbled together an adapter to connect the board to a standard Xilinx USB JTAG cable.  Initially I used xcu26-vsva1365-2L-e and had something working on the board.  However, after checking the sysmon, it seems that the board uses Vccint of 0.72v, which means the part in Vivado is actually xcu26-vsva1365-2LV-e.  After switching to that, the design miserably fails timing (-180 ns or so TNS, worse than the NetFPGA SUME).  So, it will likely be quite some time before the SN1022/AU45N can be supported upstream.

trosenkranz, 2025-04-17:
@alexforencich I have implemented a port to the Alveo U45N. With a single Intelligent Design Run (IDR) timing clousure is achieved (not yet check in hardware). Check our fork here: https://github.com/bitaggregat/corundum/tree/Alveo-U45N-port (still under development).

Are you interested in a PR, Alex?

Tobias

Headcrabed, 2025-04-17:
Sorry for off-topic, but do you guys have the newest satellite controller firmware for SN1000/u45n? I bought a SN1000 from eBay but it has an outdated firmware and seems have some compatibility issues with new versions of vivado.

alexforencich, 2025-04-17:
@trosenkranz I actually already have a port, I just haven't committed it yet since it fails timing miserably on the -2LV.  I can try adding some additional directives for Vivado to help with timing closure...what did you need to add to the makefile and/or config.tcl to get it to close timing?

Headcrabed, 2025-04-17:
> Sorry for off-topic, but do you guys have the newest satellite controller firmware for SN1000/u45n? I bought a SN1000 from eBay but it has an outdated firmware and seems have some compatibility issues with new versions of vivado.

Seems I have to contact Xilinx FAE to get the firmware, but since this card is from eBay, so no chance….  :(

trosenkranz, 2025-04-22:
> [@trosenkranz](https://github.com/trosenkranz) I actually already have a port, I just haven't committed it yet since it fails timing miserably on the -2LV. I can try adding some additional [...]

As I have written before, I achived Timing clousure using one IDR. Best P&R without IDR run was -0.227ns WNS with -19.373 TNS.

One thing I did, was setting the LOC constraint of the PCIe-Hard-IP-Block from _X0Y0_ to _X0Y1_ in the _config.tcl_. I added at around line 250 `dict set pcie_config "CONFIG.pcie_blk_locn" {X0Y1}`. I can do a full diff on the Makefile/config.tcl, if you want.

Regarding the Core Voltages of the FPGA: I do have an Alveo U45N and my Core Voltage is 0.85V. Does Xilinx/AMD sell Alveos with two different fitted FPGAs? (I have to check the FPGA ID later. Do you know, where I can readout the FPGA ID in Vivado?)

trosenkranz, 2025-04-23:
I have generated a bitstream with the FPGA ID of the Low Voltage Version of the FPGA. I had no issues programming that bitstream to the FPGA.

jxc98728, 2025-04-25:
> I have generated a bitstream with the FPGA ID of the Low Voltage Version of the FPGA. I had no issues programming that bitstream to the FPGA.

Hi, may I ask how you disabled the GTY/GTM QSFP lpmode off? Is this possible using CMS?

JakaBac, 2025-04-25:
@trosenkranz 
The speed and voltage grade is not part of the IDCODE, so all xcu26 FPGAs will look the same over JTAG. The FPGAs will also not check if you try to load different speed grade/LV grade bitstream on it. Chances are simply that the design will not work if it was loaded on a slower FPGA.

I have a SN1022 which is U45N's predecessor and my board also has 0.85V VCCINT. -2LV part is basically -2L when used with 0.85V VCCINT.

Currently nobody knows which boards have 0.72V and which ones 0.85V.

jxc98728, 2025-04-25:
> [@trosenkranz](https://github.com/trosenkranz) The speed and voltage grade is not part of the IDCODE, so all xcu26 FPGAs will look the same over JTAG. The FPGAs will also not check if you try to [...]

Is checking the SysMon to get VCCINT the only way to specify the actual speed and voltage grade?

JakaBac, 2025-04-25:
> Is checking the SysMon to get VCCINT the only way to specify the actual speed and voltage grade?

You can also use `sensor` command on the debug port UART connection to the SuC. 
It doesn't seem to be possible to distinguish the VCCINT configuration by just looking at the board.

trosenkranz, 2025-04-28:
> > I have generated a bitstream with the FPGA ID of the Low Voltage Version of the FPGA. I had no issues programming that bitstream to the FPGA.
> 
> Hi, may I ask how you disabled the GTY/GTM QSFP lpmode? Is this possible using CMS?

I did nothing. Either my Adapter doesn't passthrough this signal or my 10GBit SFP+ Module doesn't use it.

I also tested the OpenNICshell Project with the Alveo U45N and transmitted data using an Optical Cable and it worked without any extra steps involved. Maybe all you have to do is instantiate the CMS-Subsystem in your project and the rest happens automagically...

JakaBac, 2025-04-28:
Apparently there are different versions of SuC firmware and the older ones are incompatible with the CMS IP.

@trosenkranz has U45N which means he has the latest firmware. Unfortunately there are no updates available on AMD site for U45N.

There is an update for SN1022 in the SN1000 "lounge" but that requires SN1022 firmware to be running (along with drivers on the host system)

Headcrabed, 2025-04-28:
> Apparently there are different versions of SuC firmware and the older ones are incompatible with the CMS IP.
>
> [@trosenkranz](https://github.com/trosenkranz) has U45N which means he has the latest [...]

@JakaBac Thanks for reply. My SN1022's firmware version seems to be reported as 3.0.5, and not working correctly with the CMS IP and firmware provided with Vitis 2024.2. Can you upload the newest firmware here? And I will try if older versions of vitis would work on my card.

Headcrabed, 2025-04-28:
And for dumping out firmware in the SC controller, I found this: https://xilinx.github.io/Alveo-Cards/master/management-specification/sc-firmware-update.html#x37-sc-flash-read-data-block

alexforencich, 2025-04-28:
Unfortunately, I think the dump command might only work with newer SC firmware.  But I guess that works nicely if you want to dump a newer firmware.  Also, I'm not sure what address ranges need to be dumped.  The SN1022/U45N/X3522/X3 are also "special" in that they use an LPC ARM microcontroller instead of an MSP430.  I don't know what effect that might have on the process.

Headcrabed, 2025-04-28:
@alexforencich Personally I prefer to get an official packed firmware for SN1022 first, since official firmware updater seems only receive a txt format firmware. https://adaptivesupport.amd.com/s/article/73654?language=en_US

alexforencich, 2025-04-28:
I mean naturally that would be optimal.  But if it's not available, what else can you do?

jxc98728, 2025-04-28:
I have a SN1022_Update_Bundle and here are some info:

```
Xilinx Alveo SmartNIC SN1000 Series Firmware Package

Version: 1.1.1.1021-1

Component versions

  -------------------------------------------------------------------
  Component                                        Version
  ------------------------------------------------ ------------------
  SN1022 firmware bundle                           1.1.1.1021

  -- Board support microcontroller (SUC) firmware  3.0.10.1000

  -- Network microcontroller (NMC) firmware        1.4.7.0

  -- SmartNIC FPGA stage 1 bitstream               7.13.11.0 [7_N11]

  -- SmartNIC FPGA stage 2 bitstream               7.13.11.0 [7_N11]

  -- UEFI driver                                   1.2.0.0-unsigned
  -------------------------------------------------------------------
```

So apparently this bundle updates SUC firmware to 3.0.10, and since this bundle released in Oct 2021, I believe we should probably use Vivado 2021.1/2021.2 to generate corresponding CMS IP.

JakaBac, 2025-04-28:
@jxc98728 That is the last update bundle that I am aware of.
Someone with U45N posted their SuC version (obtained through SuC UART commands on the debug connector)
That seems to be newer:
3.0.25.1000 v3_0_25_1000

If the firmware is from 2021 it doesn't mean that Vivado 2021 needs to be used. The SuC firmware just needs to be new enough to understand "standard" CMS commands.

If the LPC part is not locked for reading it may be possible to read the whole SuC firmware over its JTAG (available on the SN1022/U45N debug connector). But keep in mind that SuC also has access to the all flash chips on the SN1022/U45N and it definitely has some settings/data stored there. So just updating the SuC via JTAG may cause it to misbehave, since it has control of the board power this has potential to cause trouble

JakaBac, 2025-04-28:
@Headcrabed Unfortunately the update files are on AMD's restricted access site with "confidential" written all over it, si I am not allow to upload the files to GitHub

irprk, 2025-08-09:
@trosenkranz I'm also having timing problems with U45N, even with IDR. Which version of Vivado did you use to close timing in corundum and open-nic?

trosenkranz, 2025-08-11:
Check out https://github.com/bitaggregat/corundum/tree/Alveo-U45N-port. I developed some placement rules, which are committed there too. As far as I can remember I only did a single 100GBit Port.

With OpenNIC shell it was more a pain i* t* a**. I had to use multiple runs, had to configure them to use the (?) explorative (?) type algorithm in syth and P&R and also had to apply QoR suggestions multiple times...