corundum/corundum #160: Unaligned transfer on the DMA interface for the application block

KireinaHoro, 2023-07-26:
I'm trying to issue unaligned transfers on the DMA transfer over PCIe, to match the semantics of AXI strobed writes from my application logic.  I am currently using `dma_client_axis_sink.v`, which seems to process `TKEEP`; however, the segmented DMA interface does not have byte enable signals on the descriptor.  Does the DMA subsystem support strobed writes?

alexforencich, 2023-07-26:
PCIe does not support that outside of the first and last dword in each request TLP, so the DMA engine does not support it.  And incidentally this is the main reason I haven't written an AXI slave DMA client yet; converting the byte enables to DMA operations is not so straightforward since you have to interpret the byte enables to extract contiguous blocks and issue a separate DMA transfer request for every block, even if it's only 1 byte.  

I guess maybe it could be possible to add a way to specify the byte enables in some way, but the semantics for how that might work get a bit tricky.  Perhaps if we limit it to only 32 bit writes that might be reasonable to implement.  But either way, you would still have to slice and dice each transfer into a bunch of separate DMA operations, it just might reduce the number of DMA operations that you need when you get a really strange set of byte enables.

KireinaHoro, 2023-07-27:
Thanks for the information!  I think at the moment what I need is just unaligned but contiguous transfers instead of arbitrary byte-enable configurations.  As you said that the PCIe TLP format allows for a byte-enable on the first and last dword, that should in theory allow for unaligned transfers.  Does the Corundum DMA subsystem support this, for example if I issue unaligned requests onto the descriptor bus?

alexforencich, 2023-07-27:
Yes, the DMA interface supports fully unaligned transfers.

closed 2023-08-15 by KireinaHoro