corundum/corundum #161: Line rate tests on AU50

vaniaprkl, 2023-08-09:
Hi Alex, I am doing some line rate testing on the U50 and seeing something strange. 

I have built a IF_COUNT=4 design and feeding all four interfaces from a switch at a rate close to 9.5Gbps. 

I use linux's ```sar -n DEV 1``` command to monitor the performance of the interfaces. So strangely enough when only one port is active, I see an ingress rate close to the transmission rate of 9.5Gbps. 

However, when I enable all 4 interfaces to receive the same traffic from the switch, the ingress rate drops on all 4 interfaces to around 3 Gbps. Am I running into some bottleneck or do I need to configure something more than IF_COUNT=4?

Thanks in Advance for any advice!!!

vaniaprkl, 2023-08-14:
Hi Alex, 
So I did some more digging around and found ```queue_ram_read_data_head_ptr``` and ```queue_ram_read_data_tail_ptr``` in ```queue_manager.v```. I brought out these signals to a register to observe how it changes when I enabled between 1-4 interfaces.

I noticed when only 1 interface is enabled the ```queue_ram_read_data_head_ptr``` and ```queue_ram_read_data_tail_ptr``` were far apart each time I read it and ```sar``` was showing  the transmitted data rate of 9.5Gbps being received as well.
When I enabled  additional interfaces, the ```queue_ram_read_data_head_ptr``` and ```queue_ram_read_data_tail_ptr``` were very close to each other and the ingress data rate recorded by sar on every enabled interface reduced as below:
1 port  Enabled: 9.5Gbps per port ==> Total 9.5Gbps
2 ports Enabled: 8.40Gbps per port ==> Total 16.8Gbps
3 ports Enabled: 5.53Gbps per port ==> Total 16.59 Gbps
4 ports Enables: 4.14Gbps per port ==> Total 16.56 Gbps

I have a hunch that this happens because of the head and tail getting too close to each other before requesting for more descriptors. Currently, I'm searching for where this comparison of head/tail happens in the driver to see if I can start enqueuing descriptors more often and not when head/tail gets too close.

vaniaprkl, 2023-08-15:
Hi Alex, 

In my deep dive into the code under ```modules/mqnic_rx.c``` I found  there is a function called ```mqnic_refill_rx_buffers```.  I was wondering if this is where I need to make the change for the drivers to update sooner to keep head/tail pointers further apart. I am going to play with this next but not seeing an abs around the subtraction of ```ring->prod_ptr``` and ```ring->cons_ptr``` is a bit confusing. But since its working I am assuming it's something to do with "pointers" and "u32".  

Thanks for any pointers in advance on where I should be heading.

```
int mqnic_refill_rx_buffers(struct mqnic_ring *ring)
{
	u32 missing = ring->size - (ring->prod_ptr - ring->cons_ptr);
	int ret = 0;

	if (missing < 8)
		return 0;

	for (; missing-- > 0;) {
		ret = mqnic_prepare_rx_desc(ring, ring->prod_ptr & ring->size_mask);
		if (ret)
			break;
		ring->prod_ptr++;
	}

	// enqueue on NIC
	dma_wmb();
	mqnic_rx_write_prod_ptr(ring);

	return ret;
}
```

alexforencich, 2023-08-16:
Basically, the driver is set up to maintain a certain number of free buffers in the RX queues.  I think the default is 1024.  They're not always added, they're only added when there is room for a handful of them.  Looks like maybe I arbitrarily set that at 8.  The idea here is to try to take advantage of the CPU cache by handing things in batches as much as possible.

You should be able to change the queue size via ethtool APIs.  But, it's possible that the bottleneck is actually the host system in this case; the host system simply might not be able to keep the queue full.  What are the specs of the host system that you're using?

vaniaprkl, 2023-08-16:
Hi Alex, Increasing them in batches makes sense to me. I tried playing around with the "8" but it did not matter much in the performance because I have a sense that the function that refills the buffers is not being called often enough and not soon enough(head and tail seem to be getting very close). Trying to play around with when and where I might be able to optimize this at the moment. 

Experimented with changing the queue size using ethtool, but anything more than 1024 seemed to be detrimental. 

But the value of "8" itself for a "batch-add" is logical, as I now understand from your feedback. I can now read the code as  always-going-to-refill the rx buffers as-soon-as there is room for "8". While I was experimenting,  I commented out :) the threshold check of "8" and it somewhat degraded performance. I'll play around adjusting this a little more. 

I am running this on a system with 14 cores/28 threads and 192GB memory. I'll try to beef up the specs on this system at some point. What specs did your system have where you achieved close to 10Gbps on all 4 ports? So that I can try to match maybe?

alexforencich, 2023-08-17:
Specifically what CPU?  And also, what sort of packet sizes are we talking about here?  And have you confirmed that the card is enumerating properly over PCIe with the correct speed and number of lanes?  The driver should print out a summary of the PCIe config when it loads.

In my current dev setups, I am using a combination of Intel Xeon CPUs, either 6130, 6138, or 6148.  And I can consistently hit 100 Gbps with jumbo frames, and well over 50 Gbps with 1500 byte frames.