corundum/corundum #162: How to debug host freeze

KireinaHoro, 2023-08-15:
I'm currently using a VCU1525 with an AMD Ryzen 7 2700 CPU to run Corundum with an application block that sends write DMA transactions ([PsPIN](https://github.com/KireinaHoro/pspin)).  I'm experiencing seemingly random whole-system hangs when the design is under load with Ethernet traffic and DMA transactions from my application block.  The Linux kernel seems to be completely frozen (cursor on the tty not blinking) and also with no dmesg output.  It is supposedly not a kernel panic or oops since I've configured kdump on the system to automatically reboot and perform a dump, and that's not happening here as well.  Any ideas on what the issue may be and how I could debug this?

P.S. [my Corundum branch](https://github.com/KireinaHoro/corundum/tree/pspin) tracked changes from `master` up to c708bc45cd0d5b44274d238dd9fadd16335f337d, just in case if this had been fixed later on I'll pull the changes.

KireinaHoro, 2023-08-15:
Thanks for the prompt reply!  I'll merge upstream and see if that helps; will report back.

KireinaHoro, 2023-08-16:
Ok so I merged `master` into my branch and rebuilt the design, the same set of benchmarks would still hang the system.

KireinaHoro, 2023-08-17:
Thanks.  Just checked that our card is also running on a degraded link (x8 instead of x16).  Here's the output of `lspci -vv`:

```
1d:00.0 Ethernet controller: Device 1234:1001
        Subsystem: Xilinx Corporation Device 95f5
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Region 0: Memory at 1ffe000000 (64-bit, prefetchable) [size=16M]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [60] MSI-X: Enable+ Count=64 Masked-
                Vector table: BAR=0 offset=00010000
                PBA: BAR=0 offset=00018000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 512 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x16, ASPM not supported
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s (ok), Width x8 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range BC, TimeoutDis+, NROPrPrP-, LTR-
                         10BitTagComp-, 10BitTagReq-, OBFF Not Supported, ExtFmt-, EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-, TPHComp-, ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
                         EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [1c0 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn-, PerformEqu-
                LaneErrStat: 0
        Kernel driver in use: mqnic
        Kernel modules: mqnic
```

alexforencich, 2023-08-17:
Interesting.  Is that slot limited to x8, or did it just enumerate at x8 for some reason?  Have you tried it at x16 to see if you're getting the same hang?

KireinaHoro, 2023-08-18:
Ok so the x8 was due to the motherboard (MSI X470 GAMING PRO) having either x16/x0 or x8/x8 for the two PCIe 3.0x16 slots, and we had a GPU in the other slot.  I moved the GPU to the third PCIe2.0x16 slot, the FPGA now has full x16 link; I reran the benchmark set and the whole system hung again :(

KireinaHoro, 2023-08-28:
Any update on this?

alexforencich, 2023-08-28:
I haven't had time to work on this yet, hopefully in the next week or two I'll be able to take a crack at recreating it.  I tried forcing the link rate to gen 1 x16, but that didn't crash, so I think I need to try a narrower slot.

KireinaHoro, 2023-08-28:
Thanks!  Just a remark that it's probably not (only) due to a narrow configuration, since I managed to hang it actually more often at 16x 8 GT/s.  If you have time, I can send you a bitstream and write up some short instructions on how to reproduce it almost stably on a VCU1525, with my application block.

alexforencich, 2023-08-28:
Well, I'm probably going to have to stick some ILAs in and perhaps some other stuff.  So a bitstream is not sufficient, I'll need to be able to run the whole build.

alexforencich, 2023-08-28:
Also, it's entirely possible that there is more than one bug that needs to be rooted out.

alexforencich, 2023-09-23:
As part of testing the pause frame functionality, I put a card in a narrow slot to reduce the PCIe BW.  But, this did not result in any crashes.  So, it looks like some additional work is going to be required to replicate the issue and debug it properly.  

However, moein just discovered this minor issue: https://github.com/alexforencich/verilog-pcie/commit/2074d7212f1e2136a92234ab2b09ee29bdb96fb1 .  Does applying that change anything?

KireinaHoro, 2023-09-27:
Thanks for the attention into this!  To reproduce the hardware you can clone my branch and make the bitstream: https://github.com/KireinaHoro/corundum/tree/pspin/fpga/mqnic/VCU1525/fpga_100g/fpga_pspin

The design is a bit tough to close timing; I usually get a very small WNS (~-0.05) and TNS (> -1).  This sometimes would result in PCIe (link training?) taking longer to get enumerated in the case of a reset (e.g. `mqnic-fw -b`).

The software reproduction is a bit more involved.  You need:
- Kernel module: https://github.com/KireinaHoro/corundum/tree/pspin/fpga/app/pspin/modules/mqnic_app_pspin
- User app: https://github.com/KireinaHoro/pspin/blob/vivado/examples/datatypes/run_eval.sh (run `./run_eval.sh -mp`)

You would need the [PULP RISC-V toolchain](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain) set up for the build process to work.  Please tell me if you run into problems with this; I can try to produce binary artefacts for you.