corundum/corundum #164: An erro during generating project at file "./config.tcl" line 240: ERROR: [Ipptcl 7-29] Invalid param '03f'.

Chin725, 2023-08-24:
You could change param EQN_WIDTH from 6 to 5, because PCIE IP supports the Max value of CONFIG.PF0_MSIX_CAP_TABLE_SIZE up to 2**5-1=0x01f.

Chin725, 2023-08-24:
param EQN_WIDTH is at file config.tcl line 79

alexforencich, 2023-08-28:
Which target are you building?

alexforencich, 2023-09-23:
I just checked the Xilinx docs; 7-series, US, and US+ all support the full table size (0x7ff), even the kintex 7 block which isn't currently supported.  So it's unclear what the problem is unless you can provide some more information about how you're running in to this problem.  Only thing I can think of is maybe the core somehow got set to MSI-X internal on US+ as the core's internal table is limited in size, but it should be configured for MSI-X external.

alexforencich, 2023-11-13:
Closing this as I cannot reproduce or fix the issue without more information

closed 2023-11-13 by alexforencich