corundum/corundum #165: kernel driver license stated ambigiously

Basseuph, 2023-09-07:
The license of the Linux kernel driver code is stated in an ambigious way. The source code is licensed under BSD-2-Clause-Views according to the SPDX header (https://github.com/corundum/corundum/blob/master/modules/mqnic/mqnic_main.c#L1), but the module exports a dual licensing string "BSD/GPL" (https://github.com/corundum/corundum/blob/master/modules/mqnic/mqnic_main.c#L19). The problem now is that https://docs.kernel.org/process/license-rules.html#id1 states the actual exact license conditions need to be extracted from the in code licensing statements. IMHO it is all fine for the BSD part, but as there is no statement about the GPL side, it is unclear whether this is GPLv2 only, GPLv2 or later, GPLv3, etc.

We can provide a patch to fix this, but the question now is, which license would you/we like to choose for the code?
This also boils down into the formal license attribution used by yocto, which actually raised this item internally on our side.

My personal feeling is GPLv2-only is OK, but not sure about this. Maybe @joft-mle you can add another view/ advise on this?

joft-mle, 2023-09-07:
Adding to the above comment by @Basseuph above, the rules for SPDX seem to include the possibility to state "license A or license B" (https://docs.kernel.org/process/license-rules.html#license-identifier-syntax); example:

```
// SPDX-License-Identifier: BSD-2-Clause-Views OR GPL-2.0
```

Where `GPL-2.0` **without** a `+` as an appendix would mean "GPL 2 only". And `GPL-2.0+` => "GPL 2 or later", if I understand correctly.

Basseuph, 2023-09-22:
@alexforencich any idea on this one?

alexforencich, 2023-09-23:
Does it need to be licensed under GPL?  I could change the module license to BSD instead of dual.  But I suppose since the kernel is GPL 2.0, it has to be GPL 2.0 plus maybe something else.  So I think adding the "OR GPL-2.0" is probably going to be the way to go.