corundum/corundum #169: Custom App Ports

oskar-mle, 2023-10-07:
Hey everyone,

me and @Basseuph are currently looking into the implementation of a custom app interface as first discussed in the developer meeting on 9/11/2023. The idea is to enable user-defined input and output signals for the mqnic application block, which are then routed through the hierarchy in and out of the mqnic toplevel.

Given the limitations of verilog, there seems to be no way around using preprocessor directives. There are two sections of code that must be substituted:
• custom port definitions
• custom port connections

For convenience, the custom wire definitions in the module that instantiates mqnic could also be provided. If custom signals should use additional parameters, it'd probably be sufficient to insert them at each hierarchical module that uses the custom port definitions, and set an appropriate default value.

We came up with three options to implement this, an example at EDA playground is provided for each of them:
• option 1: `include files: https://edaplayground.com/x/Msfk
• option 2: simple `define macros: https://edaplayground.com/x/q_VH
• option 3: nested `define macros: https://edaplayground.com/x/BrTL
(signal names: custom_vec_l* = vector width defined by literal, custom_vec_mp* = vector width defined by module parameter, custom_vec_cp* = vector width defined by custom parameter)

We checked if those compile and function with all commercial simulators available at EDA playground, as well as Vivado 2022.2. 

Option 1 and 2 are pretty straighforward and should work with all tools (for option 2 there are issues with xcelium at EDA playground, but that may be a matter of proper configuration). They are quite verbose, but the macros/header files could be generated via script.

Option 3 eliminates redundancy and only requires users to define the custom signals once, resulting in less code. However, this doesn't work with every tool: apparently, nested macro substitution is handled differently from tool to tool. So far it seems to work with Vivado 2022.2 and synopsys vcs 2021.09.
Also, this method has the disadvantage of always putting a comma after the last element in the custom port list, so there has to be another port after it.

Which option do you prefer? Do you have any additional thoughts?
The next step would be to actually implement a proof of concept for corundum. 

Best
Oskar

alexforencich, 2023-10-09:
As discussed in the meeting: let's look at a hybrid of 2 and 3.  Looks like adding a couple of static lines at the bottom of the included file for 3 should make it directly compatible with the HDL for 2, so you have the option of using one or the other depending on simulator support, etc.

alexforencich, 2023-10-09:
For tool support, it needs to work on Vivado, Quartus, Icarus Verilog, and Verilator.  I don't care too much about other simulators, but if some tools are having issues with a construct, it's maybe better to rework it.  

Also, I fixed up the formatting.  Apparently you need to use double backticks and separate them with spaces, like so: ```` `` `include `` ````, which results in `` `include ``.