corundum/corundum #170: Cocotb with icarus verilog gets frozen stuck at some point during simulation: How to debug?

zaidtahirbutt, 2023-10-11:
I am trying to simulate some custom logic, but when I run the cocotb simulation, it gets stuck at some point.
I did Ctrl+C and I entered "step" command in the icarus verilog console, but the simulator ticks stayed the same.
I then compiled with the simulation code with the debug flags as follows: 
`make WAVES=1 COCOTB_SCHEDULER_DEBUG=1 COCOTB_ENABLE_PROFILING=1 COCOTB_LOG_LEVEL=DEBUG
`
 and got the following error:
```
 56624.00ns DEBUG    cocotb.scheduler                   Coroutine Scheduler._do_writes yielded <<Event at 0x7f66995a3fd0>.wait() at 0x7f662fd93af0> (mode 1)

 56624.00ns DEBUG    cocotb.scheduler                   Scheduled coroutine Scheduler._do_writes

 56624.00ns DEBUG    cocotb.scheduler                   All coroutines scheduled, handing control back to simulator
```

I did go into the python files of cocotb scheduler and simulator etc, but understanding those and debugging my verilog logic using this method would take a lot of time. 

What debugging method do you suggest here? I did read your reddit suggestion (from 4 years ago) of commenting chunks of code out.

alexforencich, 2023-10-12:
So, what it really comes down to is what the nature of the hang is, as you'll want to look at different things in order to track it down. First thing to sort out: did you make a combinatorial loop and hang the simulator, or is the simulation still making forward progress but nothing is happening?

Probably the simplest way to see if you have a combinatorial loop is to enable waveform dumping, run it until it hangs, wait a few seconds, kill the simulator, check the end time of the waveform dump, and then do it again and see if it ends in exactly the same place. Usually a combinatorial loop will cause problems immediately, though. These are usually straightforward enough to find through inspection and commenting stuff out until you break the loop and the simulator no longer hangs.

If it isn't a combinatorial loop, then the process of getting that sorted out generally requires a more careful examination of the waveform dump to see where the hold up is. Sometimes data will get stuck somewhere or lost somewhere due to handshaking bugs, or a module will get stuck waiting for some operation to complete.  So checking tready/tvalid handshaking signals is a good starting place, as well as tracing out the last few operations before the hang.

zaidtahirbutt, 2023-10-12:
I found out the bug, it was in the combinatorial loop. I needed to add an extra if condition to activate a register in the combinatorial loop.

The simulation was hanging at a certain time stamp and it was always hanging at the same time stamp.

The way I found out the bug was that I pressed "Ctrl+C" after the simulation got stuck and wrote "finish" in the Icarus Verilog console so that the waveform gets dumped until the last tick before it froze.  Then after close inspection of the waveform and then my code I found the problem. 

It took quite some time to find the bug because I had written a few FSMs (that were dependent on signals from each other) without simulating them and then I simulated all the FSMs together, so I wasn't sure which FSM was causing the simulation to get stuck. 

What do you suggest with regards to simulating the logic while developing it? How often should I be simulating the code while I am focused on writing the logic first?

Also why does the cocotb and/or simulator get stuck? Should it still simulate and show all the signals going red (XX) instead of just freezing up? 

Isn't there a more reliable way inside cocotb to check simulation hangs like, "which signal/wire/register was being driven by cocotb, or which value was being read by cocotb before it froze up?"

I haven't tried out cocotb 1.8, is it better with simulation hangs?

closed 2023-10-13 by zaidtahirbutt