corundum/corundum #172: insmod mqnic.ko error

GraceDouX, 2023-11-03:
My board's pcie is Gen4x8 , Gen3x16 but when i insmod mqnic.ko, Some errors were generated,is there something wrong in my kernal? Hope you can give me some advice,thanks.

> [  146.544443] **mqnic: loading out-of-tree module taints kerne**l.
[  146.544506] **mqnic: module verification failed: signature and/or required key missing - tainting kernel**
[  146.546006] mqnic 0000:01:00.0: mqnic PCI probe
[  146.546008] mqnic 0000:01:00.0:  Vendor: 0x1234
[  146.546009] mqnic 0000:01:00.0:  Device: 0x1001
[  146.546010] mqnic 0000:01:00.0:  Subsystem vendor: 0x10ee
[  146.546010] mqnic 0000:01:00.0:  Subsystem device: 0x90fa
[  146.546011] mqnic 0000:01:00.0:  Class: 0x020000
[  146.546012] mqnic 0000:01:00.0:  PCI ID: 0000:01:00.0
[  146.546021] mqnic 0000:01:00.0:  Max payload size: 256 bytes
[  146.546022] mqnic 0000:01:00.0:  Max read request size: 512 bytes
[  146.546023] mqnic 0000:01:00.0:  Read completion boundary: 64 bytes
[  146.546024] mqnic 0000:01:00.0:  **_Link capability: gen 3 x16_**
[  146.546025] mqnic 0000:01:00.0:  **_Link status: gen 3 x4_**
[  146.546026] mqnic 0000:01:00.0:  Relaxed ordering: enabled
[  146.546027] mqnic 0000:01:00.0:  Phantom functions: disabled
[  146.546027] mqnic 0000:01:00.0:  Extended tags: enabled
[  146.546028] mqnic 0000:01:00.0:  No snoop: enabled
[  146.546029] mqnic 0000:01:00.0:  NUMA node: -1
[  146.546030] mqnic 0000:01:00.0:  PCI ID (bridge): 0000:00:1c.0
[  146.546039] mqnic 0000:01:00.0:  Link capability (bridge): gen 3 x4
[  146.546040] mqnic 0000:01:00.0:  Link status (bridge): gen 3 x4
[  146.546045] mqnic 0000:01:00.0: 31.504 Gb/s available PCIe bandwidth, limited by 8.0 GT/s PCIe x4 link at 0000:00:1c.0 (capable of 126.016 Gb/s with 8.0 GT/s PCIe x16 link)
[  146.546048] mqnic 0000:01:00.0: can't disable ASPM; OS doesn't have ASPM control
[  146.546052] mqnic 0000:01:00.0: enabling device (0000 -> 0002)
[  146.546057] mqnic 0000:01:00.0: Control BAR size: 16777216
[  146.547094] mqnic 0000:01:00.0: Configured 64 IRQs
[  146.547136] mqnic 0000:01:00.0: Device-level register blocks:
[  146.547136] mqnic 0000:01:00.0:  type 0xffffffff (v 0.0.1.0)
[  146.547138] mqnic 0000:01:00.0:  type 0x0000c007 (v 0.0.1.0)
[  146.547139] mqnic 0000:01:00.0:  type 0x0000c000 (v 0.0.1.0)
[  146.547140] mqnic 0000:01:00.0:  type 0x00000000 (v 0.0.0.0)
[  146.547141] mqnic 0000:01:00.0:  type 0x0000c006 (v 0.0.1.0)
[  146.547142] mqnic 0000:01:00.0:  type 0x0000c080 (v 0.0.1.0)
[  146.547143] mqnic 0000:01:00.0:  type 0x0000c008 (v 0.0.1.0)
[  146.547144] mqnic 0000:01:00.0:  type 0x0000c110 (v 0.0.1.0)
[  146.547145] mqnic 0000:01:00.0:  type 0x0000c101 (v 0.0.1.0)
[  146.547146] mqnic 0000:01:00.0:  type 0x0000c120 (v 0.0.2.0)
[  146.547147] mqnic 0000:01:00.0:  type 0x0000c140 (v 0.0.1.0)
[  146.547148] mqnic 0000:01:00.0:  type 0x0000c150 (v 0.0.1.0)
[  146.547149] mqnic 0000:01:00.0:  type 0x0000c150 (v 0.0.1.0)
[  146.547162] mqnic 0000:01:00.0: FPGA ID: 0x04b57093
[  146.547163] mqnic 0000:01:00.0: FW ID: 0x00000000
[  146.547164] mqnic 0000:01:00.0: FW version: 0.0.1.0
[  146.547165] mqnic 0000:01:00.0: Board ID: 0x10ee90fa
[  146.547166] mqnic 0000:01:00.0: Board version: 1.0.0.0
[  146.547167] mqnic 0000:01:00.0: Build date: 2023-10-31 10:03:31 UTC (raw: 0x6540d0f3)
[  146.547169] mqnic 0000:01:00.0: Git hash: 00000000
[  146.547169] mqnic 0000:01:00.0: Release info: 00000000

alexforencich, 2023-11-03:
It says right there that the bridge is only gen 3 x4. Try a different PCIe slot. I suspect you'll need to swap your graphics card for the FPGA board so the FPGA board can get enough bandwidth.

GraceDouX, 2023-11-03:
Thank you very much i will try that .