corundum/corundum #174: Does corundum support two devices in a server?

UnknownThis, 2023-12-01:
I put two AU200 card into one server. 
I inserted two U200 accelerator cards into the same server. I identified these two U200 cards using Vivado software and downloaded the bit file into the U200 cards.
The problem is that after installing the drivers, when I use "ifconfig," it only displays two network interfaces on one of the U200 cards.

UnknownThis, 2023-12-01:
I expected to see four network interfaces. How can I solve this problem? Or is it that the driver itself only supports one U200 card?

alexforencich, 2023-12-01:
Yes, I have definitely used it with at least 3 cards in the same machine and it works fine.

UnknownThis, 2023-12-01:
Thank you. I see 4 interface after i reboot the server.

closed 2023-12-01 by UnknownThis