corundum/corundum #176: petalinux compile error for mqnic module

DJW146, 2023-12-15:
Hi Alex, thanks for sharing about the open source project corundum NIC project.
We are evaluating the possibility for 5G fronthaul protocol.

Below is the compilation errors that prompted while using petalinux for AMD-Xilinx zcu106 board.
I'm not sure what's something that caused for the errors.
Could you pls help & guide to check what maybe the problems that caused and how to proceed to finish the compilation.

==============================
ERROR: mqnic-tools-1.0-r0 do_compile: ExecutionError('/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/temp/run.do_compile.26405', 2, None, None)
ERROR: Logfile of failure stored in: /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/temp/log.do_compile.26405
Log data follows:
| DEBUG: Executing shell function do_compile
| make -C lib/mqnic/ libmqnic.a
| make[1]: Entering directory '/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/lib/mqnic'
| ar rcs libmqnic.a mqnic.o mqnic_res.o mqnic_if.o mqnic_port.o mqnic_sched_block.o mqnic_scheduler.o mqnic_clk_info.o mqnic_stats.o reg_if.o reg_block.o fpga_id.o
| make[1]: Leaving directory '/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/lib/mqnic'
| aarch64-xilinx-linux-gcc  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0=/usr/src/debug/mqnic-tools/1.0-r0                      -fdebug-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0=/usr/src/debug/mqnic-tools/1.0-r0                      -fdebug-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot-native=  -Wall -Ilib -Iinclude -MD -MP -MF .mqnic-config.d -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0=/usr/src/debug/mqnic-tools/1.0-r0                      -fdebug-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0=/usr/src/debug/mqnic-tools/1.0-r0                      -fdebug-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -Llib/mqnic mqnic-config.o timespec.o lib/mqnic/libmqnic.a -o mqnic-config -lmqnic
| /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/../../libexec/aarch64-xilinx-linux/gcc/aarch64-xilinx-linux/11.2.0/ld: lib/mqnic/libmqnic.a(fpga_id.o): Relocations in generic ELF (EM: 62)
| /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/../../libexec/aarch64-xilinx-linux/gcc/aarch64-xilinx-linux/11.2.0/ld: lib/mqnic/libmqnic.a(fpga_id.o): Relocations in generic ELF (EM: 62)
| /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/../../libexec/aarch64-xilinx-linux/gcc/aarch64-xilinx-linux/11.2.0/ld: lib/mqnic/libmqnic.a: error adding symbols: file in wrong format
| WARNING: /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/temp/run.do_compile.26405:149 exit 2 from 'make'
| WARNING: Backtrace (BB generated script):
| collect2: error: ld returned 1 exit status
| make: *** [Makefile:39: mqnic-config] Error 1
| 	#1: do_compile, /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/temp/run.do_compile.26405, line 149
| 	#2: main, /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/temp/run.do_compile.26405, line 149
NOTE: recipe mqnic-tools-1.0-r0: task do_compile: Failed
ERROR: Task (/home/faca/corundum/meta-corundum/recipes-devtools/mqnic-tools/mqnic-tools.bb:do_compile) failed with exit code '1'
=================================
I also check that file libmqnic.a with readelf -h under /ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/lib/mqnic/
something werrid, all the mqnicXXX.o are compiled for machine AArch64, 
but for reg_if.o and fpga_id.o, they are compiled for machine Advanced Micro Devices X86-64.
This is what i observed till now. Below are come captures for you reference.
Thanks.

File: /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/lib/mqnic/libmqnic.a(mqnic.o)
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              REL (Relocatable file)
  Machine:                           AArch64
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          0 (bytes into file)
  Start of section headers:          52312 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           0 (bytes)
  Number of program headers:         0
  Size of section headers:           64 (bytes)
  Number of section headers:         26
  Section header string table index: 25

File: /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/lib/mqnic/libmqnic.a(reg_if.o)
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              REL (Relocatable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          0 (bytes into file)
[... 5 lines cut ...]
  Size of section headers:           64 (bytes)
  Number of section headers:         14
  Section header string table index: 13

File: /home/faca/corundum/fpga/mqnic/ZCU106/fpga_zynqmp/ps/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/mqnic-tools/1.0-r0/lib/mqnic/libmqnic.a(fpga_id.o)
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              REL (Relocatable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          0 (bytes into file)
  Start of section headers:          7112 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           0 (bytes)
  Number of program headers:         0
  Size of section headers:           64 (bytes)
  Number of section headers:         13
  Section header string table index: 12

alexforencich, 2023-12-15:
Is it possible that was built on the host system and petalinux is pulling in the compiled files instead of the original source files?

DJW146, 2023-12-15:
Hi Alex, i git cloned corundum repo and built this under x86-64 linux host installed with vivado & petalinux utility.
Could you pls explain more about "petalinux is pulling in the compiled files instead of the original source files?"
I'm not quite sure about it.
I'm facing this for the commit id ed4a26e checkout.
Thanks.

alexforencich, 2023-12-15:
Did you build any of the userspace tools before running the petalinux build?  I'm wondering if maybe you ran "make" and produced a libmqnic.a file for the host CPU, then petalinux just tried to use that instead of building it for ARM.  Another possibility is that I have screwed something up with how the static library build is set up; it's called via `$(MAKE) -C`, maybe there are additional arguments or such required for cross-compilation.

But, the last time I built the design with petalinux, it did build correctly.  I suspect whatever is going on might have something to do with your specific setup, but I'm not sufficiently familiar with petalinux, yocto, bitbake, etc. to even begin to debug it.

DJW146, 2023-12-15:
Thank you for your comment and guidance, Alex. 
I'm not sure if petalinux do it or not. i will check that carefully again.

I setup the ARCH, cross compilation variables for zynqMP only.
And I got to  ....../corundum/fpga/mqnic/ZCU106/fpga_zynqmp location and issued make -C ps/petalinux/ build-boot

After the error prompted, i also try to manually delete libmqnic.a and issue make -C ps/petalinux/ build-boot again to let the file compile to generate again. Unfortunately,  the error prompt symptom is the same.

alexforencich, 2023-12-15:
I just ran a build of the ZCU106 petalinux image for commit 1ca0151b97af85aa5dd306d74b6bcec65904d2ce with petalinux 2021.2 and it built successfully.  So, I really have no idea what might be going on here.  I will try again with the latest commit in my development repo, but nothing related to petalinux has changed in quite some time.

alexforencich, 2023-12-15:
Also it looks like you're using a really old commit.  Try it with the latest commit and see if that makes any difference.  I'm not going to provide free support for old commits.

DJW146, 2023-12-15:
Got it , thank you, alex !!
i will try the latest commit you mentioned with petalinux 2021.2 again if possible and the petalinux i use is 2022.1

DJW146, 2023-12-20:
after rollback to 2021.2 with https://github.com/corundum/corundum/commit/ed4a26e2cbc0a429c45d5cd5ddf1177f86838914 
there are no errors prompted and compilation to get all files. I will try it for the latest commit and 2022.1 as well.
Thanks again !!

closed 2023-12-20 by DJW146