corundum/corundum #180: AU50 working with DAC but not Optical

DragosFalk, 2023-12-21:
Hi
I am trying to bring up a Alveo U50 with corundum. Managed to get it running with a DAC but when I try to switch to optical transciever I can't get it to create a link . I have read in the documentation that it could be related to txdisable/lpmode but I can't find the right way to change that. I have tried via CMS but without any success.
Any one encountered this? Any suggestions are welcomed.
Thanks.

alexforencich, 2023-12-21:
What transceiver/AOC are you using, what rate do you want to run at, does it work in RX?

The BMC should turn on the transceiver by default.  But if you want to run a 25G module at 10G, you may need to turn off the CDRs. Unfortunately, the way the BMC is set up on the newer Alveo boards (U50, U280, U45, U55) makes this a bit of a pain and I have never tried to do it before.

DragosFalk, 2023-12-21:
I am using 100G one https://www.fs.com/de-en/products/48856.html?attribute=452&id=3439491 . On the other end I have a melanox connect x4 on which I tested and that supports it. Not sure how to test only RX. 

My company actually tried to get in contact with you . Not sure if you are willing to send some private contact info so that we could talk a bit more as we want to actually implement a few more things maybe we can cooperate if you are interested.

alexforencich, 2023-12-24:
Next thing to do is look at ethtool -m.  Are you seeing both nonzero TX bias and RX power on all lanes on both ends?  If so, that rules out problems with reset/lpmode.

My email address should be in all of the commits.  I have been really busy lately as the research project has been ramping up; hopefully after January I'll have more time to explore opportunities for collaboration.