corundum/corundum #184: Porting PCIe example to zu7cg

xiongyw, 2024-02-01:
Hi, Alex,

I am trying to port PCIe example project to [BOE board](https://github.com/hpb-project/BOE) v1.1, which is based on Xilinx zu7cg FPGA. It seems that the most similar board is ZCU106 (i.e., gen3x4 PCIe EP using GTH), so I use this as the starting point, by just chaning the fpga part number and related constraints (e.g., BOE board uses GTH bank 223 instead of 224 as in ZCU106). Bitstream builds successfully, and it can be downloaded to the board.

The issues are:
- The link status is not gen3x4, but gen2x4. The BOE board is plugged into a gen3x16 slot of x86 PC.
- The `example.ko` output shows that read/write/dma are not working.

I want to check that, for the test in `example.ko` to work properly, should the established link must be gen3x4?

Btw, one difference in BOE board is the lane assignment, which is bottom up (channel 0 for lane 0), not xilinx-suggested top-down order (channel 3 for lane 0). I am not sure if this is an issue given that the x4 link can be established (degraded to gen2 though).

Thanks!

alexforencich, 2024-02-01:
Strange.  The link speed should not affect whether or not the DMA engine works, but it's not clear why it wouldn't enumerate at the full speed, unless maybe there is some sort of signal integrity problem.  How does the device look under lspci - did it enumerate correctly?  If you only changed the part and the pin constraints, I'm not sure what else might need to be checked.

xiongyw, 2024-02-01:
Thanks for your reply, Alex.

`lspci`  looks like the following:

```
root@cmic-X99:/home/cmic# lspci -s 81:0.0 -vv
81:00.0 Memory controller: Device 1234:0001
        Subsystem: Xilinx Corporation Device 906a
        Physical Slot: 4
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        NUMA node: 1
        Region 0: Memory at fa000000 (64-bit, prefetchable) [virtual] [size=16M]
        Region 2: Memory at f9000000 (64-bit, prefetchable) [virtual] [size=16M]
        Region 4: Memory at fb000000 (64-bit, prefetchable) [virtual] [size=64K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
                Vector table: BAR=4 offset=00000000
                PBA: BAR=4 offset=00008000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ NonFatalErr+ FatalErr- UnsupReq+ AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x4, ASPM not supported
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s (downgraded), Width x4 (ok)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range BC, TimeoutDis+ NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [1c0 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: 0
        Kernel driver in use: edev
```

One related question is about the lane assignment order. If I open the package/device view after implementation, it shows that the lane order is not what I specified in xdc file. I am not sure how the link still worked: either the vivado display is wrong, or the PC link partner does lane reversal? I am quite confused.

[image]

xiongyw, 2024-02-01:
Btw, I commented out 8 DIP switches (in ZCU106), since there is no counterpart on BOE board. Vivado can still synth/impl with `sw[7:0]` not constrainted. I assume it does not affect the DMA behavior.

xiongyw, 2024-02-01:
I have some "progress", that:

- the link now is gen3x4 full, and,
- the driver test is almost ok.

I actually have two projects:

1. PCIe example integrated with a previous PS-only block design. Generating `BOOT.BIN` via Petalinux, and "cold" booting from TF card each time. This version of PCIe example was taken from VCU1525 with some manual changes.
2. ZCU106 version of PCIe example porting to BOE (i.e., without PS or any other PL logic), using the JTAG to "warmly" download bitstream.

I was working with the 1st project, which does not do full link (and with DMA test failure). So I tried the 2nd project to minimize the differences.

For some reasons the link status is now full gen3x4. I don't exactly know why the link changes from gen2x4 to gen3x4. I did some PC BIOS settings change wrt PCIe, but I believe I finally revert all thos settings back to the original. 

For the 1st project I did some changes (after comparing with ZCU106 example):

- AXIS width/freq configs changed from "256bit by 125MHz" to "128bit by 250MHz"
- Fixed reset signal: I was using a reset signal from PS. Now connecting to the one from golden finger.
- Fixed `fpga_core` parameter `AXIS_PCIE_DATA_WIDTH`, it was 512, now it's 128 (consistent with pcie config).

I am not sure any of the items above affects link training. In any case, now the link is full gen3x4 after cold TF boot. And after TF boot, if I program the device with bitstream from the 2nd project, the link is still gen3x4. The 2nd bitstream was only get gen2x4 before...

As for the driver test:

It seems that with TF boot, the DMA test is almost ok, while using the 2nd bitstream warmly, the test result is "worse".

Test result with cold TF boot (the 1st 16 bytes seems all zero, in DMA test):

```
[   43.450173] edev 0000:81:00.0: edev probe
[   43.450180] edev 0000:81:00.0:  Vendor: 0x1234
[   43.450184] edev 0000:81:00.0:  Device: 0x0001
[   43.450187] edev 0000:81:00.0:  Subsystem vendor: 0x10ee
[   43.450190] edev 0000:81:00.0:  Subsystem device: 0x903f
[   43.450193] edev 0000:81:00.0:  Class: 0x058000
[   43.450196] edev 0000:81:00.0:  PCI ID: 0000:81:00.0
[   43.450205] edev 0000:81:00.0:  Max payload size: 256 bytes
[   43.450209] edev 0000:81:00.0:  Max read request size: 512 bytes
[   43.450211] edev 0000:81:00.0:  Read completion boundary: 64 bytes
[   43.450214] edev 0000:81:00.0:  Link capability: gen 3 x4
[   43.450217] edev 0000:81:00.0:  Link status: gen 3 x4
[   43.450220] edev 0000:81:00.0:  Relaxed ordering: disabled
[   43.450222] edev 0000:81:00.0:  Phantom functions: disabled
[   43.450225] edev 0000:81:00.0:  Extended tags: enabled
[   43.450227] edev 0000:81:00.0:  No snoop: enabled
[   43.450229] edev 0000:81:00.0:  NUMA node: 1
[   43.450238] edev 0000:81:00.0: 31.504 Gb/s available PCIe bandwidth (8.0 GT/s PCIe x4 link)
[   43.450257] edev 0000:81:00.0: Allocated DMA region virt 000000007351a655, phys 0000000082d5654a
[   43.450270] edev 0000:81:00.0: enabling device (0100 -> 0102)
[   43.450291] edev 0000:81:00.0: ### enumerating 6 standard bars (physical addresses in host domain)
[   43.450293] edev 0000:81:00.0: BAR[0]: [0xfa000000, 0xfaffffff], flags 0x0014220c
[   43.450298] edev 0000:81:00.0: BAR[1]: null
[   43.450301] edev 0000:81:00.0: BAR[2]: [0xf9000000, 0xf9ffffff], flags 0x0014220c
[   43.450305] edev 0000:81:00.0: BAR[3]: null
[   43.450307] edev 0000:81:00.0: BAR[4]: [0xfb000000, 0xfb00ffff], flags 0x0014220c
[   43.450311] edev 0000:81:00.0: BAR[5]: null
[   43.450313] edev 0000:81:00.0: ### ioremap() all bars (to virtual addresses in host domain)
[   43.450522] edev 0000:81:00.0: BAR[0]: mapped at 0x0000000034cbce37 with length 16777216
[   43.450708] edev 0000:81:00.0: BAR[2]: mapped at 0x0000000017cab00d with length 16777216
[   43.450719] edev 0000:81:00.0: BAR[4]: mapped at 0x00000000ad310f17 with length 65536
[   43.451691] edev 0000:81:00.0: ### bar read/write test
[   43.451695] edev 0000:81:00.0: write 0x11223344 to 0000000017cab00d (BAR2)
[   43.451700] edev 0000:81:00.0: read from 0000000017cab00d (BAR2): 0x11223344
[   43.451704] edev 0000:81:00.0: ### dma test
[   43.451706] edev 0000:81:00.0: generate test data
[   43.451709] edev 0000:81:00.0: test data dump
[   43.451712] 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
[   43.451715] 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f  ................
[   43.451716] 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f   !"#$%&'()*+,-./
[   43.451718] 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f  0123456789:;<=>?
[... 134 lines cut ...]
[ 1846.178277] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178279] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178280] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178282] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178284] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178285] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178287] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178289] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178290] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178292] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1846.178294] edev 0000:81:00.0: test data mismatch
[ 1846.178296] edev 0000:81:00.0: start immediate write to host
[... cut ...]
[ 1846.190262] edev 0000:81:00.0: ffffffff
[ 1846.190265] edev 0000:81:00.0: read data
[ 1846.190267] 00 00 00 00                                      ....
[ 1846.190269] edev 0000:81:00.0: Read status
[ 1846.190271] edev 0000:81:00.0: ffffffff
```

I will dig a bit more on the DMA issue (for the 1st project). It would be great if you can give some hints on such error. 
I also have a qeustion regarding the BAR sizes: apparently the board does not have 16M+16M+64K RAM, where the BAR spaces mapped to in the FPGA?

Thanks again!

xiongyw, 2024-02-02:
It seems that strange DMA issue (first 16-bytes are all `00`) is due to vivado version: I was using vivado 2023.2 when the error presents. Today I switch to Vivado 2023.1 for generating the bitstream (still use petalinux 2023.2 to generate BOOT.BIN), the driver test is successful.

Btw, the reason I switch back to 2023.1, is because last night I also tested the PCIe example on another card (vu13p), it presents the similar problem that several lines of 16 byte data are `00`. But this vu13p card has a known good test result, using Vivado 2023.1 though.

Interesting...

xiongyw, 2024-02-08:
One small suggestion about the example driver code at [line 371](https://github.com/corundum/corundum/blob/master/fpga/lib/pcie/example/common/driver/example/example_driver.c#L371) that, it would be more accurate to change `phys` to `bus`, and more test-friendly to change `%p` to `%px`.

Thanks for the examples again. For me, it's a good start point for learning.

closed 2024-02-19 by xiongyw