corundum/corundum #189: mqnic is not compilent from modern kernel

ne-vlezay80, 2024-05-03:
Build log
``` 
 CC [M]  /tmp/corundum/modules/mqnic/mqnic_cq.o
  CC [M]  /tmp/corundum/modules/mqnic/mqnic_eq.o
  CC [M]  /tmp/corundum/modules/mqnic/mqnic_ethtool.o
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:615:21: error: initialization of ‘int (*)(struct net_device *, struct ethtool_rxfh_param *)’ from incompatible pointer type ‘int (*)(struct net_device *, u32 *, u8 *, u8 *)’ {aka ‘int (*)(struct net_device *, unsigned int *, unsigned char *, unsigned char *)’} [-Werror=incompatible-pointer-types]
  615 |         .get_rxfh = mqnic_get_rxfh,
      |                     ^~~~~~~~~~~~~~
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:615:21: note: (near initialization for ‘mqnic_ethtool_ops.get_rxfh’)
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:616:21: error: initialization of ‘int (*)(struct net_device *, struct ethtool_rxfh_param *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, const u32 *, const u8 *, const u8)’ {aka ‘int (*)(struct net_device *, const unsigned int *, const unsigned char *, const unsigned char)’} [-Werror=incompatible-pointer-types]
  616 |         .set_rxfh = mqnic_set_rxfh,
      |                     ^~~~~~~~~~~~~~
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:616:21: note: (near initialization for ‘mqnic_ethtool_ops.set_rxfh’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /tmp/corundum/modules/mqnic/mqnic_ethtool.o] Error 1
make[2]: *** [/usr/lib/modules/6.8.7-artix1-1/build/Makefile:1921: /tmp/corundum/modules/mqnic] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.8.7-artix1-1/build'
make: *** [Makefile:47: modules] Error 2
[user@archkde mqnic]$ make
make -C /lib/modules/6.8.7-artix1-1/build M=/tmp/corundum/modules/mqnic modules
make[1]: Entering directory '/usr/lib/modules/6.8.7-artix1-1/build'
  CC [M]  /tmp/corundum/modules/mqnic/mqnic_ethtool.o
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:615:21: error: initialization of ‘int (*)(struct net_device *, struct ethtool_rxfh_param *)’ from incompatible pointer type ‘int (*)(struct net_device *, u32 *, u8 *, u8 *)’ {aka ‘int (*)(struct net_device *, unsigned int *, unsigned char *, unsigned char *)’} [-Werror=incompatible-pointer-types]
  615 |         .get_rxfh = mqnic_get_rxfh,
      |                     ^~~~~~~~~~~~~~
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:615:21: note: (near initialization for ‘mqnic_ethtool_ops.get_rxfh’)
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:616:21: error: initialization of ‘int (*)(struct net_device *, struct ethtool_rxfh_param *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, const u32 *, const u8 *, const u8)’ {aka ‘int (*)(struct net_device *, const unsigned int *, const unsigned char *, const unsigned char)’} [-Werror=incompatible-pointer-types]
  616 |         .set_rxfh = mqnic_set_rxfh,
      |                     ^~~~~~~~~~~~~~
/tmp/corundum/modules/mqnic/mqnic_ethtool.c:616:21: note: (near initialization for ‘mqnic_ethtool_ops.set_rxfh’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /tmp/corundum/modules/mqnic/mqnic_ethtool.o] Error 1
make[2]: *** [/usr/lib/modules/6.8.7-artix1-1/build/Makefile:1921: /tmp/corundum/modules/mqnic] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.8.7-artix1-1/build'
make: *** [Makefile:47: modules] Error 2
```

alexforencich, 2024-05-03:
Looks like there were some changes in kernel 6.8.  I'll add some more ifdefs...

muvarov, 2024-12-05:
fix is in https://github.com/corundum/corundum/pull/225