corundum/corundum #191: Weird bugs meet with fb2CG and Vivado 2023.2

Yiyuan-Dong, 2024-05-14:
Recently, I encountered an issue while attempting to build Corundum with Vivado 2023.2 and deploying it onto an fb2CG@KU15P FPGA. The problem manifested in incorrect packet reception, where the first 62 bytes of incoming packets were not received accurately. For example, When I try to send a packet containing following content:

```
00 a0 01 a0 02 a0 03 a0 04 a0 05 a0 06 a0 07 a0
08 a0 09 a0 0a a0 0b a0 0c a0 0d a0 0e a0 0f a0
10 a0 11 a0 12 a0 13 a0 14 a0 15 a0 16 a0 17 a0
18 a0 19 a0 1a a0 1b a0 1c a0 1d a0 1e a0 1f a0
20 a0 21 a0 22 a0 23 a0 24 a0 25 a0 26 a0 27 a0
28 a0 29 a0 2a a0 2b a0 2c a0 2d a0 2e a0 2f a0
30 a0 31 a0 32 a0 33 a0 34 a0 35 a0 36 a0 37 a0
...
```

On the reception side I get the following content:

```
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 1f a0
20 a0 21 a0 22 a0 23 a0 24 a0 25 a0 26 a0 27 a0
28 a0 29 a0 2a a0 2b a0 2c a0 2d a0 2e a0 2f a0
30 a0 31 a0 32 a0 33 a0 34 a0 35 a0 36 a0 37 a0
38 a0 39 a0 3a a0 3b a0 3c a0 3d a0 3e a0 3f a0
40 a0 41 a0 42 a0 43 a0 44 a0 45 a0 46 a0 47 a0
...
```

The first 62 bytes does not correctly received.  Upon investigation, I traced the issue to a miscalculation within the `dma_if_pcie_rd` module, specifically in the determination of the `ram_wrap_next` value during the first cycle of a TLP reception. Let that be explained:

```verilog
  if (op_dword_count_next > TLP_DATA_WIDTH_DWORDS) begin
      // more than one cycle
      cycle_byte_count_next = TLP_DATA_WIDTH_BYTES-rx_cpl_tlp_hdr_lower_addr[1:0];  // == 7'h3e
      last_cycle = 1'b0;

      start_offset_next = addr_next;   // == 7'h00
      {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;  // ram_wrap_next == 1
  end
``` 

The erroneous calculation of `ram_wrap_nex` resulted in the first 62 bytes of incoming data being erroneously stored in the subsequent ram reg, leading to overlap with subsequent content and thus corrupting the packet.  And after I made the following change,  the FPGA NIC began functioning correctly, accurately processing incoming network packets.

```
--- a/fpga/lib/pcie/rtl/dma_if_pcie_rd.v
+++ b/fpga/lib/pcie/rtl/dma_if_pcie_rd.v
@@ -1018,14 +1018,14 @@ always @* begin
                         last_cycle = 1'b0;
 
                         start_offset_next = addr_next;
-                        {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
+                        // {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
                     end else begin
                         // one cycle
                         cycle_byte_count_next = op_count_next;
                         last_cycle = 1'b1;
 
                         start_offset_next = addr_next;
-                        {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
+                        // {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
                     end
                 end else begin
                     // last completion
@@ -1038,17 +1038,22 @@ always @* begin
                         last_cycle = 1'b0;
 
                         start_offset_next = addr_next;
-                        {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
+                        // {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
                     end else begin
                         // one cycle
                         cycle_byte_count_next = op_count_next;
                         last_cycle = 1'b1;
 
                         start_offset_next = addr_next;
-                        {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
+                        // {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
                     end
                 end
 
+                // A workaround for a Vivado bug (maybe). The original code would lead to incorrect `ram_wrap_next` value
+                // and thus is commented out
+                end_offset_next = start_offset_next + cycle_byte_count_next - 1;
+                ram_wrap_next = (RAM_DATA_WIDTH / 8 - start_offset_next < cycle_byte_count_next) ? 1'b1 : 1'b0;
+
                 ram_mask_0_next = {RAM_SEG_COUNT{1'b1}} << (start_offset_next >> $clog2(RAM_SEG_BE_WIDTH));
                 ram_mask_1_next = {RAM_SEG_COUNT{1'b1}} >> (RAM_SEG_COUNT-1-(end_offset_next >> $clog2(RAM_SEG_BE_WIDTH)));
 
@@ -1198,7 +1203,10 @@ always @* begin
                     last_cycle = 1'b1;
                 end
                 start_offset_next = addr_next;
-                {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
+                // {ram_wrap_next, end_offset_next} = start_offset_next+cycle_byte_count_next-1;
+
+                end_offset_next = start_offset_next + cycle_byte_count_next - 1;
+                ram_wrap_next = (RAM_DATA_WIDTH / 8 - start_offset_next < cycle_byte_count_next) ? 1'b1 : 1'b0;
```

As a newcomer to HDL development, the nature of this issue puzzled me, particularly the unexpected calculation of `ram_wrap_next` as 1, which looks like a complier bug. I am considering raising this issue on GitHub to help other users who may encounter similar challenges.

alexforencich, 2024-05-14:
Well.  This could certainly explain a few things.  I have been building mostly on 2021.2, and a bug in 2023.2 would certainly explain why some people have been running in to some strange issues.  Nice work!

MengEason, 2024-05-15:
Good job! The workaround has also solved my problem. #190

acostillado, 2024-05-15:
Good job Yiyuan-Dong, this solved my issue too. Thank you!

likewise, 2025-04-24:
For a newcomer to HDL, kudos for tracing the issue back.   I wonder was this issue ever reported to AMD/Xilinx?  We probably need a minimal piece of Verilog code demonstrating the bug at 2023.2 and check against newer versions.  It's creepy if the tooling can degrade like that.  This would make for a perfect regression unit test for AMD.