corundum/corundum #194: problem while building the driver

chenker-alt, 2024-05-30:
I've just got this problem while building the driver. So I'm using an Alveo U200 100g version, I've installed all the compiler packages and the kernel header, I ran the synthesis and implementation and generated the bitstream, but when I tried to build the driver, I got this issue. The kernel I'm using is Linux fpga 5.4.0-182-generic #202-Ubuntu SMP Fri Apr 26 12:29:36 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
[image]

JeffMY05, 2024-05-30:
I also got this problem which I believe is the kernel version mismatch. Which Linux kernel version should we upgrade our system to?

wegul, 2024-05-30:
Mine is 6.5.0 and no errors shown.

alexforencich, 2024-05-30:
(and yes, this is a version-related issue...but this sort of thing should be fixable with some ifdefs)

chenker-alt, 2024-05-31:
The devlink issue has been fixed, but now I have an issue with mqnic.ko: 
[image]

alexforencich, 2024-05-31:
Huh.  Did you do a make clean first?

chenker-alt, 2024-06-03:
Oh, I see the problem, it should be good now. Thanks!

closed 2024-06-03 by chenker-alt