corundum/corundum #196: RX part of fpga mqnic in reset after probing

vborchsh, 2024-07-03:
Dear colleagues,

I've been trying to run mqnic on the custom board. I use ZCU106 example since I'd like to connect mqnic cores to PS part.

So, I use the same setup as in example with two SFP modules. I run firmware, and rx_status is fine as well as other things. I see XGMII frame on the input of 10g mac core. But internal registers are still in reset. Particularly signal `rx_cfg_enable` is zero. This signal connected to `eth_rx_enable` signal from mqnic core.

So, I'm a bit confused how it should works. Probably, I'm missing part with modification of mqnic driver, but I'm no sure.

Shall I tie these enable signals to 1 as in some examples? And why mqnic driver doesn't load this registers during probing? Do I need to modify driver to implement this resets (the zcu106 examples has no any modification AFAIK)?

Thanks!

alexforencich, 2024-07-03:
The driver should enable the links when the link is brought up at the OS level.  Did you do "ip link set dev ethN up"?

vborchsh, 2024-07-04:
Yes, it does. But I do something wrong, I guess...

So, my steps are:

1. Build Linux & mqnic.ko as module in Buildroot
2. Build FPGA project in Vivado 2021.2.1
3. Load Linux
4. Load firmware
5. Run `insmod /lib/modules/5.10.0-rt17-xilinx-v2021.2-rt/extra/mqnic.ko`.
<details>
  <summary># dmesg | grep mqnic </summary>

  ```
# dmesg | grep mqnic                                                                                                                                                                                                                                                           
[ 1428.909250] mqnic: loading out-of-tree module taints kernel.                                                                                                                                                                                                                
[ 1428.913260] mqnic a0000000.ethernet: mqnic platform probe                                                                                                                                                                                                                   
[ 1428.913297] mqnic a0000000.ethernet: Resetting device                                                                                                                                                                                                                       
[ 1428.913344] mqnic a0000000.ethernet: Control BAR size: 16777216                                                                                                                                                                                                             
[ 1428.913397] mqnic a0000000.ethernet: Application BAR size: 16777216                                                                                                                                                                                                         
[ 1428.914895] mqnic a0000000.ethernet: Configured 4 IRQs                                                                                                                                                                                                                      
[ 1428.914946] mqnic a0000000.ethernet: Device-level register blocks:                                                                                                                                                                                                          
[ 1428.914957] mqnic a0000000.ethernet:  type 0xffffffff (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.914973] mqnic a0000000.ethernet:  type 0x0000c007 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.914988] mqnic a0000000.ethernet:  type 0x0000c000 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.915003] mqnic a0000000.ethernet:  type 0x00000000 (v 0.0.0.0)                                                                                                                                                                                                           
[ 1428.915017] mqnic a0000000.ethernet:  type 0x0000c006 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.915032] mqnic a0000000.ethernet:  type 0x0000c080 (v 0.0.2.0)                                                                                                                                                                                                           
[ 1428.915046] mqnic a0000000.ethernet:  type 0x0000c081 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.915061] mqnic a0000000.ethernet:  type 0x0000c008 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.915075] mqnic a0000000.ethernet:  type 0x0000c101 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.915090] mqnic a0000000.ethernet:  type 0x0000c150 (v 0.0.1.0)                                                                                                                                                                                                           
[ 1428.915107] mqnic a0000000.ethernet: FPGA ID: 0x04730093                                                                                                                                                                                                                    
[... cut ...]
Sched channel stride: 4
Sched control: 0
Sched dest: 0
EQ info
 Queue      Base Address     En A  LS  A   IRQ    Prod    Cons     Len
CQ info
 Queue      Base Address     En A  LS  A   EQN    Prod    Cons     Len
TXQ info
  Queue      Base Address     En A  B  LS   CQN    Prod    Cons     Len
RXQ info
  Queue      Base Address     En A  B  LS   CQN    Prod    Cons     Len
Statistics counters
```
</details>

Here are `Port TX ctrl: 0x00010001` and `Port RX ctrl: 0x03010000` says that RX is in reset, right?

vborchsh, 2024-07-04:
Just for the record:
[image]

`tx_enable` is 1, but `rx_enable` is 0 after `ip link set dev eth1 up`

vborchsh, 2024-07-04:
I tried to dig a bit deeper and here my ideas:

According to [ `mqnic_start_port`](https://github.com/corundum/corundum/blob/master/modules/mqnic/mqnic_netdev.c) there is a straight way to start the link. And since I see tx part enabled means I at least [at this line](https://github.com/corundum/corundum/blob/master/modules/mqnic/mqnic_netdev.c#L162).

Then, there are several steps which finish by calling `mqnic_port_set_rx_ctrl(priv->port, MQNIC_PORT_RX_CTRL_EN)`. And this function doesn't run properly, if rx part still in the reset.

The steps are:

```
	**mqnic_port_set_tx_ctrl(priv->port, MQNIC_PORT_TX_CTRL_EN);**

	// enable scheduler
	mqnic_activate_sched_block(priv->sched_block);

	netif_tx_start_all_queues(ndev);
	netif_device_attach(ndev);

	if (mqnic_link_status_poll) {
		priv->link_status = 0;
		mod_timer(&priv->link_status_timer,
				jiffies + msecs_to_jiffies(mqnic_link_status_poll));
	} else {
		netif_carrier_on(ndev);
	}

	**mqnic_port_set_rx_ctrl(priv->port, MQNIC_PORT_RX_CTRL_EN);**
```
None of these function are able to jump to `fail` point or something else. And I suppose that `link_status` is fine, because I see `rx_status` in ILA **and** in the mqnic-dump util (`Port RX ctrl: 0x03010000`, bit 16 is 1)

Is it possible that one of these functions get stuck? Or maybe there are other ideas to check?

Thanks

alexforencich, 2024-07-04:
Yeah, that really makes no sense.  I was thinking maybe the RX clock wasn't running, that might explain why the RX enable output is zero despite the driver setting it high, but that shouldn't prevent the register from reading back as zero.

By the way, what version of Vivado are you using?  Some issues have been reported with 2023.2: https://github.com/corundum/corundum/issues/191 .  That affects PCIe, but ostensibly there could be other problems with 2023.2.

vborchsh, 2024-07-04:
> Yeah, that really makes no sense. I was thinking maybe the RX clock wasn't running, that might explain why the RX enable output is zero despite the driver setting it high, but that shouldn't prevent the register from reading back as zero.

Agree. So, maybe there is something else I can check?

> By the way, what version of Vivado are you using?

I use 2021.2.1. First, I tried 2023.2.1 as my main, but after close look to mentioned issue, I moved back to 2021.2.1.

alexforencich, 2024-07-04:
I mean, next thing I would do is add printks around the register write and look at the AXI lite interface on the ILA to find the register write.  I have no clue why the RX enable bit is reading back as zero after it was supposed to be set to 1.  It worked fine the last time I tested it on the ZCU106....

vborchsh, 2024-07-04:
> is add printks around the register write

Yes, already do... Unfortunately, I have no ZCU106 around(

vborchsh, 2024-07-04:
Ok, here is dump of completed functions:

```
# ip link set dev eth1 up
[   74.328377] mqnic a0000000.ethernet eth1: mqnic_start_port on interface 0 netdev 0
# [   74.367841] mqnic a0000000.ethernet eth1: MQNIC_PORT_TX_CTRL_EN on interface 0 netdev 0
[   74.367871] mqnic a0000000.ethernet eth1: priv->sched_block on interface 0 netdev 0
[   74.367893] mqnic a0000000.ethernet eth1: netif_tx_start_all_queues on interface 0 netdev 0
[   74.367906] mqnic a0000000.ethernet eth1: netif_device_attach on interface 0 netdev 0
[   74.367924] mqnic a0000000.ethernet eth1: if (mqnic_link_status_poll) on interface 0 netdev 0
[   74.367937] mqnic a0000000.ethernet eth1: mqnic_port_set_rx_ctrl on interface 0 netdev 0
```

As I see, the function successfully goes through all steps and everything fine. But rx_enable is still 0. I'm gonna do more debug in FPGA.

vborchsh, 2024-07-05:
Here is an ILA dump of AXI lite csr interface. For some reason, there are only 0-word on every RX_CTRL address:

[image]

The sequence is: `insmod mqnic.ko` and then `ip link up`. So, there are 3 transaction to the RX_CTRL and all of them all zeros.

vborchsh, 2024-07-05:
Here we go!

[image]

The issue is AXI interconnect. Input is 128 bits AXIMM, output is 32 bits AXIL. And there is missing data after conversion. Lovely...

vborchsh, 2024-07-05:
Ok, now it works. I still see no output from from mqnic fpga core (tx part), but this is the other story. So, I'm gonna close this issue.

Thank you Alex for explanations!

closed 2024-07-05 by vborchsh

alexforencich, 2024-07-05:
Huh.  How did it read as nonzero, then???

vborchsh, 2024-07-05:
No idea. Timings are Ok. And reads are fine. But write transaction had fallen. I've changed bus width from 128 to 32 in Zynq setup, and now it works. Btw, in zcu106 example you have 32 bits.

vborchsh, 2024-07-05:
Just for the record. It's alive:

Client (PC):
```
$ iperf -c 192.168.0.33
------------------------------------------------------------
Client connecting to 192.168.0.33, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  1] local 192.168.0.103 port 54776 connected with 192.168.0.33 port 5001
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-10.0793 sec   787 MBytes   655 Mbits/sec
```

Server (MPSoC):
```
# iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  1] local 192.168.0.33 port 5001 connected with 192.168.0.103 port 54776 (icwnd/mss/irtt=14/1448/190)
[ ID] Interval       Transfer     Bandwidth
[  1] 0.00-10.07 sec   787 MBytes   655 Mbits/sec
```

And vice versa:

Client (MPSoC):
```
# iperf -c 192.168.0.103
------------------------------------------------------------
Client connecting to 192.168.0.103, TCP port 5001
TCP window size: 45.0 KByte (default)
------------------------------------------------------------
[  1] local 192.168.0.33 port 35732 connected with 192.168.0.103 port 5001 (icwnd/mss/irtt=14/1448/228)
[ ID] Interval       Transfer     Bandwidth
[  1] 0.00-10.01 sec  1.10 GBytes   941 Mbits/sec
```

Server (PC

```
$ iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  1] local 192.168.0.103 port 5001 connected with 192.168.0.33 port 35732
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-10.0091 sec  1.10 GBytes   940 Mbits/sec
```