corundum/corundum #198: Random ping loss in non-phy zynqmp implementation

vborchsh, 2024-07-16:
Hello all,

I faced tricky issue. I use corundum with ZynqMP. As mentioned [here](https://github.com/corundum/corundum/issues/196) I've bring up the interface and see ping and iperf over 10G phy. My next step is non-phy implementation in FPGA logic. Right now I've been trying to run two interface inside FPGA, without physical layer, so its just two AXI streams of mqnic_axi core connected each other.

And I see weird things. For the instance, I run several ping packets from eth2 to eth1, which are separated to different namespaces. And I control packets in the FPGA logic by ILA. And some of these pings are ok, but some are missing.

Here are two dumps for the same command `ip netns exec port1 ping <ip0> <ip1> -c 1`, with only one packet:

1. "good" ping
[image]

2. "bad" ping, response packet is missed (second window)
[image]

In "bad" case I see missing write transaction for the response packet. And indeed, tcpdump shows (pseudo) the next thing.

I do ping eth2 -> eth1 with 1 packet. Ping lost.

tcpdump -I eth2:
send request

tcpdump -I eth1:
got request
send response

In "good" case both interfaces show request and response correctly. I'll come back with more details and dumps if you'd like so.

What it could be?

[ilas.zip](https://github.com/user-attachments/files/16247967/ilas.zip)

vborchsh, 2024-07-16:
Let me try to make it a bit clearer by this picture:

[image]

Red box is missed in case of missing ping.

alexforencich, 2024-07-16:
If you send more than 1 ping packet are you still seeing this issue?  I have had at least one report of the first packet getting "stuck" somewhere until another packet comes along, but haven't yet had a chance to dig in to it further.

vborchsh, 2024-07-17:
> If you send more than 1 ping packet are you still seeing this issue?

Yes, I do. Example for 4 packets:

```
# ping 10.0.0.10 -c 4
PING 10.0.0.10 (10.0.0.10): 56 data bytes
64 bytes from 10.0.0.10: seq=2 ttl=64 time=0.247 ms

--- 10.0.0.10 ping statistics ---
4 packets transmitted, 1 packets received, 75% packet loss
round-trip min/avg/max = 0.247/0.247/0.247 ms
```

Plus, the issue is random: every time random number of packet loss..

vborchsh, 2024-07-17:
One interesting thing. When I run ping on both interfaces simultaneously no losses presents. But when I stop one of them - it starts losing the packets.

Example:

eth1
```
# ping 10.0.0.20
PING 10.0.0.20 (10.0.0.20): 56 data bytes

64 bytes from 10.0.0.20: seq=0 ttl=64 time=0.734 ms
64 bytes from 10.0.0.20: seq=5 ttl=64 time=0.480 ms
64 bytes from 10.0.0.20: seq=6 ttl=64 time=0.448 ms
64 bytes from 10.0.0.20: seq=7 ttl=64 time=0.452 ms
64 bytes from 10.0.0.20: seq=8 ttl=64 time=0.440 ms
64 bytes from 10.0.0.20: seq=9 ttl=64 time=0.452 ms
64 bytes from 10.0.0.20: seq=10 ttl=64 time=0.453 ms 
64 bytes from 10.0.0.20: seq=11 ttl=64 time=0.267 ms
^C
--- 10.0.0.20 ping statistics --- 
12 packets transmitted, 8 packets received, 33% packet loss
round-trip min/avg/max = 0.267/0.465/0.734 ms
```

eth2
```
# ping 10.0.0.10
PING 10.0.0.10 (10.0.0.10): 56 data bytes
64 bytes from 10.0.0.10: seq=0 ttl=64 time=0.674 ms
64 bytes from 10.0.0.10: seq=1 ttl=64 time=0.495 ms
64 bytes from 10.0.0.10: seq=2 ttl=64 time=0.438 ms
64 bytes from 10.0.0.10: seq=3 ttl=64 time=0.486 ms
64 bytes from 10.0.0.10: seq=4 ttl=64 time=0.494 ms
64 bytes from 10.0.0.10: seq=5 ttl=64 time=0.491 ms
64 bytes from 10.0.0.10: seq=6 ttl=64 time=0.474 ms
^C
--- 10.0.0.10 ping statistics ---
7 packets transmitted, 7 packets received, 0% packet loss
round-trip min/avg/max = 0.438/0.507/0.674 ms
```

So, here I started eth1 to eth2 ping and you can see that first 4 packets are missed (seq 1..4, except zero). But then I started ping at the second interface eth2 to eth1. And here we go: ping in both directions works with no losses. Until I stop one of them.

vborchsh, 2024-07-17:
A bit of debugging:

Good case:

AXI MM
```
rd 000006b5c3c40, 000000080654ba020000006200000000 00000000000000000000000000000000 00000000000000000000000000000000 00000000000000000000000000000000
rd 000080654ba02, <data packet, 6 words>
packet to the interface
wr 000006b5a9e20, 00000000000000000000006204f1000b 00000000000000000000000000000000
rd 000006b15cf10, 00000008073720000000100000000000
wr 000006b149e00, 000000000000000000000000000f0000 00000000000000000000000000000000
wr 0000807372000, <data packet, 6 words>
three more writes
```

Bad case:
```
rd 000006b294c40, 00000008002802020000006200000000 00000000000000000000000000000000 00000000000000000000000000000000 00000000000000000000000000000000
rd 0000800280202, <data packet, 6 words>
packet to the interface
wr 000006b27a620, 0000000000000000000000620531000b 00000000000000000000000000000000
! no read here
wr 000006b12a5a0, 000000000000000000000000000f0000 00000000000000000000000000000000
! no three writes here
```

In both cases transmitting (reading from DMA, first two transactions) works with no issues. During "bad" receiving I see missing AXI transactions. The most interesting one: missing reading with memory offset in CPU RAM for the packet.

As far as I understood, there is no DMA read transaction which provides memory offset for received packet, for some reasons.

closed 2024-07-17 by vborchsh