corundum/corundum #199: Timing violation issue in dma_if_pcie_rd.v

hmz2015011117, 2024-07-19:
Hello,

I am currently porting the open-source project Corundum to the Xilinx KCU105 platform and am facing some issues. Yesterday, I deployed the project on my personal computer (Intel i5 12400 CPU with 64GB RAM), and encountered some minor timing violations as shown in the first image. These violations seem to be due to excessive logic levels, but they appear to be not very severe, only around 0.0x ns. Today, I attempted the deployment on my work computer (Intel i5 12500 CPU with 16GB RAM). The process was initially interrupted several times due to insufficient memory allocated by VMware to Ubuntu. After increasing the memory allocation to 16GB, the bitstream was successfully generated, but as shown in the second image, the logic levels increased even further, reaching 10 levels. 

I haven't changed anything！

I would like to understand if these timing violations are significantly influenced by the system's memory, and why. What other factors could cause these timing issues? Given that this is an open-source project, it should presumably not have these issues. How should I go about resolving them?

Thank you for your assistance.
[image]
[image]
[image]
[image]
[image]
[image]

alexforencich, 2024-07-19:
Complex designs with fast clocks can have these sorts of issues, especially on smaller and slower parts.  You can try selecting the extra timing opt strategy.  The whole place and route process is chaotic, seemingly inconsequential changes can have a large impact on the result (in this case, the build time set by config.tcl will cause the design to build differently).  And if Vivado starts with a bad placement, then the timing can fail miserably.  If you can't get it to close timing reliably, you can also try dropping down to PCIe gen 2 instead of gen 3, which will halve the core clock frequency.  Although if you do that, you'll also need to change the datapath width to 128 bits to hit line rate since 125 is less than 156.25.

hmz2015011117, 2024-07-19:
emm, how should the optimization strategy be adjusted? Are there any specific reference examples or documents? I have just been working on your project for two weeks. I roughly understand that the strategy is designed through tcl scripts. I have downloaded the vivado-related tcl manual. Is the next step to read the manual and then learn some tcl commands to modify the config.tcl file? In addition, is it possible that I can pass the timing by changing the vivado version?

hmz2015011117, 2024-07-19:
Thank you very much for your response. I have successfully modified the placement and routing strategy, and after testing, the Congestion_SpreadLogic_low strategy passes the static timing analysis. I will try the other suggestions you provided later. Thank you once again.
[image]

However, there are now the following warnings. Do these warnings need to be addressed? Could you please provide some guidance on this?

[image]
3-9f01-0e38fea64508)

PoojaR07, 2024-08-02:
Can you please share the steps how you deployed the project on the Xilinx platform as we are facing issues with that.

hmz2015011117, 2024-08-02:
Modified the placement and routing strategy, and after testing, the Congestion_SpreadLogic_low strategy passes the static timing analysis.
[image]
[image]
Hope this can helps you!

PoojaR07, 2024-08-02:
Thank you for the help.
But I want the steps how you deployed corrundum on vivado , we followed the steps on readme but the terminal keeps crashing.
Can you give the entire flow of steps from the start if possible?

hmz2015011117, 2024-08-02:
I wrote all the procedures in Chinese, I don't know if you can understand it
[corundum项目验证环境搭建.pdf](https://github.com/user-attachments/files/16466831/corundum.pdf)

PoojaR07, 2024-08-05:
Thank you so much for the Document !!

hmz2015011117, 2024-08-08:
You're welcome. If you have any questions, you can send them to my email address for discussion: 2206320281@qq.com