corundum/corundum #203: About the checksum problem

hmz2015011117, 2024-08-20:
For the rx_checksum.v file, the functions it implements are not completely consistent with the checksum functions required by the official standard（Mainly the calculation of pseudo-head）. The following figure shows the official standard:
[image]

Why don't we calculate in full accordance with the official standards? What are the advantages of doing so? What are the problems?

alexforencich, 2024-08-20:
Corundum implements the standard Linux receive checksum offloading method of computing the checksum of everything after the Ethernet header. This is protocol-independent as software can simply back out headers and fix up the checksum as appropriate, without having to compute the checksum over the entire payload.

hmz2015011117, 2024-08-20:
Is the advantage of this that the hardware can remain protocol-independent and simplified, while the software can flexibly modify the checksum according to different protocols?

hmz2015011117, 2024-08-20:
Thank you very much for your answer, I found the relevant information in skbuff.h in linux:
 * - %CHECKSUM_COMPLETE
 *
 *   This is the most generic way. The device supplied checksum of the _whole_
 *   packet as seen by netif_rx() and fills in &sk_buff.csum. This means the
 *   hardware doesn't need to parse L3/L4 headers to implement this.
 *
 *   Notes:
 *
 *   - Even if device supports only some protocols, but is able to produce
 *     skb->csum, it MUST use CHECKSUM_COMPLETE, not CHECKSUM_UNNECESSARY.
 *   - CHECKSUM_COMPLETE is not applicable to SCTP and FCoE protocols.
 Thanks again for your guidance, I will continue to learn.

alexforencich, 2024-08-20:
Yep, you got it.  I was forgetting exactly where that was documented.  And yes, it's basically a nice compromise of simple hardware and simple software - software doesn't have to crank through the whole packet, only a few header fields, and hardware doesn't have to be aware of the protocol in use (which not only means the hardware is simpler, it also means that it isn't limited by what protocol variations/combinations that the hardware happens to support, which can be a problem with all of the various tunneling protocols and encapsulation methods).