corundum/corundum #204: insmod mqnic.ko & port control register block not found

comapny99, 2024-08-21:
Because linux kernal 6.8 has problems when making mqnic, I changed ubuntu from

linux-image-generic-hwe-22.04:amd64 22.04.1 6.8.0-40.40~22.04.3
linux-headers-generic-hwe-22.04:amd64 6.8.0-40.40~22.04.3
busybox-initramfs:amd64 1:1.30.1-7ubuntu3 1:1.30.1-7ubuntu3.1
busybox-static:amd64 1:1.30.1-7ubuntu3 1:1.30.1-7ubuntu3.1

degraded to

linux-generic-hwe-22.04:amd64 6.5.0.45.45~22.04.1
linux-image-generic-hwe-22.04:amd64 6.5.0.45.45~22.04.1
linux-headers-generic-hwe-22.04:amd64 6.5.0.45.45~22.04.1
busybox-initramfs:amd64 1:1.30.1-7ubuntu3 1:1.30.1-7ubuntu3
busybox-static:amd64 1:1.30.1-7ubuntu3 1:1.30.1-7ubuntu3

But when setting the network port, the mqnic device cannot be found. Use dmesg to observe the following:
[    5.779858] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
[    6.437020] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.437022] Bluetooth: BNEP filters: protocol multicast
[    6.437024] Bluetooth: BNEP socket layer initialized
[    6.472930] loop25: detected capacity change from 0 to 8
[    6.883206] Bluetooth: hci0: Waiting for firmware download to complete
[    6.883210] Bluetooth: hci0: Firmware loaded in 1345359 usecs
[    6.883236] Bluetooth: hci0: Waiting for device to boot
[    6.899165] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[    6.899221] Bluetooth: hci0: Device booted in 15624 usecs
[    6.899465] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-1040-0041.ddc
[    6.901225] Bluetooth: hci0: Applying Intel DDC parameters completed
[    6.904250] Bluetooth: hci0: Firmware timestamp 2023.24 buildtype 1 build 67068
[    6.974267] Bluetooth: MGMT ver 1.22
[    6.976496] NET: Registered PF_ALG protocol family
[    7.025884] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
[    7.185421] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x20
[    7.185449] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    7.185456] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[    7.185465] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x0
[    7.186652] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 5
[    7.276556] iwlwifi 0000:00:14.3: Registered PHC clock: iwlwifi-PTP, with index: 1

[   87.453098] mqnic 0000:05:00.0: mqnic PCI remove
[  102.040712] mqnic 0000:05:00.0: mqnic PCI probe
[  102.040715] mqnic 0000:05:00.0:  Vendor: 0x1234
[  102.040715] mqnic 0000:05:00.0:  Device: 0x1001
[  102.040716] mqnic 0000:05:00.0:  Subsystem vendor: 0x10ee
[  102.040716] mqnic 0000:05:00.0:  Subsystem device: 0x906a
[  102.040717] mqnic 0000:05:00.0:  Class: 0x020000
[  102.040717] mqnic 0000:05:00.0:  PCI ID: 0000:05:00.0
[  102.040730] mqnic 0000:05:00.0:  Max payload size: 256 bytes
[  102.040730] mqnic 0000:05:00.0:  Max read request size: 512 bytes
[  102.040731] mqnic 0000:05:00.0:  Read completion boundary: 64 bytes
[  102.040731] mqnic 0000:05:00.0:  Link capability: gen 3 x4
[  102.040732] mqnic 0000:05:00.0:  Link status: gen 3 x4
[  102.040732] mqnic 0000:05:00.0:  Relaxed ordering: enabled
[  102.040732] mqnic 0000:05:00.0:  Phantom functions: disabled
[  102.040733] mqnic 0000:05:00.0:  Extended tags: enabled
[  102.040733] mqnic 0000:05:00.0:  No snoop: enabled
[  102.040734] mqnic 0000:05:00.0:  NUMA node: -1
[  102.040734] mqnic 0000:05:00.0:  PCI ID (bridge): 0000:00:1c.4
[  102.040737] mqnic 0000:05:00.0:  Link capability (bridge): gen 4 x4
[  102.040737] mqnic 0000:05:00.0:  Link status (bridge): gen 3 x4
[  102.040744] mqnic 0000:05:00.0: 31.504 Gb/s available PCIe bandwidth (8.0 GT/s PCIe x4 link)
[  102.040746] mqnic 0000:05:00.0: can't disable ASPM; OS doesn't have ASPM control
[  102.040756] mqnic 0000:05:00.0: Control BAR size: 16777216
[  102.040767] mqnic 0000:05:00.0: Application BAR size: 16777216
[  102.041677] mqnic 0000:05:00.0: Configured 64 IRQs
[  102.041726] mqnic 0000:05:00.0: Device-level register blocks:
[  102.041727] mqnic 0000:05:00.0:  type 0xffffffff (v 0.0.1.0)
[  102.041728] mqnic 0000:05:00.0:  type 0x0000c007 (v 0.0.1.0)
[  102.041728] mqnic 0000:05:00.0:  type 0x0000c000 (v 0.0.1.0)
[  102.041729] mqnic 0000:05:00.0:  type 0x0000c005 (v 0.0.2.0)
[  102.041729] mqnic 0000:05:00.0:  type 0x0000c006 (v 0.0.1.0)
[  102.041730] mqnic 0000:05:00.0:  type 0x0000c080 (v 0.0.1.0)
[  102.041730] mqnic 0000:05:00.0:  type 0x0000c081 (v 0.0.1.0)
[  102.041731] mqnic 0000:05:00.0:  type 0x0000c008 (v 0.0.1.0)
[  102.041731] mqnic 0000:05:00.0:  type 0x0000c110 (v 0.0.1.0)
[  102.041732] mqnic 0000:05:00.0:  type 0x0000c101 (v 0.0.1.0)
[  102.041733] mqnic 0000:05:00.0:  type 0x0000c150 (v 0.0.1.0)
[  102.041743] mqnic 0000:05:00.0: FPGA ID: 0x04730093
[  102.041743] mqnic 0000:05:00.0: FW ID: 0x00000000
[  102.041744] mqnic 0000:05:00.0: FW version: 0.0.1.0
[  102.041744] mqnic 0000:05:00.0: Board ID: 0x10ee906a
[  102.041745] mqnic 0000:05:00.0: Board version: 1.0.0.0
[  102.041746] mqnic 0000:05:00.0: Build date: 2024-08-15 06:35:20 UTC (raw: 0x66bda1a8)
[  102.041747] mqnic 0000:05:00.0: Git hash: ed4a26e2
[  102.041747] mqnic 0000:05:00.0: Release info: 00000000
[  102.041749] mqnic 0000:05:00.0: Application ID: 0x12340001
[  102.041763] mqnic 0000:05:00.0: IF offset: 0x00000000
[  102.041763] mqnic 0000:05:00.0: IF count: 2
[  102.041764] mqnic 0000:05:00.0: IF stride: 0x00800000
[  102.041764] mqnic 0000:05:00.0: IF CSR offset: 0x00040000

[  102.173843] mqnic 0000:05:00.0: Creating interface 0
[  102.173925] mqnic 0000:05:00.0: Interface-level register blocks:
[  102.173926] mqnic 0000:05:00.0:  type 0x0000c001 (v 0.0.4.0)
[  102.173927] mqnic 0000:05:00.0:  type 0x0000c010 (v 0.0.4.0)
[  102.173928] mqnic 0000:05:00.0:  type 0x0000c020 (v 0.0.4.0)
[  102.173929] mqnic 0000:05:00.0:  type 0x0000c030 (v 0.0.4.0)
[  102.173929] mqnic 0000:05:00.0:  type 0x0000c031 (v 0.0.4.0)
[  102.173930] mqnic 0000:05:00.0:  type 0x0000c090 (v 0.0.2.0)
[  102.173930] mqnic 0000:05:00.0:  type 0x0000c002 (v 0.0.2.0)
[  102.173931] mqnic 0000:05:00.0:  type 0x0000c004 (v 0.0.3.0)
[  102.173946] mqnic 0000:05:00.0: IF features: 0x00000711
[  102.173946] mqnic 0000:05:00.0: Port count: 1
[  102.173947] mqnic 0000:05:00.0: Scheduler block count: 1
[  102.173947] mqnic 0000:05:00.0: Max TX MTU: 9214
[  102.173948] mqnic 0000:05:00.0: Max RX MTU: 9214
[  102.173952] mqnic 0000:05:00.0: EQ offset: 0x000c0000
[  102.173952] mqnic 0000:05:00.0: EQ count: 64
[  102.173953] mqnic 0000:05:00.0: EQ stride: 0x00000010
[  102.173957] mqnic 0000:05:00.0: CQ offset: 0x00100000
[  102.173958] mqnic 0000:05:00.0: CQ count: 16384
[  102.173958] mqnic 0000:05:00.0: CQ stride: 0x00000010
[  102.173963] mqnic 0000:05:00.0: TXQ offset: 0x00200000
[  102.173963] mqnic 0000:05:00.0: TXQ count: 8192
[  102.173963] mqnic 0000:05:00.0: TXQ stride: 0x00000020
[  102.173969] mqnic 0000:05:00.0: RXQ offset: 0x00300000
[  102.173969] mqnic 0000:05:00.0: RXQ count: 256
[  102.173969] mqnic 0000:05:00.0: RXQ stride: 0x00000020
[  102.173971] mqnic 0000:05:00.0: RX queue map indirection table size: 256
[  102.173974] mqnic 0000:05:00.0: Max desc block size: 8
[  102.175706] mqnic 0000:05:00.0: Port-level register blocks:
[  102.175707] mqnic 0000:05:00.0:  type 0x0000c003 (v 0.0.2.0)
[  102.175707] mqnic 0000:05:00.0: Port control register block not found
[  102.175711] mqnic 0000:05:00.0: Failed to create interface: 0
[  102.175846] mqnic 0000:05:00.0: Registered device mqnic0
[  102.175860] mqnic 0000:05:00.0: Registered auxiliary bus device mqnic.app_12340001.0

It looks like the problem is as follows
[102.175711] mqnic 0000:05:00.0: Failed to create interface: 0
[102.175846] mqnic 0000:05:00.0: Registered device mqnic0
I tried changing the KERNAL version, ex: 6.4, 6.3, but it still doesn’t work. Could you please tell me how to debug and solve this problem?  Thank you so much

By the way, do you have any suggestions for a newer and stable ubuntu kernal version? I would like to say that the recommended compatible ubuntu version was not found in the file. Thank you

alexforencich, 2024-08-21:
some register block not found usually means the HDL and driver were not built from the same git revision and there is a version mismatch in the register blocks

I try to keep the kernel module updated to work with the latest kernel release, but it is unfortunately a moving target, and sometimes things break with older kernels.  I would appreciate it if people would report build issues with newer kernels so I can add ifdefs as necessary.

comapny99, 2024-08-21:
Thank you for your reply. The reason why I want to downgrade the following packages is because I could still use them normally the day before, but the next day after ubuntu automatically updated the following 5 items for me, I couldn’t use them. So I want to downgrade them to the previous day. version and deleted the new version of the package, but it is still unavailable.

linux-generic-hwe-22.04:amd64 6.5.0.45.45~22.04.1
linux-image-generic-hwe-22.04:amd64 6.5.0.45.45~22.04.1
linux-headers-generic-hwe-22.04:amd64 6.5.0.45.45~22.04.1
busybox-initramfs:amd64 1:1.30.1-7ubuntu3 1:1.30.1-7ubuntu3
busybox-static:amd64 1:1.30.1-7ubuntu3 1:1.30.1-7ubuntu3

comapny99, 2024-08-23:
Hi @alexforencich 
I later solved this problem:
1. I downgraded kernal to 6.4.0 and deleted both 6.5 and 6.8 by mainline.
2. Download the github code (corundum Dec 2, 2023) and recompile the bitstream and driver under kernel 6.4.0
Thank you very much.

sivanasheesh, 2024-11-26:
HI @alexforencich, I'm working on the CORUNDUM project, for ZCU106 while I was booting it is showing "FW ID block not found" in dmesg. Can you please tell what the issue might be, Thank you