corundum/corundum #207: Error about mqnic_core_pcie_us emulation

hmz2015011117, 2024-09-02:
Hello Alex, I have another problem that I can't solve.
When I simulated mqnic_core_pcie_us using modelsim（Simulation with iverilog is successful）, an error like the following appeared:
[image]
The specific information is shown in output1.txt:
[output1.txt](https://github.com/user-attachments/files/16835681/output1.txt)
The content of my Makefile is as follows:
[Makefile.txt](https://github.com/user-attachments/files/16835712/Makefile.txt)
I would like to know what caused the error and how to solve it.

wangyajun001, 2024-10-08:
Hello, I’m facing the same issue as you and have been unable to find a solution. Have you managed to resolve this problem?

hmz2015011117, 2024-10-08:
At present, there is no solution. The solution should be to write Python programs for modelsim or vcs, but these two software are not open source, so it is difficult. At present, I still use iverilog and gtkwave for simulation learning.

alexforencich, 2024-10-08:
I would need a waveform dump to figure out what's going on here.  Presumably this is some sort of difference in how the simulator interprets the HDL, but the underlying cause will probably take some time to sort out, and I don't have access to modelsim myself which also makes debugging more difficult.