corundum/corundum #208: About DNPCIe_40G_KU_LL_2QSFP board test

hmz2015011117, 2024-09-12:
Hello
The board model I have is DNPCIe_40G_KU_LL_2QSFP, so I opened the project under your corresponding model, but found that some pins do not match. What I want to know is whether the current project can run directly on the DNPCIe_40G_KU_LL_2QSFP board without modification? Have you tested it on this board? If not, what parts should I modify to make it run smoothly? Looking forward to your reply! Thank you very much!

For example, the following is the sfp+ pinout in the DNPCIe_40G_KU_LL_2QSFP board manual. The pins in your project are qsfp+, and there are many more pins than the DNPCIe_40G_KU_LL_2QSFP board.
[image]

The reference clock in the PCIe interface also does not match, as shown in the following figure:
[image]
[image]

alexforencich, 2024-09-12:
Isn't that documentation for the kcu105, not the dining board?

hmz2015011117, 2024-09-12:
Emm, I understand... I will try to modify the pins to the K105 board and test it.