corundum/corundum #209: App block development restrictions

comapny99, 2024-09-12:
Hi @alexforencich ,
    I connected the developed IP to the s_axis_if_tx  interface (250M) of the app block, and found that as long as my IP latency is more than 6ns, after burning into ZCU106, I can't ping the connection and receive no signal. After dumping it, I found that the TXQ is blocked. 

The latency of the IP must be changed to less than 4ns or the output rate of the IP must be reduced, that is, only one output for two clocks can ping the connection smoothly. 

I would like to ask you what design restrictions there are for app blocks? 

For example, in the situation I encountered, how should I deal with or set up the settings on corundum when encountering such a problem?

alexforencich, 2024-09-12:
The latency should not be a problem (within reason). Something else is going on here, but you haven't provided enough information.

comapny99, 2024-09-13:
Hi @alexforencich, 
dump.log :
PCIe ID: 0000:05:00.0
Control region size: 16777216
Application region size: 16777216
Device-level register blocks:
 type 0xffffffff (v 0.0.1.0)
 type 0x0000c007 (v 0.0.1.0)
 type 0x0000c000 (v 0.0.1.0)
 type 0x0000c005 (v 0.0.2.0)
 type 0x0000c006 (v 0.0.1.0)
 type 0x0000c080 (v 0.0.2.0)
 type 0x0000c081 (v 0.0.1.0)
 type 0x0000c008 (v 0.0.1.0)
 type 0x0000c110 (v 0.0.1.0)
 type 0x0000c101 (v 0.0.1.0)
 type 0x0000c150 (v 0.0.1.0)
FPGA ID: 0x04730093
FPGA part: XCZU7
FW ID: 0x00000000
FW version: 0.0.1.0
Board ID: 0x10ee906a
Board version: 1.0.0.0
Build date: 2024-09-12 09:06:59 UTC (raw 0x66e2af33)
Git hash: 00000000
Release info: 00000000
Application ID: 0x12340001
IF offset: 0x00000000
IF count: 2
IF stride: 0x00800000
IF CSR offset: 0x00040000
PHC ctrl: 0x00010100
PHC time (ToD): 1726134418.428620667 s
PHC time (rel): 122729503436 ns
PHC period:     6.399999999 ns (raw 0x6 ns 0x66666666 fns)
PHC nom period: 6.399999999 ns (raw 0x6 ns 0x66666666 fns)
PHC perout ch 0 ctrl:   0x00010000
PHC perout ch 0 start:  0.000000000 s
PHC perout ch 0 period: 0.000000000 s
PHC perout ch 0 width:  0.000000000 s
Ref clock nominal period: 6.400000000 ns (raw 32/5 ns)
Ref clock nominal freq: 156.250000 MHz
Core clock nominal period: 4.000000000 ns (raw 4/1 ns)
Core clock nominal freq: 250.000000 MHz
Core clock freq: 249.540864 MHz
CH0: clock freq: 156.250112 MHz
CH1: clock freq: 156.250112 MHz
CH2: clock freq: 156.250112 MHz
CH3: clock freq: 140.255232 MHz
Interface-level register blocks:
 type 0x0000c001 (v 0.0.4.0)
 type 0x0000c010 (v 0.0.4.0)
 type 0x0000c020 (v 0.0.4.0)
 type 0x0000c030 (v 0.0.4.0)
 type 0x0000c031 (v 0.0.4.0)
 type 0x0000c090 (v 0.0.2.0)
 type 0x0000c002 (v 0.0.2.0)
 type 0x0000c004 (v 0.0.3.0)
IF features: 0x00001f11
Port count: 1
Scheduler block count: 1
Max TX MTU: 9214 B
Max RX MTU: 9214 B
TX MTU: 1514 B
RX MTU: 1514 B
TX FIFO depth: 32768 B
RX FIFO depth: 65536 B
EQ offset: 0x000c0000
EQ count: 64
EQ stride: 0x00000010
[... 718 lines cut ...]
Index 1: 105473
Index 5: 46
Index 10: 750054
Index 11: 82071
Index 12: 105473
Index 13: 858
Index 16: 38
Index 17: 18
Index 21: 82039
Index 26: 246380
Index 27: 858
Index 28: 223
Index 29: 82127
Index 32: 38
Index 33: 3393
Index 34: 23929
Index 36: 38
Index 37: 23511
Index 39: 1
Index 40: 2
Index 48: 18
Index 49: 886
Index 50: 672
Index 52: 18
Index 53: 654

comapny99, 2025-04-25:
Resolved.
I later found out that my IP was not connected to the axis's side channel in the app block.