corundum/corundum #210: Missing bytes of MTU-size frame in case of vlan tagging with mqnic@mpsoc

vborchsh, 2024-09-16:
Hello all,

I found weird behavior of mqnic, maybe somebody will clarify it.

1. Run mqnic on Zynq MPSoC
2. MTU is 1500, default Linux value
3. Run `ping X.X.X.X -s 1472`
4. Everything is fine, ping is here
5. Attach vlan tag (600, 0x258) to the interface
6. Run `ping X.X.X.X -s 1472` on this interface (the same interface, but with vlan)
7. No ping

tcpdump on the other side is (partially):

```
02:43:03.312975 7e:24:02:f5:a3:ca (oui Unknown) > e6:49:83:5b:8f:3b (oui Unknown), ethertype 802.1Q (0x8100), length 1514: vlan 600, p 0, ethertype IPv4 (0x0800), truncated-ip - 4 bytes missing! (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto ICMP (1), length 1500) 10.0.0.1 > 10.0.0.0: ICMP echo request, id 10848, seq 2, length 1480
```

So, here are 4 bytes missing for some reasons. If I'll reduce the size in `ping -s XXX` the missing bytes counter will be reduced as well to 4-3-2-1 bytes. And after 1 missing byte (-s 1469) next packet size will be fine (-s 1468).

Below are full dumps of rx side packet (4 bytes missing) and tx side.

<details>
<summary>Full RX packet</summary>
<pre>
02:49:48.409908 7e:24:02:f5:a3:ca (oui Unknown) > e6:49:83:5b:8f:3b (oui Unknown), ethertype 802.1Q (0x8100), length 1514: vlan 600, p 0, ethertype IPv4 (0x0800), truncated-ip - 4 bytes missing! (tos 0x0, ttl 64, id 41969, offset 0, flags [none], proto ICMP (1), length 1
500)                                                                                                                                                                                                                                                                           
    10.0.0.1 > 10.0.0.0: ICMP echo reply, id 835, seq 0, length 1480                                                                                                                                                                                                           
        0x0000:  e649 835b 8f3b 7e24 02f5 a3ca 8100 0258                                                                                                                                                                                                                       
        0x0010:  0800 4500 05dc a3f1 0000 4001 bd2f 0a00                                                                                                                                                                                                                       
        0x0020:  0001 0a00 0000 0000 8695 0343 0000 2be8                                                                                                                                                                                                                       
        0x0030:  4a3f 0000 0000 0000 0000 0000 0000 0000                                                                                                                                                                                                                       
        0x0040:  0000 0000 0000 0000 0000 0000 0000 0000                                                                                                                                                                                                                       
        ...
        0x05c0:  0000 0000 0000 0000 0000 0000 0000 0000
        0x05d0:  0000 0000 0000 0000 0000 0000 0000 0000
        0x05e0:  0000 0000 0000 0000 0000
</pre>
</details>

<details>
<summary>Full TX packet</summary>
<pre>
02:43:22.346826 7e:24:02:f5:a3:ca (oui Unknown) > e6:49:83:5b:8f:3b (oui Unknown), ethertype 802.1Q (0x8100), length 1518: vlan 600, p 0, ethertype IPv4 (0x0800), (tos 0x0, ttl 64, id 41969, offset 0, flags [none], proto ICMP (1), length 1500)                            
    10.0.0.1 > 10.0.0.0: ICMP echo reply, id 835, seq 0, length 1480                                                                                                                                                                                                           
        0x0000:  e649 835b 8f3b 7e24 02f5 a3ca 8100 0258                                                                                                                                                                                                                       
        0x0010:  0800 4500 05dc a3f1 0000 4001 bd2f 0a00                                                                                                                                                                                                                       
        0x0020:  0001 0a00 0000 0000 8695 0343 0000 2be8                                                                                                                                                                                                                       
        0x0030:  4a3f 0000 0000 0000 0000 0000 0000 0000                                                                                                                                                                                                                       
        0x0040:  0000 0000 0000 0000 0000 0000 0000 0000                                                                                                                                                                                                                       
        ...
        0x05c0:  0000 0000 0000 0000 0000 0000 0000 0000
        0x05d0:  0000 0000 0000 0000 0000 0000 0000 0000
        0x05e0:  0000 0000 0000 0000 0000 0000 0000
</pre>
</details>

As far as I can see that everything is ok with packet on TX side, but on RX side I definitely see 4 less bytes. These bytes are placed in the end of frame. I had the other examples with counter in bytestream. Missing 1 byte on the pictures below:

TX side
[image]

RX side
[image]

The same story for the different MTU sizes. I didn't dig deeper yet, I feel that I missing something...

So, I wonder, what is the reason of these missing bytes in the end of frame case of VLAN tagging? 

Thanks!

alexforencich, 2024-09-16:
Strange.  I suspect this is not a hardware problem, because the hardware doesn't look at VLAN tags at all.  But it definitely could be a driver issue, either with how the driver interacts with the network stack relating to VLAN tagging, or potentially relating to how the driver is configuring the hardware in terms of the MTU size.

We should confirm what the "expected" behavior is relating to MTU sizes and VLAN tags.  I suspect there could be a problem between the MTU size passed to iproute2 and the the MTU size that the driver configures for the hardware when VLAN tags are being used.  Possibly the solution is a simple driver change to add at least one VLAN tag to the max packet size configured in HW.

vborchsh, 2024-09-17:
Thanks Alex,

I'll do some debug according to your messages and I'll come back with details.

vborchsh, 2024-09-17:
We did a fix according to your comment and now it works #211

closed 2024-09-17 by vborchsh