corundum/corundum #212: How is the corundum rate 10G calculated?

comapny99, 2024-09-25:
Hi @alexforencich,

I see from cpl_queue_manager.v
 if (PIPELINE < 2) begin
 $error("Error: PIPELINE must be at least 2 (instance %m)");
 $finish

This is the fastest pipeline=2 of cpl_queue_manager representing corundum, right? , so the fastest pipeline of app_block is also 2, and this is also the case in my actual test. App_block is connected to my ip. If the ip uses pipeline=1, the ip latency can only be within 4 clocks. If pipeline=2 is used, it will pass normally.

The frequency of app_block's s_axis_if_tx_tdata is 250MHz, the input signal width is 64bit, and the pipeline of app_block must be the same as cpl_queue_manager, so the fastest is only pipeline=2, so the transmission rate is not 64*250/2=8G bit/s?
But can’t corundum reach 10G bit/s?

alexforencich, 2024-09-26:
Not sure what you're asking.  The completion queue manager has nothing to do with the actual datapath.  The datapath can accept data on every cycle, so 64 bits at 250 MHz is 16 Gbps, which is sufficient to hit 10G line rate.

comapny99, 2024-09-26:
Okay, I understand, thank you very much.