corundum/corundum #226: Change clk period to meet timing constraints

MarioPatetta, 2024-12-09:
Hello,

I am currently implementing a custom block app, for which I am failing to meet timing constraints on the Alveo U200 board.

>   Path Group:             pcie_user_clk
>   Path Type:              Setup (Max at Slow Process Corner)
>   Requirement:            4.000ns  (pcie_user_clk rise@4.000ns - pcie_user_clk rise@0.000ns)
>   Data Path Delay:        4.718ns  (logic 1.616ns (34.252%)  route 3.102ns (65.748%))
>   Logic Levels:           16  (CARRY8=3 LUT3=1 LUT4=3 LUT5=3 LUT6=6)

Interestingly, most of the delay is due to routing, even though the hw utilization is modest:

[image]

As I plan on eventually upscaling the design, I don't think that trying to fidget with it to achieve timing closure is a viable option. I rather think increasing the clk period to 5 ns would work better, as 200 MHz with a TDATA WIDTH of 512 bits would still result in more than 100 Gbps, which fits my requirements. 

That said, I have tried to change the line `dict set params CLK_PERIOD_NS_NUM "4"`  to `dict set params CLK_PERIOD_NS_NUM "5"` in the config.tcl file, but that did not change the outcome. The timing constraints are still violated, and the required clock is still 4.000ns. Am I missing something? 

Thanks in advance :)

PS. I have also tried changing the routing strategy by adding `set_property strategy Congestion_SpreadLogic_low [get_runs impl_1]` to config.tcl, as I have seen in [another issue](https://github.com/corundum/corundum/issues/199), but that also had no impact on the timing situation.

Basseuph, 2024-12-11:
Hi,

Facing timing issues before upscaling points more into the direction of RTL issues, rather than tool strategy related problems.

Before optimizing the tool flow, it is most of the times benefitial to look at the number of logic levels implemented within the additional app logic. Another pipeline register stage here or there will probably help and most likely would be required for meeting timing after upscaling anyway. The easiest way to diagnose these type of problems is to use the respective elaboration or synthesis phase reports.

Better not rely on the signals that fail after PAR as the way the modern tools work, they try to fit the most complex logic and paths first. So, in case they close timing some not so bad paths do not make timing as they need to be routed around the complex logic, which then makes them fail to meet timing, although they are not the ones we should look at at first and fix.

Have successful RTL optimization session!

MarioPatetta, 2024-12-12:
Hi @Basseuph thanks for the insight!

I will try to have a look at the elaboration and synthesis reports.

Still, I would like to have the possibility to play with the clk period eventually, do you have any idea on why changing the `CLK_PERIOD_NS_NUM` parameter had no impact?

alexforencich, 2024-12-12:
That parameter is just like the timing constraints - it's informational.  To change the PCIe user clock period, you have to change it at the source, the PCIe HIP itself.  Looks like if you drop it down to gen 2 x8, you can set the clock to 125 MHz.  You'll also have to change all of the PCIe interface config settings to match - 256 bits instead of 512, etc.

At some point, I might decouple the core clock from the PCIe HIP, mainly to avoid spread spectrum clocking of the core logic.  But this involves adding a bunch of async FIFOs and other CDC logic, and will likely increase the latency a bit.

MarioPatetta, 2024-12-13:
Ah, I see! Thanks for the explantation. 

Probably 125 MHz would be too low for my purpose, but that's still good to know.

MarioPatetta, 2024-12-30:
Hello!

I have been working at optimising the design, reducing the violations and introducing some improvements here and there.

I am writing again to ask some advice on a couple of "big" changes I would like to introduce, which I expect would have a massive impact on timing and resource usage. The changes would be:

1. Temporarily store received packets in the on-board DRAM while my block app processes/schedules them. Currently, packets are categorized and buffered in per-flow FIFOs.

2. Instantiate four interfaces using one of the two QSFP ports with the 25G MAC modules, and another interface using the other QSFP port with the 100G CMAC module.

For change 1, I guess I should make use of the DMA interface of the block app for this purpose. However, I'm unclear about how the DMA interface works and how to distinguish between the one related to the host machine and the DRAM. I had a look at the dma_bench application, but I could use some extensive documentation or any resource you may have.

I have no clue about change 2, I hope you have some tips!