corundum/corundum #229: netFPGA-SUME version rx packet does not generate interrupt

Zyy439, 2025-02-03:
Hi 
i was trying the netFPGA-SUME (Vertix 7) version, but some problems encountered~ its gonna be so appreciated if you could offer some help.
basically what i did is like this:
i connected corundum with another PC and set corundum's IP as 192.168.98.100, trying to ping the other PC. i could see there are ARP packets received by the other PC(from wireshark), showing that corundum was sending packets correctly~  However, corundum could not receive any packet as Rx count was always 0 in the ifconfig result~

[image]

[image]

then i checked the tool and there events in the EQ and RXQ~

[image]

feels like that the interrupt was somewhat abnormal ,making the rx events are not handled. 

[image]

[image]

then i dumped some siganls of the "cpl queue manager" module , it seems like that some queues are not armed when a packet is received~ 

[image]

[image]

[image]

i found that the interface can receive packets normally

[image]
then i tried to forcce it generate interrupts even if its not armed

[image]
however that didnt work either i was soooo sad

could you please offer some hint on this? sorry to bother  and thank you very much!!!!

alexforencich, 2025-02-03:
Virtex 7 is old and slow and Corundum often does not close timing on the Virtex 7.  What does Vivado report for the WNS and TNS?

Zyy439, 2025-02-03:
> Virtex 7 is old and slow and Corundum often does not close timing on the Virtex 7. What does Vivado report for the WNS and TNS?
thank you so much for the reply!
Sorry im not having my PC beside but im sure that WNS is between -0.5ns and -1ns ,TNS is around -100ns.
the project sets parameter IF_COUNT=2 as default and to optimize timing  i changed the parameter as 1~  so it may be a bit better in terms of timing

alexforencich, 2025-02-03:
Yeah, that's pretty bad, who knows what odd stuff might break with that kind of timing failure.  Both WNS and TNS should ideally be non-negative (positive WNS, zero TNS).  If the TNS is better than maybe -1 ns or you can verify in the timing report that only CDC paths are failing, then it might work OK, otherwise you're likely to have various strange issues.

gongyanhui, 2025-05-09:
Hi Any update regarding this issue @Zyy439 , recently I have same issue with the one describled above, so far have no idea yet. Any information will be appreciated.

Zyy439, 2025-05-09:
Ahh so sorry that i forgot to update here～
The netFpga sume version sets parameter IF_COUNT as 2 in default and so that the pcie gen3 ip core's msix table size was 64. One interface will use 32 different msix entries 
The driver will configure those msix address and data in a RAM discripted in pcie_msi.v ~  somehow the first interface's interrupts are generated using 32-63 entry while they are supposed to use the first 32 entries~(im still looking for reason for this haha) at the time i try it i just pluged the first interface and so that the driver detected no interrupt according to that interface

Then i changed the pcie ip msix table size as 32 and the Nic then worked fine.

gongyanhui, 2025-05-09:
@Zyy439 Thank you so much!!

Zyy439, 2025-05-12:
no worries man if you are willing to communicate more about this plz +vx zhouyuyangdashabi