corundum/corundum #232: How can I burn corundum into Alveo U50 and load the driver correctly?

comapny99, 2025-04-14:
Hi @alexforencich , 
I would like to ask you how can I burn corundum into Alveo U50 and load the driver correctly?
Thank you, very much.

I used zcu106 very smoothly, but when using Alveo U50, after make fpga_25g/fpga_10g/Makefile, I used Vivado HARDWARE MANAGER to burn fpga.bit into U50 (by Alveo programming cable and connected to the host connection port: model DMB-1), and restarted the computer, make driver and loaded it. The result was that pcie was detected, but there was a problem with driver loading. I would like to ask if there is any step I did wrong?
The log is as follows:

zyell@zyell-Server:~/Project/corundum-master-version-2023-12-2-1140411-u50/corundum-master/modules/mqnic$ **lspci -nn | grep Xilinx**
05:00.0 Processing accelerators [1200]: Xilinx Corporation Alveo U50 XMDA Platform [10ee:5020]
05:00.1 Processing accelerators [1200]: Xilinx Corporation Device [10ee:5021]

zyell@zyell-Server:~/Project/corundum-master-version-2023-12-2-1140411-u50/corundum-master/modules/mqnic$ **sudo dmesg | grep mqnic**
[144.764123] mqnic: loading out-of-tree module taints kernel.
[ 144.764127] mqnic: module verification failed: signature and/or required key missing - tainting kernel

zyell@zyell-Server:~$ **uname -a**
Linux zyell-Server 6.5.0-18-generic #18~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 7 11:40:03 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

PS. The corundum version I used was downloaded on 11/04/11, and the secure mode of the OS has been disabled. The zcu106 is working properly.

alexforencich, 2025-04-14:
dmesg is reporting that the card is running the Alveo image, so it must have been reset when the server rebooted.  You may need to generate the mcs file and actually flash it from Vivado, instead of simply loading the bit file into the FPGA.

comapny99, 2025-04-17:
Thank you, very much.
I will try it.

comapny99, 2025-04-25:
Resolved
U50 uses the I0G module project.
The dedicated line uses 40G to 4*10G optical fiber lines.
If port = 1 in config.tcl (default value is the first line), the fiber optic cable connecting to the partner must be connected to the first 10G line to get data.

closed 2025-06-12 by comapny99