corundum/corundum #233: How to port Corundum to the Intel Agilex® 7 FPGA I-Series

xingyunzhidi, 2025-04-14:
Currently, Corundum has been successfully ported to the Intel Agilex® 7 FPGA F-Series. I have observed that the Intel Agilex® 7 FPGA F-Series uses the 100G E-tile Ethernet IP and P-tile PCIe IP. However, the Intel Agilex® 7 FPGA I-Series only has F-tile Ethernet IP and R-tile PCIe IP. If I want to port Corundum to the Intel Agilex® 7 FPGA I-Series, do I only need to modify the IP and pin constraints?

alexforencich, 2025-04-14:
No, the R-tile has quite a few differences that would need to be addressed in the wrapper.  Primarily this is related to flow control, but IIRC there are several other differences as well.

My plan is to support the R-tile natively up to gen 5 x16 in the SV version of Corundum that will be built on top of taxi.  It will not be supported in the legacy version of Corundum (this repo).

xingyunzhidi, 2025-04-18:
Thank you for your response. I would like to inquire when the SV version of Corundum is expected to be released. Additionally, if I would like to switch to the R-tile PCIe IP and F-tile Ethernet IP, which specific .v files would need to be modified?

alexforencich, 2025-04-18:
I don't have a timeline for that at the moment, but probably late this year or early next year.

Long term, R-tile will get a new shim.  I'm going to do a pretty significant overhaul of the entire DMA subsystem in order to support PCIe gen 5 x16 (or at least gen 4 x16), that will take a fair amount of time to complete.  But it would likely be possible to modify the existing P-tile shim to work with the R-tile for gen 3x16 and possibly gen 4x8.  This would be the pcie_ptile_if* stuff.  For F-tile, I think mainly this would only affect the top-level integration, assuming the F-tile supports AXI stream.  Long term I'm thinking about doing something with segmented interfaces to replace AXI stream to support 400G, but that will probably be a while.

Are you using this for a commercial application?  If you're interested in licensing the new version of Corundum, I may be able to speed things up.

xingyunzhidi, 2025-04-22:
Thank you very much for your response. Currently, there are no plans to commercialize corundum.