corundum/corundum #234: Implicit behaviour of AXI streams in app block?

harris-chris, 2025-04-15:
Hello, 

I've put some placeholder code in the `mqnic_app_block`, in the `*_axis_sync_*` tx path. This code should have no effect, yet I notice that it's causing the `test_fpga_core.py` cocotb test to fail; the packet that gets received from the qsfp_sink is no longer exactly the same as that which was sent via `start_xmit` in the test (in the "Send and receive single packet" part of the test).

I've added some debug to the verilog, and the only difference that my placeholder makes is to the final chunk of the packet. I have split the original `s_axis_sync_tx_tdata` signal out into its two respective ports, and am capturing debug from only one of the two ports (on the `ADM_PCIE_9V3`) so the `tkeep` signal is 16 bits, the `tdata` signal 128 bits.

The placeholder code returns:
```
INFO cocotb: p2 out tkeep: 1111111111000000
INFO cocotb: p2 out tdata: fffefdfcfbfaf9f8f7f6000000000000
```

Whereas without any placeholder code - ie, if the `mqnic_app_block` is entirely unchanged, this signal would have been:
```
INFO cocotb: p2 source tkeep: 0000001111111111
INFO cocotb: p2 source tdata: 000000000000fffefdfcfbfaf9f8f7f6
```

This difference then causes 6 empty bytes to be added to the packet received from the qfsp_sink, causing the test to fail. 

From an AXI4-stream perspective, I'd expect these two signals to be equivalent, so I was wondering if there's an implicit contract here that the AXI stream chunks be right-justified (so to speak)?

alexforencich, 2025-04-15:
AXI stream spec 2.4.1: "It is not mandatory for masters and slaves to handle null bytes"

None of my code supports null bytes aside from at the end of a packet (corundum, verilog-axis, verilog-axi, etc.).  So you have to remove them beforehand.  First byte must go in lane 0 (bits 7:0), the rest must follow contiguously.  tkeep must be all 1s for every transfer unless tlast is set.  Supporting arbitrary null bytes would add a LOT of logic overhead.

harris-chris, 2025-04-15:
Thanks for the clarification!